MarlinFirmware/Marlin #7963: G29/UBL Mesh Coverage(Area Definition?) Issue

mapodann, 2017-10-13:
### Bug Report
- Description: --- UBL Correction Mesh does not start making corrections until X>=60 with current settings. It will function for the full range of Y from 0-200.
- Expected behaviour: --- For correction mesh to start at X,Y == 0,0
- Actual behaviour: --- No correction applied for X<60

I have used new G29 P1 measurements, loaded old ones from memory. Gone through and reconfigured my bed size definitions, mesh insets, mesh min/max, 

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

I assume this is all due to some configuration error, however I am entirely unable to find it. 

Bonus issue, my initial probe spot moves into the airspace in front of my print bed (build platform does not start until Y==35) for the closest probe point, (-5.44, -13.00). I have been able to get the probe to successfully cover the entire bed in the past, however I believe that I have the config files correctly setup, so perhaps this extra issue can help point sharper minds to my mistake?

Here is the debug printout for a G29 P1 call:
18:59:07.566 : echo:DEBUG:LEVELING
18:59:15.771 : Default storage slot 0 selected.
18:59:15.771 : Mesh invalidated. Probing mesh.
18:59:15.771 : current_position=(0.00, 0.00, 15.00) : set_probe_deployed
18:59:15.771 : deploy: 1
18:59:15.771 : do_probe_raise(3.00)
18:59:15.771 : >>> do_blocking_move_to(0.00, 0.00, 15.00)
18:59:15.771 : <<< do_blocking_move_to
18:59:15.793 : >>> probe_pt(39.56, 0.00, no stow)
18:59:15.793 : current_position=(0.00, 0.00, 15.00) :
18:59:15.793 : >>> do_blocking_move_to(-5.44, -13.00, 15.00)
18:59:15.947 : <<< do_blocking_move_to
18:59:15.947 : current_position=(-5.44, -13.00, 15.00) : set_probe_deployed
18:59:15.947 : deploy: 1
18:59:15.947 : current_position=(-5.44, -13.00, 15.00) : >>> run_z_probe
18:59:15.947 : current_position=(-5.44, -13.00, 15.00) : >>> do_probe_move
18:59:15.947 : >>> do_blocking_move_to(-5.44, -13.00, 4.70)
18:59:20.089 : Printer reset detected - initalizing
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/1381345/Configs.zip)

Roxy-3D, 2017-10-13:
You should post a G29 T and a G29 W.   It is possible there is no correction with X<60 because the mesh points are almost 0.0 mm.   I don't know.   But you should turn on
```cpp
  //#define UBL_G26_MESH_VALIDATION // Enable G26 mesh validation
```
and post a picture of the Mesh Validation Pattern.   That will tell us a lot about what needs to be done.

mapodann, 2017-10-13:
Alright, Having gone through these with fresh eyes this morning things clicked. After updating firmware and setting up a new mesh, I neglected to fill in all of the spots my probe could not reach. Apologies for the bother. I will have to go back through the G29 P* process and get those values defined. That would be why the mesh had no compensation for those values and it kicked in at X==60. So problem solved, thank you for being to ready and willing to help!

The "Bonus Issue" seems to be that the firmware is shifting all of the X&Y Axis Mesh Points by the probe offset values and then attempting to send the nozzle point to that shifted value, checking if the shifted coordinates are reachable. It would seem that ensuring that the shifted coordinates are (0,0)<=(X,Y)<=(X_BED_SIZE,Y_BED_SIZE) would make more sense. Or perhaps building out the X&Y Axis Mesh Points by first determining the closest that the probe can be put to (0,0), in my case that would be nozzle position (-8,13) which would result in the probe position being (37,0). From there, subtract that value of 5 from X_BED_SIZE, then redo the calculations of the X&Y Mesh positions to use the area of the bed that is reachable by the probe. Specific to my printer I see this approach doing away with a good amount of manual probing, but more importantly ensuring the probe is always being dropped over the bed.

G29 T:

12:43:15.657 : Bed Topography Report:
12:43:15.683 : (0,9)                                                                   (9,9)
12:43:15.706 : (0,200)                                                                (178,200)
12:43:15.882 : .          .          .      -0.015    0.017    0.082    0.108    0.125    0.086    0.039
12:43:15.882 : 
12:43:16.042 : .          .          .      -0.014    0.007    0.082    0.118    0.133    0.087    0.053
12:43:16.042 : 
12:43:16.201 : 0.027    0.090    0.117    0.112    0.093    0.062
12:43:16.202 : 
12:43:16.377 : .          .          .      -0.009    0.021    0.077    0.103    0.111    0.107    0.055
12:43:16.377 : 
12:43:16.537 : .          .          .      -0.011    0.001    0.067    0.087    0.118    0.073    0.039
12:43:16.537 : 
12:43:16.697 : .          .          .      -0.017   -0.001    0.039    0.088    0.108    0.053    0.020
12:43:16.697 : 
12:43:16.840 : .          .          .      -0.037   -0.026    0.027    0.054    0.057    0.038   -0.005
12:43:16.841 : 
12:43:17.000 : .          .          .      -0.063   -0.063   -0.005    0.009    0.017    0.003   -0.035
12:43:17.000 : 
12:43:17.160 : -0.024   -0.039   -0.070
12:43:17.160 : 
12:43:17.333 : [    .   ]      .          .      -0.411   -0.377   -0.287   -0.250   -0.212   -0.206   -0.209
12:43:17.333 : (0,0)                                                                    (178,0)
12:43:17.357 : (0,0)                                                                     (9,0)

G29 W:

12:44:44.434 : echo:Home XYZ first
12:44:44.434 : Unified Bed Leveling System v1.01 inactive.
12:44:44.482 : Mesh 1 Loaded.
12:44:44.546 : UBL object count: 1
12:44:44.546 : planner.z_fade_height : 10.0000
12:44:44.546 : # of samples: 70
12:44:44.546 : Mean Mesh Height: 0.001239
12:44:44.546 : Standard Deviation: 0.111769
12:44:44.546 : zprobe_zoffset: -1.7000000
12:44:44.546 : UBL_MESH_MIN_X  0=0
12:44:44.546 : UBL_MESH_MIN_Y  0=0
12:44:44.573 : UBL_MESH_MAX_X  (178 - (0))=178
12:44:44.573 : UBL_MESH_MAX_Y  (200 - (0))=200
12:44:44.589 : GRID_MAX_POINTS_X  10
12:44:44.589 : GRID_MAX_POINTS_Y  10
12:44:44.620 : MESH_X_DIST  19.78
12:44:44.621 : MESH_Y_DIST  22.22
12:44:44.908 : X-Axis Mesh Points at: 0.000  19.778  39.556  59.333  79.111  98.889  118.667  138.444  158.222  178.000
12:44:45.163 : Y-Axis Mesh Points at: 0.000  22.222  44.444  66.667  88.889  111.111  133.333  155.556  177.778  200.000
12:44:45.212 : ubl_state_at_invocation :0
12:44:45.212 : ubl_state_recursion_chk :0
12:44:45.259 : Meshes go from 0x0270 to 0x0F7F
12:44:45.307 : sizeof(ubl) :  1
12:44:45.308 : z_value[][] size: 400
12:44:45.339 : EEPROM free for UBL: 0x0D0F
12:44:45.388 : EEPROM can hold 8 meshes.
12:44:45.420 : Unified Bed Leveling sanity checks passed.

Roxy-3D, 2017-10-13:
You should fill in the unprobed dots of the mesh before trying to use the printer.   Technically, that isn't necessary.   But it is much easier to understand what is happening with the mesh filled in.   (In the G29 you posted...   Much of the mesh is empty.)    And once the mesh is filled in, please do a G26 and post the picture.   That will help us suggest what needs to be done to the mesh to make it perfect.

mapodann, 2017-10-14:
After realizing that earlier it has fixed the issue I was having at X==60. If my other idea produces useful work I will perhaps open a new feature request. I don't think I need to bother anyone else with me tweaking the mesh, thank you though!

closed 2017-10-14 by mapodann