MarlinFirmware/Marlin #7968: first reading from ADC without ADC start ADC in temperature.cpp

Evgen2, 2017-10-13:
temperature.cpp , version 1.1.x and 2.0

in fuction  void Temperature::isr() 
  static ADCSensorState adc_sensor_state = StartupDelay;
```cpp
  switch (adc_sensor_state) {
......
    #if HAS_TEMP_0
      case PrepareTemp_0:
        HAL_START_ADC(TEMP_0_PIN);
        break;
      case MeasureTemp_0:
        raw_temp_value[0] += HAL_READ_ADC;
        break;
    #endif
// first is executed   case MeasureTemp_0: not   PrepareTemp_0
// so first ADC value is not real value. If ADC reading use DONE flag, then we don't return from reading..
}
  // Go to the next state, up to SensorsReady
 adc_sensor_state = (ADCSensorState)((int(adc_sensor_state) + 1) % int(StartupDelay));
```
---------------
with  adc_sensor_state = StartupDelay;
the result of
`adc_sensor_state = (ADCSensorState)((int(adc_sensor_state) + 1) % int(StartupDelay));`
will be  1 (MeasureTemp_0), not 0 ( MeasureTemp_0) as intended

seems that above string should be changed to
`adc_sensor_state = (ADCSensorState)((int(adc_sensor_state) + 1) % int(StartupDelay+1));`

AnHardt, 2017-10-13:
Looks as if you are right. The very first reading after a reset of the first sensor is messed up.

But the intention here is to never reach the StartupDelay state again. In your solution we will have the delay state in every round.

Because directly after a reset no heater can be on, the current code is a purely cosmetic problem.

AnHardt, 2017-10-13:
Thinking about that we also could initialize with

static ADCSensorState adc_sensor_state = 0;

delete the StartupDelay state and
calculate the next step with

adc_sensor_state = (ADCSensorState)((int(adc_sensor_state) + 1) % int(SensorsReady+1));

A proper solution could be

   case StartupDelay: adc_sensor_state = SensorsReady; break;

Evgen2, 2017-10-14:
If the intention  is to never reach the StartupDelay state again and we use (x +1) % then may be solution is
 
```
enum ADCSensorState {
  StartupDelay = -1,  // Startup, delay initial temp reading a tiny bit so the hardware can settle
  #if HAS_TEMP_0
    PrepareTemp_0,
    MeasureTemp_0,
  #endif
  #if HAS_TEMP_1
.....
  SensorsReady // Temperatures ready. Delay the next round of readings to let ADC pins settle.
};

```
and
adc_sensor_state = (ADCSensorState)((int(adc_sensor_state) + 1) % int(SensorsReady+1));

Evgen2, 2017-10-14:
https://github.com/MarlinFirmware/Marlin/commit/ca665740627cfd4e5f9fccc562a05cede1a9abf1

closed 2017-10-14 by Evgen2