MarlinFirmware/Marlin #7969: don't know what am I doing wrong

NoWizard, 2017-10-13:
//file Configuration.h Release:tags/1.1.6
//section info
#define STRING_CONFIG_H_AUTHOR "(bob, Anet config)"
//section machine
#define BAUDRATE 115200
#define MOTHERBOARD BOARD_ANET_10
#define INVERT_Y_DIR false
#define INVERT_Z_DIR true
#define X_BED_SIZE 220
#define Y_BED_SIZE 275
#define X_MIN_POS -33
#define Y_MIN_POS -10
#define Z_MAX_POS 240
//section temperature
#define TEMP_SENSOR_0 5
#define TEMP_SENSOR_BED 5
#define TEMP_RESIDENCY_TIME 6
#define TEMP_BED_RESIDENCY_TIME 6
#define BED_MAXTEMP 130
#define PID_FUNCTIONAL_RANGE 15
#define DEFAULT_Kp 21.0
#define DEFAULT_Ki 1.25
#define DEFAULT_Kd 86.0
#define BED_LIMIT_SWITCHING 
#define PREHEAT_1_TEMP_HOTEND 190
#define PREHEAT_1_TEMP_BED 60
#define PREHEAT_2_TEMP_BED 90
//section homing
#define X_MIN_ENDSTOP_INVERTING true
#define Y_MIN_ENDSTOP_INVERTING true
#define Z_MIN_ENDSTOP_INVERTING true
#define Z_MIN_PROBE_ENDSTOP_INVERTING true
#define ENDSTOP_INTERRUPTS_FEATURE 
#define HOMING_FEEDRATE_XY (100*60)
//section motion
#define DEFAULT_AXIS_STEPS_PER_UNIT { 100, 100, 400, 95 }
#define DEFAULT_MAX_FEEDRATE { 400, 400, 8, 50 }
#define DEFAULT_MAX_ACCELERATION { 2000, 2000, 100, 10000 }
#define DEFAULT_ACCELERATION 400
#define DEFAULT_RETRACT_ACCELERATION 1000
#define DEFAULT_TRAVEL_ACCELERATION 1000
#define DEFAULT_ZJERK 0.3
//section probes
#define X_PROBE_OFFSET_FROM_EXTRUDER 0
#define Y_PROBE_OFFSET_FROM_EXTRUDER 0
#define XY_PROBE_SPEED 6000
//section bedlevel
#define RIGHT_PROBE_BED_POSITION 190
#define FRONT_PROBE_BED_POSITION 15
#define ABL_PROBE_PT_1_X 20
#define ABL_PROBE_PT_1_Y 160
#define ABL_PROBE_PT_2_X 20
#define ABL_PROBE_PT_2_Y 10
#define ABL_PROBE_PT_3_X 180
#define ABL_PROBE_PT_3_Y 10
#define UBL_G26_MESH_VALIDATION 
//section extras
#define EEPROM_SETTINGS 
//#define HOST_KEEPALIVE_FEATURE 
//#define DEFAULT_KEEPALIVE_INTERVAL 2
//section lcd
#define SDSUPPORT 
#define ANET_KEYPAD_LCD 

//file Configuration_adv.h Release:tags/1.1.6
//section temperature
#define THERMAL_PROTECTION_PERIOD 60
#define THERMAL_PROTECTION_HYSTERESIS 10
#define THERMAL_PROTECTION_BED_PERIOD 60
#define THERMAL_PROTECTION_BED_HYSTERESIS 5
#define WATCH_BED_TEMP_PERIOD 180
//#define AUTOTEMP 
//section homing
#define ENDSTOPS_ALWAYS_ON_DEFAULT 
//section extras
//#define ARC_SUPPORT 
//section lcd
//#define SD_DETECT_INVERTED

AnHardt, 2017-10-13:
Mainly giving us a proper problem description.

Tannoo, 2017-10-14:
That's a lot of work to remove all the comments. Lol

Start with following the template that was deleted. It will help us help you.

fiveangle, 2017-10-14:
To prevent creation of another issue just to read the template 😉 …

http://marlinfw.org/docs/basics/reporting_bugs.html
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/issue_template.md

closed 2017-11-03 by thinkyhead