MarlinFirmware/Marlin #7970: Questions on CNC and bed leveling

Bob-the-Kuhn, 2017-10-14:
In Issue #7907 the user was trying to use the G38 command while UBL was enabled.  That reminded me of earlier this year when I was trying to mount some snap-in connectors in Plexiglas.  I needed to hold about 0.5mm tolerance.  I gave up because my bed had a significant tilt which meant that my depth of cut varied across the bed.

Do we currently have a leveling system that would help?  I've forgotten a lot about the Marlin bed leveling systems so maybe there's already a useful system/method.

The CAD software requires an origin for the work piece to be specified.  When setting up the milling job the origin has to be located and fed back to the controller software.  Usually a corner is located and then the G92 command is used to set X0 and Y0 and Z.

My first concern is how the leveling systems handle the X & Y offsets.  Do they use the "real" coordinates when calculating the amount of leveling?  By "real" I mean the ones determined when the machine is homed.

Is adjusting the Z_offset via the M851 command a reasonable way of moving the Z plane?

Hmmm - in UBL the G29 P4 H command will raise & lower the Z plane.  Maybe UBL will do everything I need.

Roxy-3D, 2017-10-14:
>  I needed to hold about 0.5mm tolerance. I gave up because my bed had a significant tilt which meant that my depth of cut varied across the bed.

I'll re-read this in the morning.    And especially so if you reply before that.   But with a 15x15 mesh the bed would have to be tilted at some crazy angle before  the nozzle could not track it.

> My first concern is how the leveling systems handle the X & Y offsets. Do they use the "real" coordinates when calculating the amount of leveling? By "real" I mean the ones determined when the machine is homed.

Even if the X & Y don't home exactly...   If they home withing .1 mm, think about that in terms of the mesh.   You are going to have a hard time getting your piece of glass clamped down with .1 mm accuracy.   I'm way more sloppy about clamping my glass down than that.

Bob-the-Kuhn, 2017-10-14:
This would be for milling Plexiglas, aluminum, PCBs, ...

The sequence I envision is:

1. Create a mesh via UBL that covers the entire milling bed.  X0 and Y0 are established by homing to endstops.
2. Clamp my work piece to a subsection of the bed.
3. Use G92 to set X0, Y0 to the corner of the work piece.
4. Use the G29 P4 H command will raise & lower the Z plane

I've also had work pieces that were not flat.  For those I'd have to do the mesh creation on the work piece itself.

Edgar48, 2017-10-14:
@Bob the Kuhn:  I tried G29 P4 H: in my configuration the command starts to probe all 16 Points of the predefined points out of the code. In my case the predefined points correspond to the Heating bed size. In case of doing CNC the area where I can Probe is often much smaller and its not always the same position. I'm still trying to understand the G29-UBL capabilities. Is there a posibility to Tell G29 to do a 4 point Leveling starting at the current position and giving a x wide and y hight of the Probing area or manualy going to 4 points and probe?

Edgar48, 2017-10-14:
Found it !

I place the Probe over the center of the Workpiece and give the following commands:
G92 Z0
G92 X0 Y0
G29 R15 L-15  F-15 B15 T V3

this will Probe a Sqare of 30 x 30 mm around the center( -15....+15) and set Z to zero on contact plane

Roxy-3D, 2017-10-14:
Help me understand...   You want to mill into the work piece, but not cut into your bed, right?

Is it possible to clamp the work piece to the bed, but have it lifted up to provide some clearance from the work bed?   

And here is the other question I have:  If you are clamping the work piece to the bed, isn't it the top of the work piece you should be using to generate the mesh?   That is the reference surface that you want fine control of, right?

Allted, 2017-10-14:
The other way Bob would be to add the offset in the CAM so home is still home (meaning move the CAM origin to not the corner of your part). That is how I tell my users to do it. 

I have never tried to combine bed leveling with cuts yet, but if that dual end stops Pull Request I have in  goes through it is much more likely I will try it.

Bob-the-Kuhn, 2017-10-14:
@Roxy - most of the time when milling there are through holes and edges where the bit will contact the bed.  I use an oversized sacrificial bed and resurface the work area when it gets too cut up.  Because the bed gets cut up there are times that I'd like to use a mesh that was created when the work area was first surfaced.

Depending on the project, sometimes the Z0 plane is the bottom of the work piece, sometimes it's the top.

@Allted - sounds like your method results in the user trying to clamp the work piece in a specific location.  My preference is to clamp the work piece in a convenient location and then tell the computer where it's at.  

I mill T slots into my sacrificial bed for my clamps.  When clamping up my work piece I set the rough Y offset by clamping up against a gage which is pushed up against the lip of the sunken work area (which also squares up the piece to the bed).  The X offset varies depending on how the slots, clamps and work piece line up.  I'll then move my bit to a location on the work piece that provides clearance from the clamps and issue a G92 X0 Y0.

Allted, 2017-10-14:
True, usually with an initial move to mark the location. If you get this figured out I can't wait to try it. I never use the auto level (on the cnc) because of this reason, totally unsure which one to use and how for this application (so many CAM nuances to learn didn't want to throw autolevel in the mix as well). Following along to see what you come up with and can test it out along the way.

Roxy-3D, 2017-10-14:
It almost sounds like CNC's need two mesh's.   One for the bed, and one for the top surface of the part.

Bob-the-Kuhn, 2017-10-14:
@Roxy - if I do the following:
1. Build a mesh with the front left probe point being at (0,0) 
2. Move the head to (50,50)
3. Issue G92 X0 Y0

Will the Z compensation that's applied at the head's current location be for the front left probe point or the current location?

Roxy-3D, 2017-10-14:
> Will the Z compensation that's applied at the head's current location be for the front left probe point or the current location?

Please allow me change the words a little bit.  The correction is always calculated from the current destination.    Regardless of where the nozzle is, when it is moved, the correction is calculated from the desired destination.   And the planner takes care of how much the nozzle needs to move up or down (and how fast) using the the current location as the starting point.

This allows some simplifications.   Regardless of where the nozzle starts, if it is making a big, long move, only the intersections with the various mesh lines needs to be determined.   A large move can be broken up into smaller moves (that track the bed better) by just moving to the next mesh line intersection.

The front end of void unified_bed_leveling::line_to_destination_cartesian() finds the current start location and the destination based on RAW_?_POSITION().
```cpp
  void unified_bed_leveling::line_to_destination_cartesian(const float &feed_rate, uint8_t extruder) {
    /**
     * Much of the nozzle movement will be within the same cell. So we will do as little computation
     * as possible to determine if this is the case. If this move is within the same cell, we will
     * just do the required Z-Height correction, call the Planner's buffer_line() routine, and leave
     */
    const float start[XYZE] = {
                  current_position[X_AXIS],
                  current_position[Y_AXIS],
                  current_position[Z_AXIS],
                  current_position[E_AXIS]
                },
                end[XYZE] = {
                  destination[X_AXIS],
                  destination[Y_AXIS],
                  destination[Z_AXIS],
                  destination[E_AXIS]
                };

    const int cell_start_xi = get_cell_index_x(RAW_X_POSITION(start[X_AXIS])),
              cell_start_yi = get_cell_index_y(RAW_Y_POSITION(start[Y_AXIS])),
              cell_dest_xi  = get_cell_index_x(RAW_X_POSITION(end[X_AXIS])),
              cell_dest_yi  = get_cell_index_y(RAW_Y_POSITION(end[Y_AXIS]));
```

Bob-the-Kuhn, 2017-10-14:
Exactly what I was looking for.  **FANTASTIC!!!**

UBL always uses the raw position which means that, no matter what I do with G92, the correct compensation is applied. 

My depth of cut accuracy should be improved by using UBL.

----

That satisfies all my concerns.  I'll leave this open for a day or two in case there are follow on questions..

closed 2017-10-17 by Bob-the-Kuhn