MarlinFirmware/Marlin #7971: 2.7.16: Marlin fly-out on right side gets annoying

ScruffR, 2017-10-14:
When running mct 2.7.16 in Chrome on Win8.1 tablet the fly-out to select the respective settings tab keeps popping out and sliding back whenever I touch the screen to scroll the page or select an option.

Over time this gets somewhat annoying ;-)

Keeping it hidden till it's actively touched would be fine for me.

Tannoo, 2017-10-14:
Umm.. I don't think whatever you are explaining has to do with Marlin as Marlin doesn't any touch support.

ScruffR, 2017-10-14:
No, I'm talking about this fly-out in Marlin Configuration Tool on my windows tablet with touch 
[image]

closed 2017-10-14 by ScruffR