MarlinFirmware/Marlin #7972: Endstops, bed leveling and G38

Bob-the-Kuhn, 2017-10-14:
In the beginning there were no endstops and setup was a pain.

And then there were endstops and life got better.

And now we have bed leveling.  The bed probing really helps the print quality but blurred the definition of endstops a bit.

But G38 came along and confused everything.

----

G38 is broken again.  I’m scratching my head over how to implement and document the signal used by G38.

I’m proposing that a logic signal be dedicated to the G38 function.  That way changes to the bed probe logic won’t affect G38 and removes any G38 concerns when changing the bed probe logic.

It would still allow the G38 signal to be hooked up to Z_MIN or any of the other endstops.

What I’m thinking is the following:
- Create a new logical signal called G38_PROBE_PIN (or something like that).
- Set it up the same as the current endstops (pullup option and inverting option).
- Mention G38_PROBE_PIN only in the G38 section of configuration_adv.h.
- Change the endstop logic to use G38_PROBE_PIN
- Add G38_PROBE to EndstopEnum
- Change the G38 section of the Wiki.
- Add/change checks in sanity.h
- Change M119 to report G38_PROBE_PIN

----

Here’s my ramblings as to how I arrived at the above.

In a perfect world:
1.	there would be endstops
2.	a bed probe 
3.	a G38 probe
4.	the option of using the bed probe as the Z_MIN endstop
5.	the option of having a G38 probe independent of the endstops and the bed probe

The current status of bugfix-1.1.x is:
- The WIKI says the Z endstop is the only signal used by G38.  
- The endstop logic uses Z_MIN_PROBE_PIN for G38.
- Enabling Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN does NOT result in Z_MIN_PROBE_PIN being defined.
- Pins.h undefines Z_MIN_PROBE_PIN if Z_MIN_PROBE_ENDSTOP is disabled.

The Wiki says to configure the bed probe’s Z Probe Pin:
- If the probe is connected to the Z-Min pin, enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN.
- If the probe is connected to any other pin, enable Z_MIN_PROBE_ENDSTOP.
- If the probe is connected to any other pin, set Z_MIN_PROBE_PIN to its digital (DIO) pin number.

Tying the G38 probe/logic to the Z endstop isn’t always an option. Usually the G38 probe consists of a pull up resistor and something that shorts it to ground.  That’s not compatible with an endstop that’s setup as  ENDSTOP_INVERTING false.

Roxy-3D, 2017-10-15:
In a 'perfect' world... Shouldn't the Z-Probe be suitable to support G38 ???

Bob-the-Kuhn, 2017-10-16:
G38 can probe in any direction.

Roxy-3D, 2017-10-16:
> Here’s my ramblings as to how I arrived at the above.
> In a perfect world:
> there would be endstops
> a bed probe
> a G38 probe

It sounded to me like you wanted an extra probe that does nothing but G38's.   I was thinking what ever is being used for the Z-Probe should also handle G38's.

thinkyhead, 2018-02-09:
Presuming `G38` is patched by now.

closed 2018-02-09 by thinkyhead