MarlinFirmware/Marlin #7981: Board GT2560 A+ wrong servo pin

dobryj, 2017-10-14:
### Bug Report
Header for board GT2560 A+ define wrong servo pin number when BLTOUCH is enabled.
In history geeetech boards until A+ revision don't have servo connector. Therefore it exist hack to use Z-max pin to this functionality. Problem is that this hack need HW modification to solder another connector on board. This is difficult for some users.
A+ board have new functionality with servo connector on pin 11 (arduino numbering). Therefore easiest way to connect BLtouch is connect it to this connector without any HW modification.
Problem is on accepted pull request https://github.com/MarlinFirmware/Marlin/pull/7769
This modification do not solve problem on boards until A+ and devastate solution prepared on PCB.
It is extremely strange.

Correct solution is this:
on file pins_GT2560_REV_A_PLUS.h remove dependency to ENABLED(BLTOUCH) and use simply
  # ifndef SERVO0_PIN
  #    define SERVO0_PIN 11 // original servo connector on A+ board
  # endif

on file pins_GT2560_REV_A.h
  # if !defined(SERVO0_PIN) && ENABLED(BLTOUCH)
  // HW modification needed 
  // see https://www.geeetech.com/forum/viewtopic.php?t=18713
  // see http://www.geeetech.com/blog/2016/09/how-to-use-3dtouch-auto-leveling-sensor-on-your-geeetech-prusa-i3/
  #    define SERVO0_PIN 32 // Z-max pin on board 
  # endif

This solution is more universal because allow to use old hack which is not needed now (if somebody want it). And this solution didn't break solution from factory.

Best regards, Jiri

Phr3d13, 2017-10-15:
I can make these changes and shoot another PR, but I don't have BLTouch on my machine to test... Can anyone confirm that this is what it should be (before I make another blind PR :) )?

dobryj, 2017-10-15:
I have BLtouch connected on "servo connector" on GT2560 A+
It is really pin 11 (arduino numbering). I can confirm it.

fiveangle, 2017-10-16:
@dobryj - Since you know intimatley about this, perhaps you can submit PR with the pin change that also includes a warning in SanityCheck.h stating for user to double-check pin usage ( if BLTOUCH is enabled for this specific board (due to the community consenseus problem you detailed above).

If you want to get even more user friendly, you can add a USE_SERVO_PIN_FOR_BLTOUCH option at the top of the board pins file, then put the redefine in a code block as you have later down in the pins.  Note, also #undef the pin before re-defining to prevent unecessary compile warning.

wittsan, 2017-12-11:
Hello,
I have a problem with the BLTouch and the last marlin.
_(But maybe it's a configuration problem for me)_
To make it work I have to reverse in pins_GT2560_REV_A_PLUS.h the numbers of the PINs

```
#include "pins_GT2560_REV_A.h"

#undef BOARD_NAME
#define BOARD_NAME "GT2560 Rev.A +"

#if ENABLED (BLTOUCH)
   #define SERVO0_PIN 11 // 32
#else
   #define SERVO0_PIN 32 // 11
#endif
```
It only works like that for me.
On the GT2560 A +, it is the PIN11 which must be activated instead of 32. I think there is a reversal.

And dobryj says the same thing:

> I have BLtouch connected on "servo connector" on GT2560 A+
>It is really pin 11 (arduino numbering). I can confirm it.

Thank you

FabLab-du-Don, 2017-12-21:
Il confirm : 
```
#include "pins_GT2560_REV_A.h"

#undef BOARD_NAME
#define BOARD_NAME "GT2560 Rev.A +"

#if ENABLED (BLTOUCH)
   #define SERVO0_PIN 11 
#else
   #define SERVO0_PIN 32 
#endif
```

thinkyhead, 2017-12-25:
Patched!

closed 2017-12-25 by thinkyhead

drackmoon, 2018-07-23:
I'm using this board on my UM clone so i did set the board to UM and i confirm that we must change the Servo Pin to 11 on UM Pins.

#define DEFAULT_MACHINE_NAME    "Ultimaker"
`// Servos
//
#define SERVO0_PIN       11    // untested`

mudmarron, 2018-12-15:
Hi guys,

I have a geeetech i3 printer. with a gt2560 rev a+ board. I have fitted the 3d touch. 

I have the 2 pin connector fitted to the z min stop, and the 3 pin fitted to the 3d touch part of the board.

I have checked with my multimeter and I think maybe this sensor is faulty.

When I turn the printer on it will do the test up and down but then nothing. When it goes to home the z axis it drives the extruder into the bed and none of the test commands (g code) respond to the sensor. 

I am completely new to this and if I can get some help that would be appreciated.