MarlinFirmware/Marlin #7982: Anet A8 first layer with ABL nozzle too high and Z corrected wrong

mamama1, 2017-10-14:
------------------------------------------------------------
### Bug Report
- Description: After Homing and ABL (Bilinear), print starts with the nozzle too high, although z-offset is set correctly. Also, when X gets higher, Z gets lifted, although the bed is tilted down a bit on the right side (so z should be lowered a bit, not lifted).
- Expected behaviour: dat damn ABL thing doing what it is supposed to do after 12 friggin hours of fiddling around.
- Actual behaviour: I have to set the z offset so low that the left side of the print is pressed way to hard to make the right side stick to the bed because of the wrong correction of ABL.
- Steps to reproduce:

What I've already done:
- measured standard deviation: ok
- increased probing points to 5x5
- corrected X and Y offsets (measured distance nozzle to probe)
- corrected Z offset multiple times to make friction between a thin sheet of paper and the nozzle
- updated to Marlin 1.1.6 (migrated my settings from my old config using a diff tool)
- turning off heated bed while probing
- printer is always warmed up
- calibrated extruder steps

sensor: LJ18A3-8-Z/BX using with an optocoupler circuit
mount: https://www.thingiverse.com/thing:1911146
#define X_PROBE_OFFSET_FROM_EXTRUDER 65 which is noted in the description of the mount is wrong, i have measured 71.

This is how I adjust Z offset:
M851 Z0
G28
G1 X110 Y110
M211 S0
Move the nozzle down so it is just gripping a piece of standard printer paper
Set the Z-offset to the displayed value. E.g. if the printer displays a Z-Value of -1.23 enter
M851 Z-1.23
M500
M211 S1

Config file:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1385082/Marlin.zip)

I have absolutely no idea what is going on here. Wasted huge amounts of time and filament already (this: [https://reprapworld.de/products/filament/real_filament/pla/real_pla_satin_sky_spool_of_0_75kg_1_75mm/](https://reprapworld.de/products/filament/real_filament/pla/real_pla_satin_sky_spool_of_0_75kg_1_75mm/))

Thanks for any advice.

AnHardt, 2017-10-15:
#7927

Roxy-3D, 2017-10-15:
> but desperately needs an, apart from this, correct geometry of the printer to get correct probes.

Yeah...  But I think I would argue...  In most cases it is best for the user to actually square the X & Y axis.   That is not that hard  to do.  And it drastically simplifies the firmware.

AnHardt, 2017-10-15:
A lot of possible geometric errors are avoided when the probe has no offset to the nozzle, or the nozzle is the probe. That was the big advantage of grid-leveling when it only worked by probing manually. Now we have UBL what can be hand-tuned to make perfect adhesion.

mamama1, 2017-10-15:
Hi

I also once probed manually and as far as I remember, i had the same results. but i'll try that again. when probing manually, i'd have to remove g28 and g29 from my slicers start g-code, right? because at g28 any abl information gets cleared.
how do i save my manually probed data to eeprom and how do i recall it so i don't need to level before each print? didn't find anything on this.

thanks!

AnHardt, 2017-10-15:
Too lazy to search and read about UBL?

benlye, 2017-10-15:
> Too lazy to search and read about UBL?

Harsh ;-) 

OP has an Anet A8 which has an Atmega1284p with too little flash space for UBL. UBL isn't the answer to every bed levelling problem.

In my experience, having been through it, the problem described is most likely down to printer or probe  geometry. The software works, whether it's ABL or UBL, and it's much more productive to assume that when there are problems, the hardware is at fault.

mamama1, 2017-10-15:
that's why i wanted to try the manual probing once again.
"too lazy" after 12 hours mocking around (once again)? definitely not. desperate and information-overloaded, more likely.

mamama1, 2017-10-15:
any advice on this? "when probing manually, i'd have to remove g28 and g29 from my slicers start g-code, right? because at g28 any abl information gets cleared.
how do i save my manually probed data to eeprom and how do i recall it so i don't need to level before each print? didn't find anything on this."

thanks

guiseco, 2017-10-15:
Move to LINEAR. Bilinear sucks!

mamama1, 2017-10-15:
already tried linear...

guiseco, 2017-10-15:
Anet A8 with ABL LINEAR

https://youtu.be/6kTknGnr9Ds

benlye, 2017-10-15:
> Move to LINEAR. Bilinear sucks!

I disagree, bilinear works perfectly for me on my A8. Has done since I spent quite a lot of effort getting everything true and square and the probe offsets perfectly dialled in. None of this stuff is magic, you have to set the basics up correctly first.

guiseco, 2017-10-15:
So, why linear works great and a lot of people have problem with bilinear ?

mamama1, 2017-10-15:
maybe because of:

> you have to set the basics up correctly first.

as soon as my printer works again (ptfe tube got jammed), I'll print a mount for the probe which is less far away from the nozzle, especially on the Y axis and then we'll see.

I'd like to change the shafts and rods with new ones but atm I also don't want to invest too much money into this thing as I have ordered a new Creality CR-10 a few weeks ago, which should arrive soon.

Everything else should be square and even. I've also printed various frame stabilizers and put quite an effort into making the printer well aligned.

benlye, 2017-10-15:
> So, why linear works great and a lot of people have problem with bilinear ?

Because bilinear is more sophisticated it is also more prone to being upset by hardware that is not correctly set up or calibrated.

I went through weeks of trying to work out why my printer wouldn't level to eventually figure out that I had a couple of different hardware issues (Y axis linear rods bowed and probe Y offset out by 1mm).  The probe offset being incorrect was definitely the main issue.  Now that the hardware is sorted, I get great first layers every time - all I have to do is use Z baby-stepping to dial in the perfect Z adjustment while it's printing the skirt, and away it goes.

My main advice is to double and triple check the probe-to-nozzle X and Y offset.  If you're using a probe mount you downloaded and printed, and the offsets were given to you by the designer, don't trust them - measure them yourself.

guiseco, 2017-10-15:
Thats not true. 

My printer is perfect aligned. Works amazing without (manual) probe and using LINEAR. With bilinear there is no way to work properly.

See this: https://github.com/MarlinFirmware/Marlin/issues/7278

mamama1, 2017-10-15:
> My main advice is to double and triple check the probe-to-nozzle X and Y offset.

that's interesting! everywhere in the guides it says, that this is not too important. a few millimeters tolerance make no difference, the various guides say.

> If you're using a probe mount you downloaded and printed, and the offsets were given to you by the designer, don't trust them - measure them yourself.

been there, done that. however, not ultra precise, as I was thinking that it is not necessary to get these values 100% correct.

i think, all my rods and shafts are bent. mainly, because quite often i had a really hard time getting my PLA prints off the blue tape. it is always either not sticking at all (first layer not sticking) or sticking so good, that I barely can remove it.

benlye, 2017-10-15:
Well, the most accurate you can set is to the nearest mm because the values don't allow decimals, but for sure it must be accurate.  Mine was out by only 1 or 2mm in the Y axis (I don't recall the exact number now) and it caused the familiar too close on one corner and too far on another problem.

@guiseco, I promise you that in my case, it's true - bilinear levelling works for me.  If it doesn't work for you, I seriously doubt it is a software problem.  If you look at the comments in #7278 there is at least one user who had the same issues, but resolved the problem by re-routing his wiring harness so that it wasn't tilting the print head.

I'll say it again, the most common cause of problems with levelling is the hardware.

mamama1, 2017-10-15:
> resolved the problem by re-routing his wiring harness so that it wasn't tilting the print head.

now THAT is one nasty cause.
Didn't check that yet.

guiseco, 2017-10-15:
Nice first layer with LINEAR: 
https://youtu.be/4eTScOfJzlc

benlye, 2017-10-15:
> Nice first layer with LINEAR

Can't argue with you - that's a nice layer - if it works I'd stick with it.  Watched your drone footage videos too - your S500 is stable!  I've been thinking about building an S550...

guiseco, 2017-10-15:
@benlye, s500 is nice but i love my mine talon! Amazing plane !!!

mamama1, 2018-01-03:
i guess this can be closed as that ABL setup doesn't exist anymore and I can not contribute any more testing to that issue...

closed 2018-01-03 by mamama1