MarlinFirmware/Marlin #7985: Printer stops heating while doing G28

dulfe, 2017-10-15:
**Description:**

When `G28` is executing, the printer stops heating the extruder and the bed.

**Expected behavior:**

Keep heating extruder and bed.

**Actual behavior:**

Extruder and bed _stops_ heating.

**Steps to reproduce:**

- Go to Z150 (somewhere height)
- Preheat bed and extruder (wait until they stabilize) 
- `G28`

While it is doing `G28`, you will notice the extruder and bed stops heating. (target temperature still showing the correct value, but actual temperature _decreases_)

After `G28` completes, temp raises.

I am using stable version 1.1.6 and I think it was also happening in 1.1.5 and it was working fine in 1.1.4. I haven't tested the older versions because I need to print some parts.

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1385129/Config.zip)

Thank you!

benlye, 2017-10-15:
You have `PROBING_HEATERS_OFF` enabled.

Find this line in Configuration.h and comment it out:
`#define PROBING_HEATERS_OFF       // Turn heaters off when probing`

dulfe, 2017-10-15:
@benlye Thank You!... I guess I missed that when I merged my configuration file with the new one. Sorry about that!

benlye, 2017-10-16:
@dulfe, glad I was able to help.  If you're all set, please close the issue.

closed 2018-02-09 by thinkyhead