MarlinFirmware/Marlin #7989: [branch bugfix 2.0.x] ### Bug Report: inverted mosfet config did not work

tximy, 2017-10-15:
[branch bugfix 2.0.x] ### Bug Report: inverted mosfet config did not work
I have an DUE with Geektech RAMPS_FD_V1 it has inverted mosfets
the pin configuration has this definitions but without any effect

#define INVERTED_HEATER_PINS
#define INVERTED_BED_PINS
#define INVERTED_FAN_PINS

and I try the full graphics reprapdiscount lcd without video

Thx.

psavva, 2017-10-16:
Please post your configuration files, it makes it easier to get help...

tximy, 2017-10-16:
Only change "#define MOTHERBOARD BOARD_RAMPS_FD_V1" at the default config and the heaters and the fans are enabled.

tximy, 2017-10-16:
The RepRapDiscountFullGraphics display work with one change at this file
[ultralcd_impl_DOGM.zip](https://github.com/MarlinFirmware/Marlin/files/1387113/ultralcd_impl_DOGM.zip)
// LCD selection
#if ENABLED(REPRAPWORLD_GRAPHICAL_LCD)
  U8GLIB_ST7920_128X64_4X u8g(LCD_PINS_RS); // 2 stripes
  // U8GLIB_ST7920_128X64 u8g(LCD_PINS_RS); // 8 stripes
#elif ENABLED(U8GLIB_ST7920)
  U8GLIB_ST7920_128X64_4X u8g(LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS); // Original u8glib device. 2 stripes **************
                                                                            // No 4 stripe device available from u8glib.
  //U8GLIB_ST7920_128X64_1X u8g(LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS);    // Original u8glib device. 8 stripes
//**********  U8GLIB_ST7920_128X64_RRD u8g(0); // Number of stripes can be adjusted in ultralcd_st7920_u8glib_rrd.h with PAGE_HEIGHT

thinkyhead, 2018-02-09:
@tximy — Is this change still required for `RAMPS_FD_V1` + `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`? Would it apply to all `U8GLIB_ST7920` or just this specific display?

tximy, 2018-02-09:
This display works nice with marlin 1.x.x and ramps 1.4, this fix is for marlin 2.0.x and Ramps_FD_V1

AnHardt, 2018-02-09:
Not `Ramps_FD_V1` is the problem, but the 32bit `DUE` below it.
The original u8glib driver should work for 8 and 32 bit, but is slooooow for at least 8bit.
So don't make it the general case. This has to be conditional.

thinkyhead, 2018-02-18:
> This has to be conditional.

At this time the code for `U8GLIB_ST7920` reads:
```cpp
#elif ENABLED(U8GLIB_ST7920)
  // RepRap Discount Full Graphics Smart Controller
  #if DISABLED(SDSUPPORT) && (LCD_PINS_D4 == SCK_PIN) && (LCD_PINS_ENABLE == MOSI_PIN)
    U8GLIB_ST7920_128X64_4X_HAL u8g(LCD_PINS_RS); // 2 stripes, HW SPI (shared with SD card, on AVR does not use standard LCD adapter)
  #else
    //U8GLIB_ST7920_128X64_4X u8g(LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS); // Original u8glib device. 2 stripes, SW SPI
    U8GLIB_ST7920_128X64_RRD u8g(LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS); // Number of stripes can be adjusted in ultralcd_st7920_u8glib_rrd.h with PAGE_HEIGHT
                                                                           // AVR version ignores these pin settings
                                                                           // HAL version uses these pin settings
  #endif
```
…so for a working display it sounds like this can be changed to…
```cpp
#elif ENABLED(U8GLIB_ST7920)
  // RepRap Discount Full Graphics Smart Controller
  #if DISABLED(SDSUPPORT) && LCD_PINS_D4 == SCK_PIN && LCD_PINS_ENABLE == MOSI_PIN
    U8GLIB_ST7920_128X64_4X_HAL u8g(LCD_PINS_RS); // 2 stripes, HW SPI (shared with SD card, on AVR does not use standard LCD adapter)
  #elif defined(__SAM3X8E__)
    U8GLIB_ST7920_128X64_4X u8g(LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS); // Original u8glib device. 2 stripes, SW SPI
  #else
    U8GLIB_ST7920_128X64_RRD u8g(LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS); // Number of stripes can be adjusted in ultralcd_st7920_u8glib_rrd.h with PAGE_HEIGHT
                                                                           // AVR version ignores these pin settings
                                                                           // HAL version uses these pin settings
  #endif
```

…assuming no other PR already addressed this.

thinkyhead, 2018-02-18:
In fact, the three options mentioned in the OP have not been implemented at all. Adding those to the PR linked above.

closed 2018-02-22 by thinkyhead