MarlinFirmware/Marlin #7990: 2.0 compile errors ' create process ' no such file eccc  cfg files included here

studiodyne, 2017-10-15:
Hi , is there someone to try to compile 2.0 with these three files please
My compiler stops and shows  avr-gcc: error: CreateProcess: No such file or directory

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1385794/Marlin.zip)
thks

bobc, 2017-10-15:
This is the same issue as #7931

fiveangle, 2017-10-15:
And search still works @studiodyne :grin: 

https://github.com/MarlinFirmware/Marlin/issues?utf8=%E2%9C%93&q=is%3Aissue%20is%3Aopen%20CreateProcess

https://github.com/MarlinFirmware/Marlin/issues?utf8=✓&q=is%3Aissue%20is%3Aclosed%20CreateProcess

studiodyne, 2017-10-16:
🔵🔵🔵🔵🔵🔵🔵🔵🔵🔵🔵🔵🔵🔵🔵🔵
Thank you very Much !!!
🔵🔵🔵🔵🔵🔵🔵🔵🔵🔵🔵🔵🔵🔵🔵🔵
@fiveangle :  sorry , it's my first interest for 2.0 and never look the topics about it , now , i will follow 😄

closed 2017-10-16 by studiodyne

TateA, 2020-02-03:
This did not work for me, I think it is just windows 10. Booting into Linux now, where shit just works. Screw you Microsoft! I give up trying to get windows to work.

FLYINGSC0T, 2020-02-28:
Hi. I have a similar issue where the compilation ends with an error. I'm using the Marlin Configurator app and the final output is as follows : 

Building in release mode

Linking .pio\build\sanguino_atmega1284p\firmware.elf

avr-g++: error: CreateProcess: No such file or directory

*** [.pio\build\sanguino_atmega1284p\firmware.elf] Error 1

========================= [FAILED] Took 16.16 seconds =========================

Can anyone help in solving this issue please.

Also, I notice that on the Compile menu dropdown to select the environment for compilation that it is not possible to navigate to the end of the list. Is this normal ?

Thanks in advance for any advice.

thinkyhead, 2020-02-28:
@FLYINGSC0T — The problem is your build environment. I suggest you use **PlatformIO** in **VSCode** with the **Auto Build Marlin** extension to do your builds. Anyway, we don't answer questions or do user support here. Please read:

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).