MarlinFirmware/Marlin #7994: Marlin compiled size issue, won't fit

jonathanlundstrom, 2017-10-16:
Hello,

I apologize for not using any of the provided templates but I didn't see them as necessary in this particular issue. I'm well traversed with Marlin and yesterday I was tinkering with a friend of mine installing the BLTouch probe on the CR-10. The result was great and we repurposed pin 27 on the CR-10 Melzi for the BLTouch servo control. The tricky part was software though, and not because of Marlin per se, but because if it's size. The Melzi has a very limited storage capacity, so with BLTouch, SD-Card, Full graphic LCD and Bilinear leveling it simply won't fit.

I spend a few hours trying to compress the firmware by disabling features such as bed thermal protection, cold extrusion prevention etc. but it wasn't enough, and to be frank, these features ***should not*** be disabled.

So, my question is the following; is there anything that I can remove or change in the firmware to free up space? As of writing moment I'm about 2216 bytes over the limit, but with security measures disabled (not preferable) I can get it down to ~944 bytes.

I've attached my configuration files so that you can have a look.

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/1386674/Configurations.zip)

Thanks,
Jonathan

jeffyl, 2017-10-16:
You haven't disabled arc support in your configuration_adv.h which should give you enough space. 

You can also compare with the sample files here https://github.com/MarlinFirmware/Marlin/tree/1.1.x/Marlin/example_configurations/Creality/CR-10

fiveangle, 2017-10-16:
Enable/disable every feature starting from the top until you find the magical combo you can live without, that’s pretty much it.

https://github.com/fiveangle/Marlin/blob/progmem_usage_comments/Marlin/Configuration.h
https://github.com/fiveangle/Marlin/blob/progmem_usage_comments/Marlin/Configuration_adv.h

fiveangle, 2017-10-16:
At some point it’s easier to buy a new controller… or ditch SD + LCD and just use Octoprint. That was my takeaway from the same experiment on Printrboard.

jonathanlundstrom, 2017-10-16:
I just disabled the arc support and it did indeed give me the additional free space that I needed. Now, what are the drawbacks of doing so?

fiveangle, 2017-10-16:
You won’t be able to run CNC toolpath outputs ;)

jonathanlundstrom, 2017-10-16:
Thanks guys, this is perfect, now it's time to try it out! :)

psavva, 2017-11-01:
Please cost this issue if all is well

closed 2018-02-09 by thinkyhead