MarlinFirmware/Marlin #7995: ABL probe points, Bed Size, problems

weekendwarrior1999, 2017-10-16:
I am having trouble setting this version of marlin up and getting ABL working correctly. This is what I have.

X axis maximum travel= 360mm
Bed width = 300mm
Y axis Maximum travel = 310
Bed length = 300

Problem:
I need to offset what would be 0,0 on XY as the bed is actually about +30mm from X axis Min endstop and about +5mm from Y axis Min endstop.

Without being able to set this (haven't found a variable so far) when I set the bed size, and Min/Max on Y/X coordinates for probing I run into an issue.

Cant probe close to the right side edge of the bed because my probe offset is -26 on X axis and the IDE complains the probing point is not reachable, cause I sat the point to something like 320mm and the only way to overcome the error is to make the probing point <=300mm which actually lands on the bed about 250mm. and if I increase the bed size to a larger size to accommodate the needed probing point the IDE then complains that the X axis travel distance is too small to accommodate the bed within that axis

If there was a variable (if there isn't  one I haven't found) for X and Y bed offset I think this would correct the issue.

benlye, 2017-10-16:
It sounds like you need to set:
```
#define X_MIN_POS -30
#define Y_MIN_POS -5
```
(both in Configuration.h)

weekendwarrior1999, 2017-10-16:
Ah, I was putting "30 and 5" as the values, the description in the comments made it sound like it was to simply create a "software min limit".  The designers may need to add a comment here so as to better describe it's function. Such as (Also functions as "X Min Limit offset from bed, Y Min Limit offset from bed".

GigaTech, 2017-10-16:
Hi Guys/Gals trying to do the same thing... so confusing (could be a lot easier)

X_BED_SIZE 230
Y_BED_SIZE 146

X_MAX_POS 287
Y_MAX_POS 149
Z_MAX_POS 130

X_BED_OFFSET 20
Y_BED_OFFSET 0

so with this logic (and my logic) am I correct?

**ADD** the offset to the ?_MIN_POS **_and_** ?_MAX_POS values

#define X_BED_SIZE 230
#define Y_BED_SIZE 146

#define X_MIN_POS -20
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 267
#define Y_MAX_POS 149
#define Z_MAX_POS 130

Many Thanks

thinkyhead, 2018-02-09:
There's no "bed offset." Your nozzle in simply able to move left of where 0 is.

closed 2018-02-09 by thinkyhead