MarlinFirmware/Marlin #7997: This is a bug - can't kill print when starting

ruggb, 2017-10-16:
------------------------------------------------------------
### Bug Report
- Description: ---If I start a print, I cannot kill it immediately.
- Expected behaviour: ---whenever I kill a print it should kill it immediately.
- Actual behaviour: ---if I kill a print b4 it reaches temp and starts, it continues heating until it reaches temp, then kills
- Steps to reproduce:
  - Do this -- load and start a print,
  - Do that --  as it is heating kill it.

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1388574/Marlin.zip)

------------------------------------------------------------
1.1.4? bug fix #10d72f3

FHeilmann, 2017-10-19:
Enable the EMERGENCY_PARSER in Configuration_adv.h

that should allow you to send M112 at any point to kill the printer or, at least, make it execute faster. Alternatively, using the reset button on your control board or your LCD will accomplish an immediate kill as well.

AnHardt, 2017-10-19:
Via kill-button or via M112?

ruggb, 2017-10-19:
@AnHardt I hit the kill button and also tried M109 s0. I don't know what Repetier sends as a kill sequence. Instructions indicate the script I entered is additional code - which does not have M108, M112, M410 commands. I don't know what M410 is as it is not defined in the riprap gcode listing.

Subsequently, as shown below, I tried M108, and M112 - no change with or without emergency parser enabled.

@FHeilmann why would you want to inhibit those commands at all? Then have a function to uninhhibit them? just strange logic. 
Reading further, I have no idea what the verbage after that EMERGENCY PARSER is trying to say. Another issue I quess...
----------------
// Some other clients start sending commands while receiving a 'wait'.
// This "wait" is only sent when the buffer is empty. 1 second is a good value here.
//#define NO_TIMEOUTS 1000 // Milliseconds

// Some clients will have this feature soon. This could make the NO_TIMEOUTS unnecessary.
//#define ADVANCED_OK
--------------
why do clients not respect a 'wait'?
why is a 'wait' sent if the buffer is empty? - AND which "**this** 'wait'" is sent - the NO_TIMEOUTS????????
what does NO_TIMEOUTS actually do?
what does ADVANCED_OK do and why would it make NO_TIMEOUTS unnecessary?

I'm an engineer and don't understand - I can't imagine a 'normal' user would either.

**BOTTOM LINE** - this issue - as configured default it responds to an emergency stop - which apparently is a reset. With emergency parser uncommented, it still doesn't respond to a kill from Repetier. Here is the Repetier log with my comments:

Printer connected, file loaded and sliced
Start print was clicked

09:57:24.035 : N16 G28*36
09:57:24.035 : N17 M117 ETE 41m 29s*118
09:57:24.036 : N18 G0 Z10 F9000*4
09:57:24.038 : N19 M107*29
09:57:24.039 : N20 M104 S205*80
09:57:24.039 : N21 G91*34
09:57:24.040 : N22 G1 Z10 F9000*12
09:57:27.882 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:0 B:0 Z:0
09:57:27.882 : N23 G90*33
09:57:27.882 : N24 G28*37
09:57:27.886 : N25 G1 Z10 F9000*11
09:57:27.888 : N26 M109 S205*91
09:57:27.888 : N27 G21*47
09:57:27.888 : N28 G90*42
09:57:27.888 : N29 M82*34
09:57:27.890 : N30 G92 E0*116
09:57:27.891 : N31 G1 E-4 F2400*6
09:57:34.577 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:0 B:0 Z:0
09:57:34.594 : N32 M117 ETE 41m 28s*112

hotend heating and waiting......
note: maybe something is sent but it doesn't show in the log unless marlin responds?? I don't know.

kill was clicked - nothing
M112 was sent manually - nothing
M108 sent manually - nothing
heating icon clicked and off icon appeared but nothing happened and it kept heating
emergency stop was clicked -- 

09:58:16.947 : Printer reset detected - initalizing
09:58:16.947 : start
09:58:16.947 : echo:Marlin 1.1.4BF
09:58:16.949 : echo: Last Updated: 10-16-17 | Author: Bill Ruggirello
09:58:16.950 : echo:Compiled: Oct 19 2017
09:58:16.952 : echo: Free Memory: 3575  PlannerBufferBytes: 1232
09:58:16.955 : echo:EEPROM version mismatch (EEPROM=V40 Marlin=V42)
09:58:16.956 : echo:Hardcoded Default Settings Loaded
09:58:16.957 : echo:  G21    ; Units in mm
09:58:16.959 : echo:  M149 C ; Units in Celsius
09:58:16.960 : echo:Filament settings: Disabled
09:58:16.961 : echo:  M200 D1.75
09:58:16.961 : echo:  M200 D0
09:58:16.963 : echo:Steps per unit:
09:58:16.964 : echo:  M92 X200.00 Y200.00 Z800.00 E185.20
09:58:16.965 : echo:Maximum feedrates (units/s):
09:58:16.969 : echo:  M203 X500.00 Y500.00 Z8.00 E25.00
09:58:16.971 : echo:Maximum Acceleration (units/s2):
09:58:16.972 : echo:  M201 X9000 Y9000 Z100 E12000
09:58:16.975 : echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
09:58:16.978 : echo:  M204 P5000.00 R12000.00 T9000.00
09:58:16.981 : echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
09:58:16.984 : echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
09:58:16.985 : echo:Home offset:
09:58:16.987 : echo:  M206 X0.00 Y0.00 Z0.00
09:58:16.988 : echo:Material heatup parameters:
09:58:16.989 : echo:  M145 S0 H180 B70 F0
09:58:16.989 : M145 S1 H240 B110 F0
09:58:16.991 : echo:PID settings:
09:58:16.992 : echo:  M301 P18.43 I1.21 D70.22
09:58:17.157 : N1 M110*34
09:58:17.157 : N2 M115*36
09:58:17.157 : N3 M105*36
09:58:17.157 : N4 M114*35
09:58:17.183 : N5 M111 S6*98
09:58:17.186 : N6 T0*60
09:58:17.186 : N7 M20*22
09:58:17.187 : N8 M80*19
09:58:17.187 : N9 M105*46
09:58:20.221 : N10 M105*22
09:58:21.567 : FIRMWARE_NAME:Marlin 10d72f3 SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:CC_BOT EXTRUDER_COUNT:1 UUID:277494b7-b7c8-4eb1-a653-c2e695c24911
09:58:21.567 : N11 M220 S100*81
09:58:21.568 : Cap:EEPROM:1
09:58:21.568 : Cap:AUTOREPORT_TEMP:1
09:58:21.570 : Cap:PROGRESS:0
09:58:21.570 : Cap:PRINT_JOB:1
09:58:21.571 : Cap:AUTOLEVEL:0
09:58:21.571 : Cap:Z_PROBE:0
09:58:21.572 : Cap:LEVELING_DATA:0
09:58:21.573 : Cap:BUILD_PERCENT:0
09:58:21.573 : Cap:SOFTWARE_POWER:0
09:58:21.575 : Cap:TOGGLE_LIGHTS:0
09:58:21.576 : Cap:CASE_LIGHT_BRIGHTNESS:0
09:58:21.576 : Cap:EMERGENCY_PARSER:1
09:58:21.604 : N12 M221 S100*83
09:58:21.615 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:0 B:0 Z:0
09:58:21.615 : N13 M111 S6*85
09:58:21.632 : echo:DEBUG:INFO,ERRORS
09:58:21.632 : N14 T0*15
09:58:21.633 : echo:Active Extruder: 0
09:58:21.634 : N15 M155 S1*84
09:58:21.635 : Begin file list
09:58:21.635 : End file list
09:58:21.646 : echo:DEBUG:INFO,ERRORS
09:58:21.646 : echo:Active Extruder: 0
09:58:24.101 : echo:SD init fail

AnHardt, 2017-10-21:
@ruggb 
So you hit the stop button in the hast, not at the lcd-panel.

When we developed M108 we had expended and detailed discussions under what circumstances this and M112 can work at all. If you want search for that.

Short answer:
It can not work, except the host supports it.

A bit more detailed answer:
The normally used communication between the 3D-printer client (here Marlin) and the host is called ping-pong-protocol. The host is sending a g-code command and waits for an 'ok' from the client. When that was received the next command is sent. That warrants no buffer overflow can occur. In case of long lasting commands like heat_and_wait for their duration no further command can be sent without violating this protocol by will.

Even more details in https://github.com/MarlinFirmware/Marlin/issues/7076#issuecomment-337373889

ruggb, 2017-10-21:
The only way one can cancel a started but not printing yet print (hotend not at temp) is to do an emergency stop. Repetier has an issue with this because if you do a kill, nothing happens and if you turn the heater off it appears like it went off (off icon appears) but it really does not and Repetier sets the display based on what you tell it and not what it accomplishes. The only way you would know is to touch the hotend (ouch!) or notice that the temp is still rising. There is no info on this "feature" and I only stumbled upon it. Not a good user interface. Somehow Repetier must figure out that the kill didn't or isn't going to function NOW and do an auto emergency reset or Marlin needs to be able to receive a kill command while it is heating. I don't know that this can be done by either side independently, but maybe you do or can configure something to inform Host. I'm sure other hosts have the same issue.
I have not tried it with the hotbed but I am sure the situation is similar. This should be considered a safety issue.

AnHardt, 2017-10-21:
With the 'emergency command parser' we did our part. We can receive, evaluate and execute M108, M112, M410 at every time, even when all buffers are full - and try to inform the hosts via the EXTENDED_CAPABILITIES_REPORT about this possibility.
Now the hosts have to do their homework.

ruggb, 2017-10-21:
I passed that on to Repetier. Maybe something will come of it.

closed 2018-02-09 by thinkyhead

AnHardt, 2018-02-09:
The "Emergency Stop" button in RH is configurable.
If set to "Send emergency command and reconnect" or "Send emergency command + DTR high->low" most boards (at least MEGAs) will do a reset - except this is disabled by cutting a jumper.