MarlinFirmware/Marlin #7998: oversight - re Babystep_XY

ruggb, 2017-10-16:
The good thing is you commented out babystep_xy - it is bad for my CoreXY
The bad thing is you didn't change the associated note to instruct ppl what to do with it.
Not that I care but someone might. It is no longer ALSO ENABLEing
  //#define BABYSTEP_XY              // Also enable X/Y Babystepping. Not supported on DELTA!

maybe "//do not enable for CoreXY, not supported on DELTA!"

FHeilmann, 2017-10-19:
The comment seems fine to me, when the line is uncommented it "also enables babystepping for XY" which is exactly what the comment suggests. The comment is supposed to tell the user what the line does if uncommented, not in which state it is right now.

ruggb, 2017-10-19:
interestingly, it is the same comment that was there when it was not disabled. Maybe removing ALSO would help. ALSO implies that it is doing it - which now it is not.

FHeilmann, 2017-10-19:
As I said, the comment is supposed to descirbe what the line does WHEN UNCOMMENTED, not what it does in its current state (which may be commented or uncommented)

The comment indicates, "if this line is uncommented, babystepping is also enabled for XY". So if you look at your config, the line is commented, hence no XY babystep. I understand that this is confusing to you, but seing as this issue doesn't crop up constantly, other users interpret it the way I described it.

ruggb, 2017-10-19:
this was just commented out recently and the note is the same as it was. **Previously**, others may have interpreted it properly. As a note - NO ONE there (developers) knew that this was the reason babystepping did not work for CoreXY. A user had to figure it out and I stumbled upon his note and told you all. It took you 5 issues to finally believe it did work if you commented this out. You finally changed it but still don't tell users not to enable it for CoreXY. It is all about communicating.

closed 2018-02-09 by thinkyhead