MarlinFirmware/Marlin #8000: [FR] Configurable long-press delay

dmsquirrel, 2017-10-16:
Anet A8 2017 - Marlin 1.1.6 Release, standard ANETV1.0 board

Menu is difficult to use, buttons seem to "double press" if you click the button for down more than a split second. Not so fun when trying to do manual z homing <crash>.

Was previously working flawlessly with Skynet 2.3.2, but this had no hold down of button functionality.

Happy to mod code if someone points me in right direction (dev)

tcm0116, 2017-10-31:
Are you using an encoder with a single button or a 5 button setup?

dmsquirrel, 2017-11-02:
5 button setup, however board failed shortly after due to a bad firmware flash (serial port dc'd mid flash) so would be good if another 5 button a8 user can verify this 'quirk'. Board has been ISP reflashed, so will get latest marlin back on to verify if this is still present.

closed 2018-02-09 by thinkyhead