MarlinFirmware/Marlin #8008: 2.0.x - U8G vs. U8G2

Bob-the-Kuhn, 2017-10-17:
I've gotten U8G2 functional enough on Marlin 2.0.x that I could do some performance testing.  Here are the [details](https://www.dropbox.com/s/y1rgkeu1d8v8mj2/U8G2%20vs%20U8G%20rev%201.xlsx?dl=0).

U8G2 offers three types of drivers.  They trade off speed vs. RAM usage.  All three are in the table.

I only looked at the time from the start of the display update code to the end of the last bit being transmitted.  

For the LPC1768 the CPU load is about 1% for both.  I don't see this as a factor in deciding if we want to convert to U8G2.

For the mega2560 , U8G2 takes about 300 bytes more RAM and 4K more FLASH than U8G for the same performance.  

U8G2's big advantage for 2.0.x is that adding new CPUs is fairly straight forward and does NOT require changes to the U8G2 library.

----

I've got some problems to fix before U8G2 is ready for limited testing.  The biggest problems are:
- Rotating the display by 180 degrees screws up the text.
- The bit maps are not displaying properly.

AnHardt, 2017-10-18:
So let's have a look into the numbers,

Some assumptions:
There is no stepper running.
We are on the info-screen.
"total update time" is about the time from ".firstPage();" to the last ".nextPage()" -> rendering + transmission of a full screen.
"start to first XMIT" is about the time from ".firstPage();" to the first bit transfer -> rendering of the first "stripe or page"
"bit speed" is the frequency of clk-pulses when a byte is transferred.
"update interval" is the time between two ".firstPage();" -> time until the next complete screen update begins - when no change of a value, keypress, ... it is forcing.
You don't trust in the number in the yellow field.

Questions:
In what tests ROT180 was on or off. Im my experience with u8glib this is a major time killer during rendering. (The reason for not recommending VIKI2 as a good Marlin display. The new device may have improved that - but i don't think so - only the transfer should be affected)

Under the further assumption ROT180 was used everywhere.
What do i see?
u8g runs considerably faster when compiled for Arduino. That's unexpected. Rendering as well transfer is affected - so its not only the different device. The higher spi-clock tells a story about lost time - maybe in unnecessary waits.
The clipping-macros in u8g² do work as expected. Else for all tree devise the first rendering would take the same time.
3-4MHz spi-clock - but pixel errors. Try to slow it down. From my experience with the ST7920 i know this effect. Are ST7565_DELAYs needed? That would be a clear sign for being to fast now.
Different update intervals for different numbers of stripes do make me nervous. Should always be a second. Likely a minor Marlin bug.
If the 4MHz spi clock is correct i want to have VIKI2s for my ReARM (and the DUEs and ...). My ST7920s are at their limits with the 16MHz AVRs
Comparing total update times of devices with the same amount of stripes is surprising. While for the AVRs the time is nearly constant, the difference for the rearm is nearly 100%. But that could be the problem with the failed measurement.
Codesize should not be a problem. If using a graphical display you already have to make too many hurting compromises with the smale prozessors. Up to now on the 2560s is plenty of space. Probably also no problem with the 32biters.

Writing your own devices without altering the library is also possible with u8glib. We already have two of them

Conclusion:
If you can fix the problems i have no complains against switching to u8g²lib.
Switching to u8g²s UTF8 handling can probably made later - and it's much easier when you already have Marlin running on the right graphic system. As far i can see the impact to the char-based displays will not be very high. In case the u8g²fonts can only hold 255 symbols i imagine an utf-mapper not picking symbols but switching fonts with the needed code pages.

thinkyhead, 2018-02-09:
A whole new display and menu system would be great. Something built outside of Marlin and then imported into it would be the best way to go, in my humble.

closed 2018-05-20 by Bob-the-Kuhn