MarlinFirmware/Marlin #8010: 2.0 Does not compile with MAX31855

Grogyan, 2017-10-17:
### Compile Error
When I compile with `MAX31855` I get an error:
```
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:939:4: error: #error "MAX6675_SS (required for TEMP_SENSOR_0) not defined for this board.
"
#error "MAX6675_SS (required for TEMP_SENSOR_0) not defined for this board."
```

------------------------------------------------------------
### Bug Report
- Description: --- 2.0 bugfix does not compile with MAX31855
- Expected behaviour: --- compile with MAX31855
- Actual behaviour: --- Not compiling with said feature
- Board:  Re-Arm EFB
------------------------------------------------------------

Grogyan, 2017-10-23:
Tried again over the weekend with a fresh copy of both PlatformIO and Marlin 2.0 bugfix

Still cannot compile

Grogyan, 2017-12-30:
Tried this again with the latest bugfix, and still unable to compile.
Found that MAX6675_SS was not declared in the pins_RAMPS_RE_ARM.h

However that only complicated things further because the Re-ARM doesn't have SPDR, SPSR and SPIF registers like the ATMEGA 2560 does.  And so compiling fails on that in private_spi.h

thinkyhead, 2018-02-11:
Trying to test with `BOARD_RAMPS_14_RE_ARM_EFB` and `TEMP_SENSOR_0 -3` but can't get it to compile (with Deviot in Sublime Text). The compiler can't find `debug_frmwrk.h`, `lpc17xx_i2c.h`, or `lpc17xx_pinsel.h`.

Grogyan, 2018-12-05:
Have tried again with the latest 2.0 bugfix
nearly 12 months later it still does not compile for MAX31855

A bit of messing around, seeing what works

In spi_pins.h
added to the bottom #define MAX6675_SS_PIN P1_28 (using the MAX Z Endstop, as an example CS pin)

in private_spi.h, modified the lines that call the SPSR and SPDR to
`
    #if HAL_AVR
      FORCE_INLINE static uint8_t receive() {
        SPDR = 0;
        for (;!TEST(SPSR, SPIF););
      return SPDR;
    }
    
    #else
    FORCE_INLINE static uint8_t receive() {}
    #endif
`

This presents the error
`
Marlin\src\module\../libs/softspi.h:50:14: error: 'g_APinDescription' was not declared in this scope
`

However the pin was set in spi_pins.h

I don't know how to get this to compile and work

boelle, 2019-02-19:
@Grogyan Please provide your `Configuration.h` and `Configuration_adv.h` in a ZIP file for compiling and testing.

Grogyan, 2019-02-19:
@boelle not necessary, as it is just two lines changed from the default, to replicate

Board
MOTHERBOARD BOARD_RAMPS_14_RE_ARM_EFB

Temp Sensor
TEMP_SENSOR_0 -3

Grogyan, 2019-02-19:
I'll add the changed files later, next week probably.

boelle, 2019-02-19:
Awesome

Grogyan, 2019-02-20:
I just pulled down the latest version,

Had to still add to the bottom of the `pins_RAMPS_RE_ARM.h`
`#ifndef MAX6675_SS_PIN 
  #define MAX6675_SS_PIN P1_28
#endif`

It compiled.

Highly suspect this has been inadvertently fixed by #8686
However I have no time to actually test it until next week

closed 2019-02-20 by thinkyhead