MarlinFirmware/Marlin #8013: T parameter not working without argument

LVD-AC, 2017-10-18:
`parser.seen('T')`

seems no longer to be working when no argument is passed (I guess since the introduction of the T-codes)

e.g. GXX T is not recognized while GXX T0 and GXX T123 are.

Roxy-3D, 2017-10-18:
I was just using this yesterday and it worked as expected.    But I have #define FASTER_GCODE_PARSER turned on.   Do you have that disabled?

LVD-AC, 2017-10-18:
FASTER_GCODE_PARSER turned off/on makes no difference.

I retested and the problem disappeared largely. Now it is the V parameter that is sometimes not recognized (if put after T without arguments).

e.g: 

e.g.
•GXX T is recognized now
but
•G33 P2 **T V**0: P2 and T are recognized V0 is not
•G33 **T V**0 P2: same thing

•G33 P2 V0 T works fine
•G33 P2 T0 V0 works fine
•G33 P2 **T V**0 C0.05 works fine
•G33 V0 T P2 works fine
•G33 **T V**0 C0.05 P2 works fine

 It seems a bit 'at random' depending of the order of the parameters whether it works or not.

Roxy-3D, 2017-10-18:
I did make a change to my branch in one place of the UBL-G29 code to used the parsed value for map type.   I don't think this is what you are seeing, but it should insulate us from the parser.
```cpp
 g29_map_type = parser.intval('T');
```
Right now, the change is only at:  https://github.com/Roxy-3D/SD-Reselect-and-Print-v5   but it should be merged soon.

LVD-AC, 2017-10-18:
Seems the trouble comes from (parser.h):
`  FORCE_INLINE static bool     boolval(const char c, const bool dval=false)   { return seen(c)    ? value_bool()         : dval; }
`
when using this without an argument, it gives undefined results.
I suggest to change this to:
`
  FORCE_INLINE static bool     boolval(const char c)  { return seenval(c) ? value_bool()         : seen(c); }`

LVD-AC, 2017-10-18:
See #8016

closed 2017-10-18 by LVD-AC

Roxy-3D, 2017-10-18:
Yes...   It is already merged.    I just checked that it was in place because I did a new fork and if it wasn't in place...  I was going to add it to my fork.  (That way, it would get merged in the next day or two.)