MarlinFirmware/Marlin #8018: Autobed ​​level inverted correction on the X and Y axes

greg60greg, 2017-10-18:
Hello, sorry for English translator.
I have a problem with the autobed ​​level. The correction is applied upside down on the X and Y axes.
The correction taken on this site works:
https://plus.google.com/+MateuszPerlak/posts/XNpAhkCKPCE

    vector_3 pt1 = vector_3 (ABL_PROBE_PT_1_X, ABL_PROBE_PT_1_Y, -z_at_pt_1);
    vector_3 pt2 = vector_3 (ABL_PROBE_PT_2_X, ABL_PROBE_PT_2_Y, -z_at_pt_2);
    vector_3 pt3 = vector_3 (ABL_PROBE_PT_3_X, ABL_PROBE_PT_3_Y, -z_at_pt_3);

On the other hand, it only works on the three point system.
I am forced to comment AUTO_BED_LEVELING_GRID. But how does it work with the grid has nine points?

Thanks for telling me the fix this firmware is a perfect copy of mine but with unadjusted values. Just for my "test".

Thank you in advance.
[firmware.zip](https://github.com/MarlinFirmware/Marlin/files/1395985/firmware.zip)

Roxy-3D, 2017-10-20:
> The correction is applied upside down on the X and Y axes.

You may not have your axis in the 'correct' direction.   You can try inverting the direction on X or Y and see if the bed leveling does the 'right' thing.

greg60greg, 2017-10-20:
The machine is fully finished. I spent a lot of time doing it (one and a half years). I do not want to edit anything except the firmware.

If I invert my X and Y I should reverse the direction of the engines if I do not touch the endstop what would be the same.

I am currently looking at the code with my father and am looking for a solution.

Many people have already had the trouble but I found no answer apart for the debug of the three probing point.

AnHardt, 2017-10-20:
Please look up "right handed coordinate system" and "left handed coordinate system".
To make work any kind of ABL you have to have a right handed. Lots of beginners are confused by this - especially when the bed is moving up and down instead of the nozzle.

greg60greg, 2017-10-20:
Or can I see this parameter?

AnHardt, 2017-10-20:
No. There is no parameter.
You have to reverse an axis - and logically your max-endstop (on this axis) becomes a min and the other way around.

An other symptom of a left handed coordinate system is getting mirrored parts.

In our example configurations the 3 points are always defined counter clockwise. That can't be so bad.

```
  // 3 arbitrary points to probe.
  // A simple cross-product is used to estimate the plane of the bed.
  #define ABL_PROBE_PT_1_X 15
  #define ABL_PROBE_PT_1_Y 180
  #define ABL_PROBE_PT_2_X 15
  #define ABL_PROBE_PT_2_Y 20
  #define ABL_PROBE_PT_3_X 170
  #define ABL_PROBE_PT_3_Y 20
```

AnHardt, 2017-10-20:
When you are talking about `AUTO_BED_LEVELING_GRID` you are not using one of our recent versions. From where did you get it? When did you get it? What version?

greg60greg, 2017-10-20:
I had mirror parts but following a reversal of X problem solved.

So if I do G29 with:
  // 3 arbitrary points to probe.
  // A simple cross-product is used to estimate the plane of the bed.
  #define ABL_PROBE_PT_1_X 15
  #define ABL_PROBE_PT_1_Y 180
  #define ABL_PROBE_PT_2_X 15
  #define ABL_PROBE_PT_2_Y 20
  #define ABL_PROBE_PT_3_X 170
  #define ABL_PROBE_PT_3_Y 20

I would have clockwise? And if it is not the case I have to reverse as I quoted previously?

I attach the firmware sorry for the empty file and thank you again for your help.

For the latest firmware version normally.

greg60greg, 2017-10-22:
Thank you the problem was the coordinate system.
I am trying to install the 1.1.6 marlin now.

Thanks again to all.

ripp2003, 2017-10-29:
Hi, I've got the same problem : compensation of the bed is inverted.
I've got an Anet A8 with Anet board inside : I'm running 
*  Marlin 1.1.6 
*  3 point ABL

This printer is a Right Handed so, counterwise.....
Can you tell me how did you solve your problem because, I know it's relative to coordinate system but really don't know how to solve this...

Many thanks

Roxy-3D, 2017-10-29:
> Can you tell me how did you solve your problem because, I know it's relative to coordinate system but really don't know how to solve this...

You can invert the direction of travel of either the X or Y axis.   By doing that, you will change that axis endstop from Max to Min or vice versa.

greg60greg, 2017-10-29:
Hello,
 I still have my worries. I'm trying to reduce my offsets. I reverse the X and Y and the minimum and maximum.
I printed with the reversed correction on the X and Y. I now have the impression that the Y is compensated upside down.
If I ever find the solution I will come and post it but I start to hope.

ripp2003, 2017-10-29:
oK, thanks for the quick reply, I'm trying to invert my Y axe, and will post my result ;-)

Thanks to Roxy and Greg !!!

ripp2003, 2017-10-29:
Here the variable I've modified to invert Y axe wich seems to be inverted:
->  Commented define USE_YMIN_PLUG
->  Commented out define USE_YMAX_PLUG
->  Y_MIN_ENDSTOP_INVERTING from true to FALSE
->  Y_MAX_ENDSTOP_INVERTING from false to TRUE
->  Y_HOME_DIR from -1 to 1

And now, Homing (G28) only center X Axe 
M114 tell me 
*  X = 45 (with offset probe (65) , it's 110 so OK)
*  Y = 155 (with offset probe (-45), it need to be 110 but is not.....

So I've modified the Y offset of nozzle probe  from -45 to 45 so from a negative value to a positve one : 

Now, Probing seems to works well (clockwise : when front of me, LEFT-DOWN then LEFT UP then RIGHT up then print) but, problem still persist, overcompensate at left and not really....

greg60greg, 2017-10-29:
Try to reverse X and Y:
#define X_STEP_PIN 54
#define X_DIR_PIN 55
#define X_ENABLE_PIN 38
#define X_CS_PIN 53

#define Y_STEP_PIN 60
#define Y_DIR_PIN 61
#define Y_ENABLE_PIN 56
#define Y_CS_PIN 49

In :
#define Y_STEP_PIN 54
#define Y_DIR_PIN 55
#define Y_ENABLE_PIN 38
#define Y_CS_PIN 53

#define X_STEP_PIN 60
#define X_DIR_PIN 61
#define X_ENABLE_PIN 56
#define X_CS_PIN 49

and the pins for the endstop reverse X and Y. I had some improvement in doing that.

ripp2003, 2017-10-29:
so with this mod, you use the X stepper driver for Y and the Y for X ? very strange, but I' will test

ripp2003, 2017-10-30:
Hi, for now, I'm waiting for a ramps board so, I 'will make some test to see if, for me, it's an anet board bug

greg60greg, 2017-10-30:
hello, yes that's right for X and Y. Let us know about the map but I do not think it will change. for my part I am waiting for a sensor that I would bring the maximum of the nozzle and I would try all the positions

ripp2003, 2017-10-31:
I'll test this night but, what slicer do you use ? (most important question is height of first layer)
I use Simplify and 1st layer is set to 90% so very low, maybe this could be the problem.....
I compare with same model in Cura 15 and, first layer start at 0.200

greg60greg, 2017-10-31:
I'm using Slic3R and simplify3D. My first layer is 0.3mm.

greg60greg, 2017-11-03:
I resumed my tests. I brought the sensor closer to the offsets and put a sensor with a smaller one.
The problem is small but is still existent.
Before I varied from 0mm to 0.6mm of the plateau and now from 0.20 to 0.40 or so.

A question ripp2003 how much do you have of extruder informed in configuration. I wonder if I have two extruders if the autobed ​​does not average between the two or something else.

Still sorry for my English and I hope to find a solution.

ripp2003, 2017-11-03:
Hi, sorry for the late reply but, I have received my ramps 1.4 3 days ago and just finished the assembly.... god this is a nightmare loooooool. 
For now, I've tested directly UBL and, I'm printing a third model successfully (no inverted no problem)

I'm printing models to fix the ramp on my anet and then, I' will try to go back to 3 points BL to see if everyhting's OK .

In my config, I've just one extruder, no more

ripp2003, 2017-11-10:
I've try the ABL linear and for me,  no more problem.
I suspect a bug on my anet board or, misconfiguration (but really don't know what I was doing wrong) but now, with Ramps, I don't have any problem.

For now, I stay with UBL and BabyStep .

closed 2018-02-09 by thinkyhead