MarlinFirmware/Marlin #8019: BABYSTEP_ZPROBE_GFX_OVERLAY will not compile.

Tannoo, 2017-10-18:
### Compile Error
Version: bugfix-2.0.x
This was also mentioned in #7797.
When I compile with `BABYSTEP_ZPROBE_GFX_OVERLAY` I get errors:
```
Marlin\src\lcd\ultralcd.cpp: In function 'void _lcd_babystep_zoffset_overlay(float)':
Marlin\src\lcd\ultralcd.cpp:1073:45: error: 'cw_bmp' was not declared in this scope
const unsigned char *rot_up   = cw_bmp;
^
Marlin\src\lcd\ultralcd.cpp:1074:45: error: 'ccw_bmp' was not declared in this scope
const unsigned char *rot_down = ccw_bmp;
^
Marlin\src\lcd\ultralcd.cpp:1084:82: error: 'nozzle_bmp' was not declared in this scope
if (PAGE_CONTAINS(3, 16))  u8g.drawBitmapP(nozzle + 6, 4 - dir, 2, 12, nozzle_bmp);
^
Marlin\src\lcd\ultralcd.cpp:1085:76: error: 'offset_bedline_bmp' was not declared in this scope
if (PAGE_CONTAINS(20, 20)) u8g.drawBitmapP(nozzle + 0, 20, 3, 1, offset_bedline_bmp);
^
Marlin\src\lcd\ultralcd.cpp:1091:58: error: 'up_arrow_bmp' was not declared in this scope
u8g.drawBitmapP(right + 20, 48 - dir, 2, 13, up_arrow_bmp);
^
Marlin\src\lcd\ultralcd.cpp:1092:58: error: 'down_arrow_bmp' was not declared in this scope
u8g.drawBitmapP(left  + 20, 49 - dir, 2, 13, down_arrow_bmp);
^
```

------------------------------------------------------------
### Bug Report
- Description: Will not compile with `BABYSTEP_ZPROBE_GFX_OVERLAY` enabled.
- Expected behaviour: Compile with `BABYSTEP_ZPROBE_GFX_OVERLAY` enabled.
- Actual behaviour: Does not compile with `BABYSTEP_ZPROBE_GFX_OVERLAY` enabled.
- Steps to reproduce:
  - Enable `BABYSTEP_ZPROBE_GFX_OVERLAY`
  - Try to compile

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
[Uploading Configuration_Files.zip…]()

cdedwards, 2017-10-18:
in dogm_bitmaps.h

move the the following  line above the #if defined(BABYSTEP_ZPROBE......

#endif // HAS_TEMP_BED

it's the very last line in the file.

Tannoo, 2017-10-18:
Really? Wow...

closed 2017-10-18 by Tannoo

Tannoo, 2017-10-19:
Why hasn't this been corrected yet?

Roxy-3D, 2017-10-20:
Maybe because people work on things that are of personal interest to them...

cdedwards, 2017-10-20:
I submitted a patch as part of a larger part. I had added GFX OVERLY to the ubl mesh adjust screen's. I guess it was not accepted by thinkyhead due to it being closed out with no comments.

Roxy-3D, 2017-10-20:
> I had added GFX OVERLY to the ubl mesh adjust screen's.

Were there some issues that never got resolved?   It would be good to have the Overlay Option on UBL Mesh Editing screens.

cdedwards, 2017-10-20:
I think it was because I was submitting both mario-io and my patch adding it to the UBL mesh edit. a quick note would have clarified this however.

cdedwards, 2017-10-20:
I've made a push #8038 to address the GFX OVERLAY in UBL