MarlinFirmware/Marlin #8022: ABL Giving very variable output - some diag test result attached

lesotter, 2017-10-19:
### Bug Report
- Description: ABL on Marlin 1.1.5 gives erratic bed height results

- Expected behaviour: To give acceptably consistent results across G28 and G29s, performed within seconds of each other

- Actual behaviour: Often there is acceptable consistency between consecutive G29s, but if I perform a G28 and then a set of G29s, the new G29s vary considerably from those done prior to the G28.

I probed using ABL on a 4 x 4 grid to perform my tests.  I then placed 1.6mm copper pennies under probe points 0,0 and 3,3.  Expecting to see height differences of approx 1.6mm, I was surprised to see wild variances.

Probe repeatability test show capacitive sensor producing consistent results.

All mechanical aspects of bed, extruder wiring tension etc checked and OK.  Completely baffled!

I attach a summary of the G28, G29 and M48 output with my commands/actions commented.  I can provide further detail if required (eg M111 S247 output, Configuration.h and adv.h)
[Levelling Issues2.txt](https://github.com/MarlinFirmware/Marlin/files/1398356/Levelling.Issues2.txt)

lesotter, 2017-10-19:
By the way, I am using AUTO_BED_LEVELING_BILINEAR, apologies for not making that clear

lesotter, 2017-10-19:
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1398421/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/1398422/Configuration_adv.h.txt)

lesotter, 2017-10-20:
After almost all day on code reading, micrometers etc, I have found my issue.  It may help others.
I have an adjustable sense distance capacitive 18mm sensor and I was using it to sense the bed from a distance of approx 12mm.  Despite showing healthy repeatability tests, this proved the problem.  As a result of lowering the sensitivity (and hence shortening the triggering distance), my Z Offset has now been changed from -8.75 to -1.05 and the G29 stats are repeatable and satisfactory.  Looking forward to be able to use ABL as intended now.

closed 2017-10-20 by lesotter

empakoso, 2017-10-21:
In that same note.  I found that my capacity sensor was reading the surface of my head bed (aluminum).  It was not reading my tape thickness (or glass for that matter.  If you want even more accuracy try using a light based (IR) Sensor or a touch sensor.
I found that variability also comes in the form of electrical noise from the PWM of the heaters.  Turn them off while probing (in the firmware), or use shielded cable (ground the shield on one end, and use a ferrite bead at the controller end).
These are the things that caused my varying results.
Great find on your part.

Roxy-3D, 2017-10-22:
The good news is, with the mesh based bed leveling systems, some probe inaccuracy doesn't matter.  With both Bi-Linear and UBL, if you have a mesh defined, you can edit it if some part is not perfectly describing the bed.    So as long as your probe can generate a mesh that is 'close', no worries.   

You just print a mesh validation pattern, and edit the areas that the probe returned not completely accurate numbers.

lesotter, 2017-10-23:
Thanks for those comments - do you find you need to set a new Z Offset depending on the temperature of the bed/nozzle?  I have noticed some variation.  I wonder if the sensing distance varies with temperature (both of the target and the sensor itself)

empakoso, 2017-10-23:
Capacitive and inductive type sensors are affected by the heat and switching on/off of the heated bed.  Those are known elements of these types of sensing principles.  I’ve even tested these sensors based on density and they do change (air-gap distances).
This is were a BL touch is so effective (mechanical measured point). I went with a stationary IR Sensor (no moving parts due to decades of sensor repair experience).  IR has its own limitations like reflected light - pay very close attention to shiny surfaces when using IR.  I needed to use black matte high temp paint or even black paper.   It took me a long time to iron that one out :)

empakoso, 2017-10-23:
One last thing, I find that a startup or on connect gcode that is useful is to turn on your heaters and a G4 S300 to wait before a G28.  Very repeatable results due to all that thermal expansion.  I try to get my z offset as close to 0.02mm (10x’s that of my layer thickness).
The absolute best solution in my case was to perfectly level the z axis (I have dual motors), then level the bed to my nozzle at all four corners. (FR- allow the four-corner-bed-leveling feature, no matter what Bed Level feature is enabled .... please, pretty please ... I know there is alway the UBL G29 J3 ... ) then the final step to testing your nozzle offset is to ‘feel’ the nozzle touching the metal of your heated bed.  Yeah I know crazy! ‘Feel’ WTH! I know, trust me within 0.02mm you can actually feel it making contact.  It’s been my most successful offset yet. (I am only 6mos into this hobby so take it for what it’s worth) Measuring/testing method, by feel.  Then I load that tested value in my fw and let G28 take over!!
Good luck.

lesotter, 2017-10-23:
Wow.. lots to take in there.  Due to using almost everything from PLA up to Polycarbonate, I use a wide variety of bed and nozzle temps to worry about, so my sensing distance will change - so my Z offset will need to change too - based on these temps.

lesotter, 2017-10-23:
oh, and isnt 0.02mm actually 1/10 of your layer thickness?  Not sure I understood that bit unless it was a typo!  Thanks

Roxy-3D, 2017-10-23:
> I use a wide variety of bed and nozzle temps to worry about, so my sensing distance will change - so my Z offset will need to change too - based on these temps.

People must be getting tired of hearing how I handle this...   But it does work well for me:   https://github.com/MarlinFirmware/Marlin/issues/8052#issuecomment-338514331

empakoso, 2017-10-23:
Exactly what Roxy said ^^^ . I agree because I have the gouge marks to prove it ;)
My offset test is really for my own understanding. 
And as for thermal expansion there won’t be much difference from PLA to ABS (insignificant really).  The key is heat/expansion must be saturated/stable. For me it’s 5min.
And yes typo - I meant to say my offset is resolved at 1/10 my layer height in accuracy (and repeatability, hence all my notes above)
Good luck.

fiveangle, 2017-10-24:
> And yes typo - I meant to say my offset is resolved at 1/10 my layer height in accuracy (and repeatability, hence all my notes above)

I began to write my disbelief that you were running layer height of 1um - glad to see it was a typo :)

> I find that a startup or on connect gcode that is useful is to turn on your heaters and a G4 S300 to wait before a G28. Very repeatable results due to all that thermal expansion.

I'm too impatient to wait 5 min every print when it's not necessary when printing from cold, so I turn the part cooling fan on high, which blows residual air over the sensor,  and move the gantry away from the bed, then issue an M190 R35 to wait until the bed temperature drops (only gets triggered if immediately starting from previous print, and will then drop out automatically once cooling_slope is low enough).

This ensures inductive sensor is guarenteed to be normalized from the standard start-up gcode.  It can add an extra 3 or 4 minutes if printing back-to-back but at least my first layer is always assured, and adds no delay when printing from cold.  I'm looking forward to implement a bed cooling fan with https://github.com/MarlinFirmware/Marlin/pull/6115 if it doesn't get permanently abandoned :)

empakoso, 2017-10-24:
Yup definitely a typo because I can’t afford that triangulating laser at $5K each :)
Interesting, M190... I like that.
Thanks