MarlinFirmware/Marlin #8025: MEMCOPY(  pointer need , please merge this

studiodyne, 2017-10-19:
@thinkyhead @Roxy-3D @Bob-the-Kuhn 
set_destination_to_current()  NOT WORKING 

The issue is  '  set_destination_to_current() = COPY( = 'memcopy ' and memcopy need ' pointers '
In motion.h i have modified like this and all becomes perfect
Before 
```
inline void set_current_to_destination() { COPY(current_position, destination); }
inline void set_destination_to_current() { COPY(&destination, &current_position); }
```
After
```
inline void set_current_to_destination() { COPY(&current_position, &destination); }
inline void set_destination_to_current() { COPY(&destination, &current_position); }
```
And pointers are transmitted
all works fine

Roxy-3D, 2017-10-19:
current_position[] and destination[] are arrays.   And as such, they are passed as pointers.   So the & operator to get their address should not be needed.

Technically... by using just the name, you are actually doing a &current_position[0] and &destination[0]

studiodyne, 2017-10-19:
Yes i know now , but it' doesn't work on 2.0 in M600 function
Destinations are not applied

AnHardt, 2017-10-19:
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/macros.h#L136

memcopy() needs pointers and gets them in both cases.
How about sizeof()?

studiodyne, 2017-10-19:
inline void set_current_to_destination() { COPY(current_position, destination); }
inline void set_destination_to_current() { COPY(destination, current_position); }
Same allocation. .. " size of "disabled in this case ? No ?

studiodyne, 2017-10-19:
Float array vs float array

studiodyne, 2017-10-19:
I can give you a proof ...
Tomorrow i give you a test code  overloaded of serial echoes
You will see by yourself 
With and without "&"

studiodyne, 2017-10-19:
Hey !!! I have seen the initial declaration of destination and current have only 2 items and not 4 .. x + y + z + e 

May be the compiler is assuming an array of 2 items 
I will try tomorrow

studiodyne, 2017-10-19:
I think you have given me the explanation 
I test my code after boot .. and arrays are not filled

Destination have not  x y z or e datas 
Current too
If one if this two arrays are different  ... memcopy is down 

I will change declaration of destination and current with 4 item equal to zero to fill at max .
I'm sure this is a sizeof  bad trip ☺

studiodyne, 2017-10-19:
But this ... not explain why it works with "&"

AnHardt, 2017-10-19:
Can you point us to the location where only 2 elements are defined? All i can find are ether XYZE or NUM_AXIS - both is 4.

studiodyne, 2017-10-20:
I HAVE FOUND THE BUG , and it's a big BUG 

Code bad written
Memcpy(destination , source )

Look a this 
inline void set_current_to_destination() { COPY(current_position, destination); }
Inversion in arguments
inline void set_destination_to_current() { COPY(destination, current_position); }
Inversion in arguments

1.1.6 is dead
2.0  dead too
It's why we have tons of issues ! Why M600 buggy and more

AnHardt, 2017-10-20:
These functions/macros follow the logic `functionname(target, source)`, with the same order as in `target = source`. That's a very common scheme, the same scheme as in memcopy().

If something is going wrong with this functions, its the use of this function by some developers, ignoring the required order of arguments. It may help to remember the correct order by looking at the functions names - same order.

AnHardt, 2017-10-20:
Maybe things would be more clear to you, if the function names are changed

set_current_to_destination() -> set_current_from_destination()
set_destination_to_current() -> set_destination_from_current()

At least if translated to my native language this is the same, because "Set target" is the *strong* part in this order.

Even better would be to omit this redundant functions at all. Let's expand them to:

memcpy(target, source, min( sizeof( target), sizeof(source)));

That would be readable by any c-novis knowing a bit about the standard libraries. Because of inlining they expand to this anyway, always and everywhere. But it's contrary to @thinkyhead 's philosophy of deepy hiding everything in functions/macros, so only Marlin experts, having looked up functions/macros -  definitions long enough to know then by heart, can read/use this code.

Roxy-3D, 2017-10-20:
> set_current_to_destination() -> set_current_from_destination()
> set_destination_to_current() -> set_destination_from_current()

I'm in favor of the change.   If there are no objections...   Let's make this change.

studiodyne, 2017-10-20:
Roxy make a simple batch and original files will no be impacted , i have tried it and it's perfect ,

set_current_to_destination  ----> set_current_to_destination2
set_destination_to_current -----> set_current_to_destination 
set_current_to_destination2 ----> set_destination_to_current

Roxy-3D, 2017-10-20:
No...   People are pretty much used to the way it is.   And AnHardt is correct that the established order of parameters match the standard memcopy (strcpy) etc.      

We can put the 'from' in the name to make things much more clear.

AnHardt, 2017-10-20:
Thanks @Roxy-3D.
That could be step 1.
The one has to search for the not working parts of the code where, maybe, the wrong direction was used.
A **global** swap will, for sure, destroy more than it helps.

studiodyne, 2017-10-20:
Yes it's a good idea for all that have already develloped some code , and this will avoid confusion 
If there are some new name , it's better to mix the old inverted names with our newest code that have the old inverted name 
LOLLL

Yes , to avoid a big crisis , it's better to change radically the names

studiodyne, 2017-10-20:
I have made the PR and detailled explanation 
[Advanced Pausing - M600 fix , works fine now](https://github.com/MarlinFirmware/Marlin/pull/8040)

studiodyne, 2017-10-20:
@AnHardt  A global swap is good , if the dev work alone , but in a git communauty .... This simple batch can make so many wasted time for so many people 
Changing name is radical , visible , easy to understand and to fix

AnHardt, 2017-10-20:
A global renaming will just help to lower the likelihood someone assumes they do the opposite of what they do.
A global reversion of what they do, will mess up all the code, currently properly working, with this functions - will (maybe) help at the few places currently not working.

dot

Roxy-3D, 2017-10-20:
Agreed.    A global renaming to 
```cpp
set_current_to_destination() -> set_current_from_destination()
set_destination_to_current() -> set_destination_from_current()
```
is safe and potentially will help people.    But changing the order of parameters is guaranteed to cause huge problems.

closed 2018-02-06 by thinkyhead