MarlinFirmware/Marlin #8026: [FR] Change Thermistor Tables Live?

PhatHub, 2017-10-19:
Suggestion:

Request: Is it possible to have some sort of way to change thermistor tables without reburning firmware?

Use case: If someone has a broken thermistor, or purchases a hot end with a different thermistor, there should be a way to change the thermistor table to (at least temporarily) run the different resistance curves before recompiling and uploading the new firmware/configuration. (much like changing the steps on an extruder with gcode if you change extruders.)

Some thoughts:

1. Store all tables. For those of us with lots of space, try to store all the thermistor tables. Let it be adjustable via a short gcode command to select the correct thermistor table.
2. Rewrite tables on the fly. For those who don't have the space, create a gcode command that allows a user to painstakingly replace all the values in the table.

PhatHub, 2017-10-19:
Oops. Didn't see "M305."

PhatHub, 2017-10-19:
Oops Sorry, I discovered M305 and closed this request, then saw again that it wasn't in Marlin, so I re-opened it.

thinkyhead, 2018-02-09:
Demand is very low and devs have a lot on their plates. You may be able to hire a developer for $$$ to do this if you very much need it.

boelle, 2019-02-19:
i would not use this, if the thermistor is not up to spec i would get a new one so i can know for sure its working

and having different thermistors will just be a mess and you end up mixing them up

InsanityAutomation, 2019-10-15:
@boelle @shitcreek both a duplicate and done. Can be closed.

closed 2019-10-15 by boelle