MarlinFirmware/Marlin #8027: Setting X_MIN_POS to a negative value?

gcormier, 2017-10-19:
When homed at the corner, 0,0, I have spare travel on my X axis. I want to put a brush there for nozzle wipe.

Is setting X_MIN_POS to -10 acceptable? From my side, it's easiest - no offsets or anything needed in slicer, no worries about ever printing outside the bed. (Slicer will never print below 0,0)

Thanks!

fiveangle, 2017-10-19:
Yes.  That's exactly how it is supposed to be used.  In conjunction with the BED_SIZE parameters, it locates the movable/probable area (with probable areas beyond the bed available by manually hard-coding the probing limits).

-=dave

gcormier, 2017-10-19:
Thanks @fiveangle ! And it will still respect the end-stops, so if I (stupidly) set X_MIN_POS to -100, it would stop sooner? (Let's say my bed starts at 20 offset from the X stop).

Roxy-3D, 2017-10-20:
Sorry...   My touch screen thought I clicked close...

Roxy-3D, 2017-10-20:
And some additional Sanity Checks would be wise.    The switch to X_BED_SIZE and Y_BED_SIZE wasn't done as transparently as it should have been done.   It broke a lot of the example Configuration.h files.

gcormier, 2017-10-20:
Yeah.. I was worried about UBL, it talks about MIN_POS.. I think it should be fine however as I traced some of the header files and it's ok, UBL starts at some offset (1) for X and Y and moves to the max bed size.

closed 2018-02-09 by thinkyhead