MarlinFirmware/Marlin #8028: [FR] Enable SD Card Logging while printing from SD card

kwisatzsawyer, 2017-10-19:
------------------------------------------------------------
### Feature Request
I want to write code to attempt to resume prints from interrupted power. I'd prefer to do this all in Marlin, but anytime I run a M928 log.txt command the print stops (traced that to cardreader.cpp). Essentially, the code is setup to only be able to write to OR read from one file. And whenever a M928 command is encountered the firmware stops printing from the SD card and closes the gcode file.
Is there a way to implement reading from one file and writing to another inside of the resources we have? 

If so, I think I can add basic resume functionality (estimate which was the last commend executed, calculate a new start point, create new gcode file containing 1) startup conditions; 2) start location; 3) uncompleted gcode from the print file, and then instruct the printer to resume printing.

NUU-B2-417, 2017-11-08:
我也想知道

thinkyhead, 2018-02-09:
No, you cannot write to the same SD card that you're reading from.

Print recovery must be done in a way other than continuous wasteful logging.

When a power-loss is detected, remember the position in the SD file. Close the SD file. Save the last position to EEPROM or to a hidden SD file. At reboot look for the EEPROM token or SD file and ask the user if they want to resume.

Anyway, this is a duplicate feature request. Closing.

closed 2018-02-09 by thinkyhead

AnHardt, 2018-02-09:
I hoped it would be that simple.
Between the move we are actually executing (and its questionable if we always can finish this move (could last several seconds)) 
and the file pointer we have, at least, the planner buffer. That's contend would be lost.

AnHardt, 2018-02-14:
My 'feeling' says we have to track file_position/line_number and submove_number down into the planner buffer to have the exact point.

AnHardt, 2018-02-14:
From a todays Travis run:
```
The command "opt_enable ULTRA_LCD" exited with 0.
16.04s$ build_marlin_pio ${TRAVIS_BUILD_DIR} ${TEST_PLATFORM}
In file included from Marlin/src/lcd/../inc/MarlinConfig.h:39:0,
from Marlin/src/lcd/ultralcd.h:26,
from Marlin/src/lcd/ultralcd.cpp:27:
Marlin/src/lcd/../inc/../core/../lcd/language/language_en.h:44:66: warning: initializer-string for array of chars is too long [-fpermissive]
#define WELCOME_MSG                         MACHINE_NAME _UxGT(" Ready.")
^
Marlin/src/lcd/../inc/../core/language.h:28:18: note: in definition of macro '_UxGT'
#define _UxGT(a) a
^
Marlin/src/lcd/ultralcd.cpp:101:51: note: in expansion of macro 'WELCOME_MSG'
char lcd_status_message[MAX_MESSAGE_LENGTH + 1] = WELCOME_MSG;
^
The command "build_marlin_pio ${TRAVIS_BUILD_DIR} ${TEST_PLATFORM}" exited with 0.
```
Seems to be related.

thinkyhead, 2018-02-14:
Changed the code to init `lcd_status_message` with `LCD_MESSAGEPGM(WELCOME_MSG)`.

thinkyhead, 2018-02-14:
> My 'feeling' says we have to track file_position/line_number and submove_number down into the planner buffer to have the exact point.

The more data the better. We'll have a peek at what Prusa Firmware does.

vjvarada, 2018-05-14:
there is a variable called  "sdpos" in cardreader.cpp that can be used to check the last SD position. If you have this, you can use the M26 command to resume.

thinkyhead, 2018-05-14:
> there is a variable called "`sdpos`" in `cardreader.cpp`

Indeed. If you examine the power-loss recovery code it saves this value (along with other useful data) to the recovery BIN file.