MarlinFirmware/Marlin #8034: Tevo Black Widow does not load at startup (1.1.6)

KeMpeRr, 2017-10-20:
### RC6

"The flight is normal"

But it is necessary to update the firmware

### 1.1.6

The printer does not start. 
Why? i do not know :/

AnHardt, 2017-10-20:
If not you, how can we - with the provided information.
Config-files please. If possible the old and the new once.

KeMpeRr, 2017-10-20:
Ok! 

[Marlin-1.1.6conf.zip](https://github.com/MarlinFirmware/Marlin/files/1402038/Marlin-1.1.6conf.zip)
[Marlin-RC6.zip](https://github.com/MarlinFirmware/Marlin/files/1402039/Marlin-RC6.zip)

AnHardt, 2017-10-20:
Differences at:

POWER_SUPPLY
TEMP_SENSOR_0
TEMP_SENSOR_BED
???ENDSTOP_INVERTING block
...

stopped searching here.
Please have an other look into your new configuration files

KeMpeRr, 2017-10-20:
I tried to correct all the inconsistencies in configuration.h
[Marlin-1.1.6conf2.zip](https://github.com/MarlinFirmware/Marlin/files/1402636/Marlin-1.1.6conf2.zip)

[image]

¯\_(ツ)_/¯

jannoke, 2017-10-20:
You don't have a display controller defined...
//
// RepRapDiscount Smart Controller.
// http://reprap.org/wiki/RepRapDiscount_Smart_Controller
//
// Note: Usually sold with a white PCB.
//
//#define REPRAP_DISCOUNT_SMART_CONTROLLER

This needs to be set. 
So be lazy.. Get this. Just got my printer and got fw updated.
[Marlin1.1.6_conf.zip](https://github.com/MarlinFirmware/Marlin/files/1403087/Marlin1.1.6_conf.zip)

closed 2018-02-09 by thinkyhead