MarlinFirmware/Marlin #8036: NeoPixels with NEOPIXEL_STARTUP_TEST causes bootloop

NiklasOlofsson, 2017-10-20:
### Bug Report
- Description: Using NEOPIXEL_LED with NEOPIXEL_TYPE   NEO_GRB
- Expected behaviour: Leds should cycle colors on startup
- Actual behaviour: It displays red, green and the firmware reboots, nothing on the LCD.
- Steps to reproduce:
  - Enable NEOPIXEL_STARTUP_TEST

AnHardt, 2017-10-20:
Without having looked at the actual code:
watchdog is already active but not refreshed in your long lasting (>3s) procedure.

Tannoo, 2017-10-23:
I did this to mine to make it work.

```
void setup_neopixel() {
  pixels.setBrightness(NEOPIXEL_BRIGHTNESS); // 0 - 255 range
  pixels.begin();
  pixels.show(); // initialize to all off

  #if ENABLED(NEOPIXEL_STARTUP_TEST)
    delay(2000);
    #if ENABLED(USE_WATCHDOG)
      watchdog_reset();
    #endif
    neopixel_set_led_color(255, 0, 0, 0, 255);  // red
    delay(2000);
    #if ENABLED(USE_WATCHDOG)
      watchdog_reset();
    #endif
    neopixel_set_led_color(0, 255, 0, 0, 255);  // green
    delay(2000);
    #if ENABLED(USE_WATCHDOG)
      watchdog_reset();
    #endif
    neopixel_set_led_color(0, 0, 255, 0, 255);  // blue
    delay(2000);
    #if ENABLED(USE_WATCHDOG)
      watchdog_reset();
    #endif
  #endif
  neopixel_set_led_color(0, 0, 0, 255, 255);    // white
}
```

Tannoo, 2017-10-23:
Neopixel strip now looks like a disco when warming up any heater.
Neopixel strip is also now unreliable on color selections.

AnHardt, 2017-10-23:
Someone promised this.
https://www.youtube.com/watch?v=YxvBPH4sArQ from about 1:46
Do you remember? :-(

Tannoo, 2017-10-23:
Yup. Lol

Tannoo, 2017-10-24:
I've changed this to what it should have been:

```
void setup_neopixel() {
  pixels.setBrightness(NEOPIXEL_BRIGHTNESS); // 0 - 255 range
  pixels.begin();
  pixels.show(); // initialize to all off

  #if ENABLED(NEOPIXEL_STARTUP_TEST)
    safe_delay(2000);
    neopixel_set_led_color(255, 0, 0, 0, 255);  // red
    safe_delay(2000);
    neopixel_set_led_color(0, 255, 0, 0, 255);  // green
    safe_delay(2000);
    neopixel_set_led_color(0, 0, 255, 0, 255);  // blue
    safe_delay(2000);
  #endif
  neopixel_set_led_color(0, 0, 0, 255, 255);    // white
}
```

Xeroxxx, 2017-10-24:
I can confirm this happens for me too. LCD Display stays empty, LED cycles through green, red, off etc.

Tannoo, 2017-10-25:
In `neopixel.cpp`.

Tannoo, 2017-10-25:
It's not an addition... it's a change.

In the funtion `setup_neopixel()`, change all the `delay(2000)`'s to `safe_delay(2000)`.

Tannoo, 2017-10-25:
That will insure all background processes keep processing.

Xeroxxx, 2017-10-25:
Hmm...and this file can be found where?

Can't find the function in Marlin or NeoPixel Library.

Tannoo, 2017-10-25:
Marlin/src/feature/leds/neopixel.cpp

Xeroxxx, 2017-10-25:
Ahh okay this is 2.x. However search function of github does not work proberly and I wasn't able to find the code.

Code for 1.1.x can be found in Marlin_main.cpp

Xeroxxx, 2017-10-25:
Works great, are you going to push this to the repo?

Tannoo, 2017-10-25:
I can't push. I'm not that important. Lol

Tannoo, 2017-10-25:
I can nudge.

Xeroxxx, 2017-10-25:
gj :) 👍

Tannoo, 2017-10-25:
Did I have a type-o?

Xeroxxx, 2017-10-25:
I guess so.

SCiunczyk, 2017-11-01:
Issue could be closed I think. There are PR fixing this.

closed 2018-02-09 by thinkyhead