MarlinFirmware/Marlin #8042: IDEAS FOR POWER OUTAGE FEATURE

studiodyne, 2017-10-20:
I'm sure this topic have so many clones , but i want to talk about my new idea , now i have experience of robotic printing and i know what is ' folkloric idea ' and what is 'probable '

And now , i need help because , i have a ' probable idea '

The theory 
An UPS ' of course ' 
+ Enough power to finish the z current layer ' it's a good thing to have a finished layer '
On resume , we probe manually somewhere in the layer and the bltouch give the Z height to start print

Now , we have z  , and no stress of colision

But i need an idea to make a probing to retrieve exactly the x and the y .
Use a BLTOUCH algorythm to probe a tower and find the edges , will never give enough precision 

Is there someone that have a good idea , to use a sensor ' if possible a bltouch vertical probe ' , to make a probing algorythm that will probe a printed form that can give X and Y with high precision
IF we solve this , we can restart the print , and to make the code is not very hard

Thks

AnHardt, 2017-10-20:
If gravitation is no problem for the BLTouch - if it works horizontally with a similar precision.
To make a 3D-scanner from a BLTouch it needs two additional servos to make it possible to point in all directions and _some_ software.

studiodyne, 2017-10-20:
we have written the same thing a the same moment  lol
I don't know if you know well bltouch , but , it's not a precise sensor for me , may be i'm wrong , and the bad repeatability is just mechanical , but  to place it in x and y , can give enough precison to resume a homing , i don't know
Do you think it will give serious results  ?

studiodyne, 2017-10-20:
We can use only one servo and port , the software can use three at the same time , but only the axis in movement will be concerned when trigged

studiodyne, 2017-10-20:
here we need now 
Ups or battery to finish a layer
Three bl touch 
Some days and nights of coding and UPLOADING the Avr !
and it can be possible

studiodyne, 2017-10-20:
@thinkyhead , do you think , to have three bl touch for high precision homing in all axes , one big battery or ups 'enough power to finish the current layer ' , and a resuming code  , that ask a human , to fiind manually , an area to probe the z on the object already printed , and after make a x y homing , to restart printing after power outage is enough possible and enough probable?
I think yes !

studiodyne, 2017-10-20:
But , in a case of spiral vase , not recommended
I think in a normal print we will see the restart trace , sometimes we don't care , but sometimes it's important and the user will restart all the print 
But in many case ' VERY LONG PRINT' , we will be all happy to resume even though the print have resumes traces on a certain layer

studiodyne, 2017-10-20:
This system don't need high end material and can work on all printer , bltouch holder is easy to make 
The code is not terrific , when i see ' ubl ' , i know now all is possible lol 
I think it's a good idea and very probable !

vido89, 2017-10-20:
Im sure that you cant probe printed object with BlTouch its just not possible. On other hand did you saw Prusa i3 mk3 and there solution for power cut if not you should check it out.
And plese close this issue

Grogyan, 2017-10-20:
I would expect that Josef would have just completed the buffer, or dump the buffer onto the SD card.
Likewise he could also have virtual memory on the SD card to monitor where in the file it was at. 
Thereby not using EEPROM at all

studiodyne, 2017-10-21:
@vido89 , If you don't like this topic , you are not forced to stay here and to ask to close , be cool please. And for information, probing an already object is possible , because this will be made manually , the user , will move himself the carriage to an area , and probe , and will have a good z ! ' already explained' , nothing that is terrible, may be you have understant ' automatic probing ' , no , the area will be set by lcd or manually , z is not a problem for me
@Grogyan : The software is not a problem , if we solve the way to restore x and y precisely , we win the help of the communauty for the software
It's not the software that afraid me , i have tons of way to solve , my only wish is mechanical
How to restore , for all printers that exist , the previous x and y before power outage 
2 choices , I ensure a perfect ultra repeatable homing that can be the same everytime , with perfection limit switche , or , i make a edge probing on a ' special for this ' printed object.

We can restore x and y , if we build a special machine that lock x and y , but we want to make it for all printers as possible. The only way  is to restore x and y with an intelligent method.

All my friends have CREATBOT D600 with ups inside and power outage resuming , and yes it's true , it works , but it's a special machine made for this , with axis that doesn't loose step , and  with ups inside that permit to achieve the current buffer before stopping

Give me all idea ' except high end precision limit switch ' , to restore x and y , by probing a printed object if possible ... I think not , but I listen all

vido89, 2017-10-21:
@studiodyne Its not that I dont like this topic its you guys over-thinking this with probe and stuff and one question how will printer know x,y,z coordinate if you move print head ? 
The solution is to to detect power cut, quickly cut power to bed/hotend and save current position in eeprom and thats it with one condition you will need z axis to stay in place :) Cheers

AnHardt, 2017-10-21:
@vido89 
Please explain your recovery strategy.

mcmatrix, 2017-10-21:
I also agree with vido89 it has to simple. I found a topic in arduino forum that may help. It needs additional electronics but you’ll get the idea.
https://forum.arduino.cc/index.php?topic=417123.0
Post #2

Sineos, 2017-10-21:
Actually such systems seems already available:
[LERDGE 3D Printer Power Monitoring Module Continued to Play Printing Automatically Put off Management Module for Lerdge Board](https://de.aliexpress.com/item/3D-Printer-Power-Continued-to-Play-Module-Printing-Automatically-Put-off-Module-for-Lerdge-Motherboard/32800910006.html?spm=a2g0x.10010108.1000016.1.758a66da0owPga&isOrigTitle=true)

Rigid3D, 2017-11-17:
I am planning to add this feature in the following months. Although additional hardware is needed, I see this process as the most fail proof one. I am planning to power the controller with a cheap powerbank and record the remaining gcode to SD card whenever a power outage is detected. After the power outage user may decide to continue the print. Below is the process steps:

1. Power outage detected.
2. Print paused, heaters are shut down and Z lifted using the power stored in the caps.
3. incomplete-print.gcode file created on the SD Card.
4. Start gcode which is defined in configuration.h is added to the gcode file. (probably HomeX HomeY, HomeZmax and temperatures are added)
5. Remaing part of the interrupted print is added to the gcode file.
6. Message user to shut down the printer for powering off the powerbank.
7. After power outage if the print was not detached from the bed the user may just print incomplete-print.gcode to resume the print.

An additional process such as calculate the Zmax location is needed on Marlin. So that whenever a bed leveling is done, user can run this process and the printer does a sequential probing of Zmin and Zmax to calculate the Zmax location and record it to the eeprom.
 
Currently I am trying to develop a power outage detection circuit that uses an optocoupler. This circuit will detect the power outage and also supply 5V to powerbank for charging. Afterwards I will work on addition of the features to Marlin.

Your comments are wellcome for the proposed solution.

Gaben5224, 2018-01-03:
Hey Guys!

I think this is not so complicated. (but I may be wrong)
Of course I can't do this, because I'm not a programmer, but some people from this community could.

The important things:
First of all we need a UPS. 
Second we need some kind of special hardware that can sense the outage, like this: https://www.aliexpress.com/store/product/3D-printer-Parts-MKS-DET-outage-Detection-module-work-with-MKS-TFT-LCD-perfect-power-continue/1502993_32667488260.html
Third, we need to create a new gcode.

My idea for the gcode creating:
When the outage is happening, it should read the actual temps and store it for a moment. After that it should read the actual Z position. 

The start gcode may look like this:
G28 X Y ; home x and Y
Maybe we need a new gcode that turns off the Z home position requirement.
G92 Z[stored_position] ; Sets the latest Z position back
M109 S[stored_extruder_temp] ; Sets extruder temp back to the stored temp 
M190 S[stored_bed_temp] ; Sets bed temp back to the stored temp

Than the gcode could be continued from the last line.

What do you think? Would it work?

thinkyhead, 2018-02-11:
Duplicate of #2085

closed 2018-02-11 by thinkyhead