MarlinFirmware/Marlin #8045: [FR] dual X carriage Cartesian disable any unused stepper motors including the unused extruder

bruce356, 2017-10-21:
### Feature Request Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`
Hi, on a dual X carriage Cartesian printer is it possible to disable the carriage that is not being used. Very often only one extruder is being used to print with (X or X2) so the other carriage is not required to be powered up for the duration of that print. How if it is possible does one disable any unused stepper motors including the unused extruder stepper motor. If its currently not possible can this option be added.
Thanks and regards.

thinkyhead, 2018-02-09:
Yes, this would be a good idea to save wasted energy. The X/X2 and E0/E1 steppers should be powered and time-out separately when not in duplication mode.

boelle, 2019-06-22:
wrong label

InsanityAutomation, 2019-06-22:
Id consider this as needing another mode in M605 that will force the unused head to rehome on tool change otherwise there will be issues. Unused extruder maybe but not the movement stepper.

Regardless this is something id be very hesitant to use or recommend using.