MarlinFirmware/Marlin #8052: ABL bilinear weird behaviour

executivul, 2017-10-21:
Hi,
Using bilinear leveling for ABL does not compensate for bed height, I have high/low spots.
My bed has under 0.1mm of error across 300x300mm (piezo probe, 3 micron deviation) so I've put some paper on the bed for a fake huge 0.5mm bump under 4 probing points out of 7x7 grid to be able to measure by eye ([https://i.imgur.com/pKrwyyr.jpg](url))
After board reset I run G28 and G29 then I go to (0,20,0.1) G1 X0 Y20 Z0.1 F2400 and:
- G1 X300 F2400 to simulate a straight path parallel to X axis across the "bump" all I get is a little "twitch" from Z steppers, no real 0.5mm compensation.
- G1 X120 F2400 (so line ends on top of the bump) then I get some real bed correction.
If I need to print a large part the slicer won't fragment longer G1 moves. 
What can I do?

I don't need or want EEPROM usage, that's why I use bilinear instead of UBL, I miss the old grid style compensation from RCx version. Pleasae help! Thank you!

Roxy-3D, 2017-10-21:
It sounds like you need and should want to use the EEPROM to store your mesh.

If you are willing to store the mesh, both Bi-Linear and UBL allow you to edit the mesh.

executivul, 2017-10-21:
Ok. I shall try and use the EEPROM. (even though it's easier for me to modify the config and flash than to make changes which I forget about and then later flash a config that doesn't have them and wonder why something is no longer working... been there)

Meanwhile can you tell me if it's Marlin's job to fragment a long G1 and apply Z correction along it? Or shall I use another program to fragment the long G1 moves so Marlin only applies the correct Z for each segment?

And is there a thorough documentation for UBL, how do I tell what type of compensation to use? what gcode with what parameters to use, etc

Roxy-3D, 2017-10-21:
> Ok. I shall try and use the EEPROM. (even though it's easier for me to modify the config and flash than to make changes which I forget about and then later flash a config that doesn't have them and wonder why something is no longer working... been there)

Whoa!   I have always had the EEPROM enabled.  And I never store my setting or configuration information in the EEPROM.   I always update the configuration.h files when I want a change to my configuration (even though that is not necessary).   I prefer to have the 'source code' for my printers at various points in time archived away.     

So on one hand I understand your point of view.   But that doesn't mean you can't use UBL.    And UBL provides a way to back up your mesh so it can be moved from machine to machine or reconstructed at a later point in time.   Or for that matter...  so it isn't stored in the EEPROM.

> Meanwhile can you tell me if it's Marlin's job to fragment a long G1 and apply Z correction along it? Or shall I use another program to fragment the long G1 moves so Marlin only applies the correct Z for each segment?

Some of the bed leveling schemes will break up a long G1 into smaller segments.   UBL does this into segments that cross mesh cell boundaries.   If you give a G1 command to go from one corner of the bed to another corner, you will see the nozzle move up and down depending on where the nozzle is positioned at the moment.   The long G1's are broken into smaller segments so the nozzle can accurately track the bed.

> And is there a thorough documentation for UBL, how do I tell what type of compensation to use? what gcode with what parameters to use, etc

If you follow the 'cheat sheet' at the front end of:  https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md   you should end up with a usable mesh saved to EEPROM.   If you follow the order of commands in the 'cheat sheet' UBL should be active with the mesh loaded automatically when the printer resets or powers up.   At that point, you only need to do a G28 and UBL is alive and working...

executivul, 2017-10-21:
@Roxy-3D UBL seems to do the trick. Is there any way to make it probe the bed as a grid instead of the diamond shaped movement? Takes a long time just moving in this diamond/dizzy/random pattern for a 15x15 grid.

Roxy-3D, 2017-10-22:
Well...   The initial probing is a one time event.   And that pattern is deliberate.   It isn't making a diamond.  It is trying to fill in the next closest mesh point to the starting location.   The reason is it is slowly expanding the size of the known mesh.  And if there is a problem, it gives the user time to stop the machine so it doesn't drive the nozzle into bed and damage something.

Bob-the-Kahn actually changed the code to have a grid option for the initial probing.   But he decided the 15 or 20 seconds it saved wasn't worth the extra complexity.    G26 behaves similarly.   But G26 allows a C parameter to let the nozzle complete mesh point patterns close to where the nozzle is currently located and that does dramatically speed up the validation pattern.

You will also see that expanding outward pattern when you have a mesh validation pattern printed and you do a G29 P4 R to edit an area of the mesh.   It will start with the closest mesh point to the nozzle, and expand outward until the user stops it because the whole bad area has been adjusted.

The pattern is a little bit unorthodox.  But you will get used to it.   And it is there for a purpose.

executivul, 2017-10-22:
Thank you for your kind explanation. I asked because I kinda like to run the probing before every print just in case I bumped on something while removing the last print. I know my bed is perfectly flat (8mm preciosion milled alu + 4mm glass clipped on top) and all my errors are due to mechanical errors (2020 profiles, corner joined cube, 0.2mm bowtie twist shaped error) and every time I probe I get the same pattern, but just in case :)

Roxy-3D, 2017-10-22:
Rather than probing every time...   You might want to consider doing what I do.    I have my nozzle home .5 mm too high.   I have a BL-Touch on the machine and it pretty much works.  But some times, it can return a goofy value.   I don't want my nozzle to gouge into the bed.   So I home .5 mm too high...

And when my print starts, I double click to get to the Z-Baby Stepping screen and lower the nozzle to exactly the right height as it is printing the objects skirt.    Usually it is exactly 3/4 of a twist of the encoder wheel to get to the correct height.   But every once in a while, I'm glad I'm doing it the way I am.  

That method saves you the probe time.   And really...   you want to be there as the print starts any way, so it isn't really lost time.

closed 2018-02-09 by thinkyhead