MarlinFirmware/Marlin #8053: Compile Error M48 Z_MIN_PROBE_REPEATABILITY_TEST

cdedwards, 2017-10-21:
Is there any reason why the following is in M48.cpp

#if DISABLED(DELTA)
  #include "../../feature/bedlevel/bedlevel.h"
#endif

with this the way it is, enabling Z_MIN_PROBE_REPEATABILITY_TEST in Configuration.h fails due to a missing function definition. removing the #if DISABLED(DELTA) #endif pair allows marlin to compile

Roxy-3D, 2017-10-21:
> Is there any reason why the following is in M48.cpp

Yes.   It is easy to understand how this can happen with all the refactoring going on.  Can you update the thread with you Configuration.h files and the error message you are seeing?   I'll try to get that cleaned up.

cdedwards, 2017-10-21:
all I did was enable Z_MIN_PROBE_REPEATABILITY_TEST in configuration.h.  which gave me an error message during compile. removing the #if DISABLED(DELTA) #endif found the function definition needed for M48.cpp and everything compiled.

cdedwards, 2017-10-21:
function definition for set_bed_leveling_enabled() was missing.

closed 2017-11-17 by cdedwards