MarlinFirmware/Marlin #8054: Succesful compile with errors

Turtle269, 2017-10-21:
------------------------------------------------------------
### Compile Error
When i try to compile with an old version of marlin, it will only compile with arduino 1.6.1, this is for a flsun delta kossel, i was having problems the new firmware recommended, which has lead me back a couple of steps, what was happening is, when i go to heat HE0 heater_0 on board, the led for mosfet would light, but the 12 volt rail on psu wouldnt load, giving me PRINT HALT within temp timing hystersis, i was able to compile the old marlin that will only compile on 1.6.1, with the change of code from ''fpost , to ''filepost in the sdbasefile.h/cpp, which has enabled it to compile (it would error, not compile without the change of name in all variations), So after achieving a compile on arduino1.8.5, i notice this in the compiling message, has the program "rewrote" new variables for these messages, or is it something i should look into?, im only new to all this coding, i am almost certain there is a pwm out somewhere with the heater, but probably a faulty board. any help with this error greatly appreciated.
```

In file included from sketch\Marlin.h:21:0,

                 from sketch\thermistortables.h:4,

                 from sketch\Configuration.h:598,

                 from C:\Users\Turtle\Desktop\çº¿è§„Marlin\Marlin\Marlin.ino:34:

sketch\fastio.h:1418:0: warning: "DIO73_PIN" redefined

 #define DIO73_PIN  PINJ4

 ^

sketch\fastio.h:1406:0: note: this is the location of the previous definition

 #define DIO73_PIN  PINJ3

 ^

In file included from sketch\Configuration.h:598:0,

                 from C:\Users\Turtle\Desktop\çº¿è§„Marlin\Marlin\Marlin.ino:34:

sketch\thermistortables.h:72:1: warning: narrowing conversion of '2.405e+2' from 'double' to 'const short int' inside { } [-Wnarrowing]

 };

 ^

sketch\thermistortables.h:72:1: warning: narrowing conversion of '2.265e+2' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortables.h:72:1: warning: narrowing conversion of '2.225e+2' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortables.h:72:1: warning: narrowing conversion of '2.185e+2' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortables.h:72:1: warning: narrowing conversion of '2.135e+2' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortables.h:72:1: warning: narrowing conversion of '2.095e+2' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortables.h:72:1: warning: narrowing conversion of '1.955e+2' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortables.h:72:1: warning: narrowing conversion of '9.35e+1' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortables.h:72:1: warning: narrowing conversion of '7.35e+1' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortables.h:72:1: warning: narrowing conversion of '5.85e+1' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortables.h:72:1: warning: narrowing conversion of '4.85e+1' from 'double' to 'const short int' inside { } [-Wnarrowing]

In file included from sketch\Marlin.h:21:0,

                 from sketch\Marlin_main.cpp:30:

sketch\fastio.h:1418:0: warning: "DIO73_PIN" redefined

 #define DIO73_PIN  PINJ4

 ^

sketch\fastio.h:1406:0: note: this is the location of the previous definition

 #define DIO73_PIN  PINJ3

 ^
[... 671 lines cut ...]

```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

Roxy-3D, 2017-10-21:
> Attach a ZIP of Configuration.h and Configuration_adv.h by dropping here.

Turtle269, 2017-10-21:
[MarlinKurtKossel1.6.zip](https://github.com/MarlinFirmware/Marlin/files/1404470/MarlinKurtKossel1.6.zip)

Please let me know how that went, Thanks.

Regards, Kurt

Turtle269, 2017-10-21:
Thanks for your reply Roxy,

I have come to a conclusion to just use 1.6.1 and not worry about updating any firmware, if i choose to do this ill just buy my own board and do a config from the start...

Thanks Roxy.

Phr3d13, 2017-10-25:
Did you try any of the delta example configurations? 1.1.6 has 2 different FLSUN ones and a few other generic ones

closed 2018-02-09 by thinkyhead