MarlinFirmware/Marlin #8060: DEFAULT_MAX_ACCELERATION issue

magdanoz, 2017-10-22:
Hi, folks!
How to deal with this issue? I had the one with the U8glib, too but somehow I managed to fix it. Thanks in advance! I'm using the latest version of Marlin - 1.1.x.
Best regards!

### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Arduino: 1.8.1 (Windows 8.1), TD: 1.35, Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from C:\Users\User\AppData\Local\Temp\arduino_build_258412\sketch\MarlinConfig.h:39:0,

                 from C:\Users\User\AppData\Local\Temp\arduino_build_258412\sketch\G26_Mesh_Validation_Tool.cpp:27:

SanityCheck.h:1301: error: 'DEFAULT_MAX_ACCELERATION' was not declared in this scope

                 sanity_arr_3[] = DEFAULT_MAX_ACCELERATION;

                                  ^

SanityCheck.h:1304: error: static assertion failed: DEFAULT_MAX_ACCELERATION requires 4 (or more) elements.

 static_assert(COUNT(sanity_arr_3) >= XYZE, "DEFAULT_MAX_ACCELERATION requires 4 (or more) elements.");

 ^

Using library U8glib in folder: C:\Program Files (x86)\Arduino\libraries\U8glib (legacy)
exit status 1
'DEFAULT_MAX_ACCELERATION' was not declared in this scope
```

GMagician, 2017-10-22:
DEFAULT_MAX_ACCELERATION should be declared in Configuration.h. Sure that it is all ok such file?

magdanoz, 2017-10-29:
Sorry about the late reply but I have a lot of stuff to do lately and didn't have enough time for my printer! :)
It was completely my fault! I was updating my Configuration file from an older Marlin version and the issue I've reported was actually a copy/paste error.
Everything is fine now! Thanks for the help!

fiveangle, 2017-10-29:
good it's all sorted now.  can you please close ?

-=dave

closed 2018-02-09 by thinkyhead