MarlinFirmware/Marlin #8062: new prusa Filament Runout Sensor

Radiu, 2017-10-23:
I just one question
i was looking into the new prusa mk3 that have a sensor filament runout that detects presense and movement of filament, so if filament runout or if hotend clogs will stop machine, pushes filament out and when replace will restart from where it was
i know that marlin already have filament runout option implement, i didnt get a chance to test it yet
does marlin also pushes out the remaining filament out when it have runout also?
from what i see they using optical mouse sensor
is there any schematic how to build that circuit available?

thanks

Roxy-3D, 2017-10-23:
> i know that marlin already have filament runout option implement, i didnt get a chance to test it yet

Yes,   And GCode implimentation with M600

> does marlin also pushes out the remaining filament out when it have runout also?

Yes.   You can set the retraction amount and a whole host of other parameters too.

> from what i see they using optical mouse sensor

It can be made to work that way.  But a simple micro-switch (with a roller on the lever arm) is very reliable and simple to make work.

> is there any schematic how to build that circuit available?    

Do the same thing you would do to make an end stop switch.  That will work for a filament sensor.
```cpp
/**
 * Advanced Pause
 * Experimental feature for filament change support and for parking the nozzle when paused.
 * Adds the GCode M600 for initiating filament change.
 * If PARK_HEAD_ON_PAUSE enabled, adds the GCode M125 to pause printing and park the nozzle.
 *
 * Requires an LCD display.
 * This feature is required for the default FILAMENT_RUNOUT_SCRIPT.
 */
#define ADVANCED_PAUSE_FEATURE
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  #define PAUSE_PARK_X_POS 75                 // X position of hotend
  #define PAUSE_PARK_Y_POS 75                 // Y position of hotend
  #define PAUSE_PARK_Z_ADD 10                 // Z addition of hotend (lift)
  #define PAUSE_PARK_XY_FEEDRATE 100          // X and Y axes feedrate in mm/s (also used for delta printers Z axis)
  #define PAUSE_PARK_Z_FEEDRATE 5             // Z axis feedrate in mm/s (not used for delta printers)
  #define PAUSE_PARK_RETRACT_FEEDRATE 60      // Initial retract feedrate in mm/s
  #define PAUSE_PARK_RETRACT_LENGTH 2         // Initial retract in mm
                                              // It is a short retract used immediately after print interrupt before move to filament exchange position
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE 10  // Unload filament feedrate in mm/s - filament unloading can be fast
  #define FILAMENT_CHANGE_UNLOAD_LENGTH 100   // Unload filament length from hotend in mm
                                              // Longer length for bowden printers to unload filament from whole bowden tube,
                                              // shorter length for printers without bowden to unload filament from extruder only,
                                              // 0 to disable unloading for manual unloading
  #define FILAMENT_CHANGE_LOAD_FEEDRATE 6     // Load filament feedrate in mm/s - filament loading into the bowden tube can be fast
  #define FILAMENT_CHANGE_LOAD_LENGTH 0       // Load filament length over hotend in mm
                                              // Longer length for bowden printers to fast load filament into whole bowden tube over the hotend,
                                              // Short or zero length for printers without bowden where loading is not used
  #define ADVANCED_PAUSE_EXTRUDE_FEEDRATE 3   // Extrude filament feedrate in mm/s - must be slower than load feedrate
  #define ADVANCED_PAUSE_EXTRUDE_LENGTH 50    // Extrude filament length in mm after filament is loaded over the hotend,
                                              // 0 to disable for manual extrusion
                                              // Filament can be extruded repeatedly from the filament exchange menu to fill the hotend,
                                              // or until outcoming filament color is not clear for filament color change
  #define PAUSE_PARK_NOZZLE_TIMEOUT 45        // Turn off nozzle if user doesn't change filament within this time limit in seconds
  #define FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS 5 // Number of alert beeps before printer goes quiet
  #define PAUSE_PARK_NO_STEPPER_TIMEOUT       // Enable to have stepper motors hold position during filament change
                                              // even if it takes longer than DEFAULT_STEPPER_DEACTIVE_TIME.
  #define PARK_HEAD_ON_PAUSE                  // Go to filament change position on pause, return to print position on resume
  //#define HOME_BEFORE_FILAMENT_CHANGE       // Ensure homing has been completed prior to parking for filament change
#endif
```

Radiu, 2017-10-23:
@Roxy-3D  wow that was a very fast responce with very detailed information
thanks alot mate
chears :)

Radiu, 2017-10-23:
by the way where you wire and define the end stop pins?
on a available free endstops pins on ramps 1.4?

closed 2017-10-23 by Radiu

Radiu, 2017-10-23:
ok i see is on pin 4

i guess the only thing that can do yet is detect if the hot end is clog?

Roxy-3D, 2017-10-23:
Marlin does not currently have support to detect hot end clogs...

AnHardt, 2017-10-23:
Some more thought about this from a thread about two yeas ago.
https://github.com/MarlinFirmware/Marlin/issues/1558

tareko, 2017-12-08:
Here is the relevant file from the Prusa firmware:

https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/fsensor.cpp

I have several genuine Prusa's and many self-built printers using Marlin. I think it would be wonderful to figure out the sensor and try to have the same features in the Marlin FW as well with some way to make the flow sensor as well. I think clog detection would be quite wonderful.

tarek : )

davidperrenoud, 2017-12-25:
The sensor is a PixArt PAT9125EL but it doesn't seem possible to buy just one piece:
http://www.pixart.com.tw/upload/PAT9125EL_GDS_V1.2_31052017_20171011150843.pdf

tareko, 2017-12-26:
I've been looking around for a similar product and don't know enough about this to find anything useful on mouser or digikey. Here are the schematics for this part from Prusa:

Roxy-3D, 2017-12-28:
Personally...   I think using the optical sensor of a mouse might yield good results.   If the mouse can give us 'movement pulses' we can probably tell when the filament isn't advancing enough.

At that point...   the firmware could be configured to do a number of things.   A user might want
- move the nozzle to a safe location
-- the printer to slow down all movement for a short period of time (or filament length) to see if the clog is temporary. 
-- increase the nozzle temperature to see if that can help clear the clog.
- park the nozzle and let the user manually clear the clog (and then re-prime the nozzle)

JerryDurand, 2017-12-28:
When I was thinking of doing this I was going to use an encoded ball
bearing pressing against the filament. 

Encoding done by marking some stripes on the side of it with paint or
ink and then bouncing a reflective sensor off it.  The marks don't need
to be consistent, you're just looking for movement.

I got the idea from a high-end trackball that had an engraved aluminum
disk glued to a ball bearing for the encoder.

tareko, 2017-12-28:
> Personally... I think using the optical sensor of a mouse might yield good results. If the mouse can give us 'movement pulses' we can probably tell when the filament isn't advancing enough.

I wonder if there can be a usable signal simply from making a holder involving the mouse. Obviously this would be an intermediary step until we figure out how to dismantle the mouse and get the sensor to be usable. Also, the filament detector can be mounted close to the spool, not necessarily on the extruder as with Prusa's version. It's possible that this should then be an Octoprint plugin rather than a Marlin component.

I found this paper that used X and Y coordinates: [Measuring Viscoelastic Deformation with an Optical Mouse](https://github.com/MarlinFirmware/Marlin/files/1591593/ng2004.pdf). Here is the relevant part from the methods section:

>The vertically mounted optical mouse is then secured on any table using a C-clamp. The connector of the optical mouse is plugged to the computer’s USB port. On the computer, a program that polls the x and y positions from the optical mouse is activated. 

There are [several](https://makezine.com/2007/12/15/using-an-optical-mouse-for-rob/) more articles on this topic, but none are particularly useful.

Here is [a tindie project](https://www.tindie.com/products/jkicklighter/adns-9800-laser-motion-sensor/) which [includes the open source schematics, firmware and software](https://github.com/mrjohnk/ADNS-9800) (hi @mrjohnk!) that may also be a great candidate to explore. I can order one and explore its use. @Roxy-3D : Happy to send one your way too if you want to PM / email me your mailing address (I'm at tarek@tarek.org).

Thank you,

tarek : )

tareko, 2017-12-28:
Here's another potentially useful project [showcased on hackaday](https://hackaday.com/2016/11/21/diy-optical-sensor-breakout-board-makes-diy-optical-mouse/). Here's the repository:

https://github.com/Tom101222/Adns-3050-Optical-Sensor (hi @tom101222!)

These seem to use ADNS 3050 chips which [digikey](https://www.digikey.com/products/en/sensors-transducers/optical-sensors-mouse/526?k=Adns%203050) and [mouser](https://ca.mouser.com/ProductDetail/Broadcom-Limited/ADNS-3050) shows as "obsolete".

This chip was suggested, though, which might be a drop-in replacement: https://www.digikey.com/product-detail/en/silicon-labs/SI1102-A-GMR/336-1819-1-ND/2172912

gavatron3000, 2018-01-19:
also this could be extended to do what we want...

http://www.instructables.com/id/Digital-3D-Printer-Filament-Counter-Use-PS2-Mouse/

Roxy-3D, 2018-01-19:
Yes.  That might be an easy way to proto-type the code.

pyr0ball, 2018-02-02:
There's a mostly printable version up on thingiverse that is usable with a smoothieboard: https://www.thingiverse.com/thing:2515750

A simple optical endstop, a hobbed bolt, a bearing, and a few other odd bits and pieces are used. Instead of using a full optical encoder like with a mouse, I believe this simply checks boolean (on/off) against a time table calculated from the extruder speed

bula87, 2018-02-12:
Personally I tried to use mouse sensor to detect filament movement and unfortunately, it doesn't work for me. Mouse sensor is just low-resolution CMOS camera. and filament with glossy finish doesn't provide enough differences during the move to determinate correct and constant movement detection. So I just drop this idea. Now when Josep released schematics of his sensor I guess in few months, on aliexpres will appear clones of his project.

pyr0ball, 2018-02-12:
What about using a wheel with differentiated patterning? Have the filament
get pulled through a mechanism that has a hobbed bolt and an idler on a
spring loaded armature keeping pressure on it, and have the differentiated
wheel on the same axel as the hobbed bolt. The sensor watches the wheel,
not the filament directly

gavatron3000, 2018-02-12:
The one in the link I shared uses the old PS2 mouse's with the old ball which has simple encoders to detect the ball movement. I think it would be quite reliable

pyr0ball, 2018-02-12:
Trouble with that setup is finding an old ps2 mouse lying around

gavatron3000, 2018-02-12:
Agreed

bula87, 2018-02-12:
My idea was to use magnetic rotary encoders (MLX90316 module / angle sensor) because the cheap, small, easy to install and has SPI but I cannot buy it in Spain and don't want to wait 4 weeks to get it from aliexpress, so I just installed switch which pause my printer when filament ends.

IMMRMKW, 2018-02-27:
In the piece of code in which the PAT9125 is initialized, there is a small line saying:

"Now continues the magic sequence from the PAT912EL Application Note: Firmware Guides for Tracking Optimization."

Does anybody know where I can obtain this Application Note?

pyr0ball, 2018-03-08:
I've obtained an HC-020K module for testing, if anyone might be able to provide a test sketch I could start poking around at making this work

thinkyhead, 2018-03-08:
@pyr0ball — Something like this? https://sites.google.com/site/myscratchbooks/home/projects/project-11-infrared-speed-sensing-module

bula87, 2018-03-12:
Does anyone get this module? I wrote even to pixart team (manufacturer) and they responded that they sell modules only through resellers. Afer mail to the reseller in Spain, I get the info that it is not possible to buy it in small quantity...  
So no idea where to buy it

alexcocan, 2018-03-14:
Hi everyone, I just got into 3Dprinting last fall. Started by making a MK2 clone. I got it to work better than i expected and I am slowly upgrading it since. 
My brother just bought an MK3 and now I'm excited about the safety features (missed step, power runout and filament runout/jam detection) of it. 
I'm good at electronics but not that good at coding especially when it comes to something like this. I've done some searching and I found the [APDS-9960](https://cdn.sparkfun.com/assets/learn_tutorials/3/2/1/Avago-APDS-9960-datasheet.pdf) gesture sensor from AVAGO. Could it be possible to replace the PAT9125EL sensor from the original design with it?

AcrimoniousMirth, 2018-11-30:
Slightly old thread but thought it was worth mentioning you can find clones of the Prusa sensor online now. For about £12. Just search eBay for “Prusa filament sensor”.

motymen, 2019-01-12:
as Acrimoniousmirth pointed above, you can buy an optical sensor below is one example
https://www.aliexpress.com/item/1Set-Prusa-i3-Laser-Sensor-3D-Printer-Filament-Sensor-Detect-Stuck-Filament-With-Cable-And-PINDA/32926072510.html

does marlin supports this as of now?

AnHardt, 2019-01-12:
No.
Made some initial test, but have the same difficulties, to reliably detect motion of a lot of filaments, as Prusa. The sensor seems to have a lot of adjustable parameters but hasn't any documentation about that - at least i could not find enough. So the sensor has to be used on a stainless steel surface - what it's made for. My software project is put aside until i constructed a sensor-holder, looking on a idler-baring, for my extruder - one day - maybe.

Ayunken, 2019-05-07:
Hi guys.
I have implemented the Prusa sensor on Marlin 2.0 through a small MCU that interprets the sensor reading and sends pulses as an encoder that Marlin accepts. All info at https://github.com/Ayunken/prusa_sensor_marlin2.0