MarlinFirmware/Marlin #8064: NEOPIXEL is jacked up

Tannoo, 2017-10-23:
### Bug Report
- Description: Neopixel color selection is unreliable.
- Expected behaviour: Reliable color selection.
- Actual behaviour: Unreliable color selection of anything other than WHITE.
- Steps to reproduce:
  - M150 R255
  - M150 U255
  - M150 R255
  - M150 B255
  - M150 U255
  - M150 R255

  Adding the P255 to each or any of the above commands doesn't seem to help.

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1406889/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1406890/Configuration_adv.txt)

Tannoo, 2017-10-23:
I hope some can tell me that I'm just messing it up.

fiveangle, 2017-10-24:
@SCiunczyk ?

SCiunczyk, 2017-10-24:
Hi,

Are you using the my version of leds.cpp and other files send to you last week? Without modification?
What type of neopixel strip do you have? I had similar effects when I was using incorrect type e.g. NEO_GRBW on GRB strip.

Tannoo, 2017-10-24:
I am using bugfix-2.0.x. I only modded the startup test function, to make it work correctly there. Nothing else.

As you can see in my configuration.h file, I have NEO_GRB and 45 LEDs in the strip.

It is not an rgbw strip.

Tannoo, 2017-10-24:
Upon booting up Marlin, I can send `M150 R255` and all leds turn on... but no always red. I was showing my wife this afternoon and it took 4 tries of the same command to light them up red.

1st attempt... blue.
2nd attempt... purple.
3rd attempt.... yellow.
4th attempt.... red

SCiunczyk, 2017-10-24:
But bugfix-2.0.x. without my files it's hard to compile ;-) So I assume that you include my files.

What kind of LED strip do you have ? Original Adfruit ? What chips WS2812B ?

I have WS2812B and it's working as expected with NEO_GRB parameter.

Tannoo, 2017-10-24:
There have been changes since we last spoke about this.

There were other changes made to rcbugfix-2.0.x. The changes were too much for what I wanted to deal with.
So, I started from a fresh download and it compiles just fine.

I do not have the strip from adafruit, but it is a WS2812B, yes.

SCiunczyk, 2017-10-24:
just checked - I still can't compile fresh fetched bugfix-2.0.x branch with NEOPIXEL (without my corrections). Can you point me to rcbugfix-2.0.x branch you mentioned ?

Tannoo, 2017-10-24:
http://github.com/Tannoo/Marlin/tree/My_AVR_Printer

SCiunczyk, 2017-10-24:
Strange. Looks exactly like my code. The only idea I have is to compile one of the Adfruit Neopixel examples and upload it to the printer just to check the LEDstrip driver itself.

Tannoo, 2017-10-24:
All 45 drivers are possibly bad?

SCiunczyk, 2017-10-25:
I don't think so, but it's worth to check.

Tannoo, 2017-10-26:
Well, I have found several strips doing the same thing. Then, I moved the data wire to an UNO with the same strand_test demo and did the same thing.

I then moved the power to the UNO and it works just fine. My 5v power supply (ATX) is not stable now.
A new 5v supply is in order now.

closed 2017-10-26 by Tannoo

SCiunczyk, 2017-10-26:
A good 5V always is a problem. That was why I designed my own 5V/3A StepDn converter. On my printer it's powering the Pi Board with OctoPrint and 43 chip Neopixel strip.
[image]

Tannoo, 2017-10-26:
Well, one would think that a 500W ATX power supply would be plenty stable. I am tossing in the trash.

fiveangle, 2017-10-26:
I've found them to be pretty horrible in general.  I've got a 550W Single-rail 80+Antec which claims good to 37A and even at 25A the 12v rail has dropped to 11.4V.  I can't imagine how bad the supply of a cheap ATX might be.

-=dave

SCiunczyk, 2017-10-26:
Me too, To avoid such problems finally in my printer I have 230V silicone heated bed with SSR so the 12V consumption is almost nothing like 3-5 Amps.

Slawek

Tannoo, 2017-10-27:
I currently have them powered by a lab bench supply set to 5.01v (capable of 60A max) and it's running just fine.