MarlinFirmware/Marlin #8065: HELP ! For devs !

studiodyne, 2017-10-23:
Hi , i have worked on a feature , to swap extruder on spool end
I use M600 'Thks to the writter of this feature '
and i have a bug on the end 

M600 --> go on pause --- Load/unloard --> Return to print , all is ok !
But
M600 -->go on pause --unLoad --Toolchange()--Load-- return = ok but after Print 3 or 4 lines and I have a big extrusion in the print but all continue normally

I think ' may be ', when go on pause , M600 don't wait to purge the planner , and have some lines already loaded ... And the tool change breaks something in the extrusion.
Is there someone that can help me to turn on m600 only after planner empty to test my code
Or may be , you have the solution

Thks

Roxy-3D, 2017-10-23:
When the filament sensor detects the filament has run out...   It gracefully allows the planner's buffers to empty before it initiates the filament change.    Try looking at that code.

studiodyne, 2017-10-23:
THks to answers 👍  , but after many days to make my extensions ' look my PR about M600 new features ', i never seen something like ' purge planer before m600'
The M600 is in a long list of switching routine  , same for all Gcode
Do you know a gcode function that need absolutely a purged planner , i will see how it works and i will try 
Do you know what name can have a purge function
Give me all that can help me to find a purging routine 
Thks

Roxy-3D, 2017-10-23:
Start by looking for any place in the code base that ADVANCED_PAUSE_FEATURE or FIL_RUNOUT_PIN causes special code to be generated.    You will need a tool such as Astro-Grep to search the whole code base.

Probably...   In the idle() routine there will be a check of that pin for that condition.    That will get you close.   It is important to learn how to find the relevant pieces of code.

studiodyne, 2017-10-23:
Roxy .... It's a JUNGLE lol , how many little humans can learn this jungle in a few minutes lol
I prefer someone here , write me a code that make this

M600 --- While Movesqueued==0 ----go to pause 
But , to make it , my knowledge of this jungle is so little  ...

Thanks for the keyword ' idle() , it's a good thing '

'i'll be back'   lol

studiodyne, 2017-10-23:
@thinkyhead @tannoo @Bob-the-Kuhn @tcm0116 , please can you take a few time to write me something to purge the planner command queue before to go on pause , it's too much for me 
thks

tcm0116, 2017-10-23:
Just call stepper.synchronize() and it will block until the queue is empty.

studiodyne, 2017-10-23:
Yes ... but stepper synchronise just finish the current block of a only one command cutted in 2 ms slices ... but not the queued commands in the buffer

tcm0116, 2017-10-23:
No. It waits for all queued blocks to finish:

```
/**
 * Block until all buffered steps are executed
 */
void Stepper::synchronize() { while (planner.blocks_queued()) idle(); }
```

studiodyne, 2017-10-23:
Impossible. ...you are wrong ...
This function can not made this . Because used everywhere. .. fr example in fwretract ...
If this empty the buffer command lines .. it means all fwretract breaks all smoothed movement. .
No it's just for the current movement

AnHardt, 2017-10-23:
> to purge the planner command queue before

> If this empty the buffer command lines

This is not the same. @tcm0116  exactly answer your question. Your question did not express what you were looking for. You are wrong!

Roxy-3D, 2017-10-23:
> Start by looking for any place in the code base that ADVANCED_PAUSE_FEATURE or FIL_RUNOUT_PIN causes special code to be generated. You will need a tool such as Astro-Grep to search the whole code base.

> Probably... In the idle() routine there will be a check of that pin for that condition. That will get you close. It is important to learn how to find the relevant pieces of code.

@studiodyne    Seriously...   follow this advice verbatim.   Get Astro-Grep.    And search for those two symbols in the code base.   This is the method I use when I need to find or figure out something.   The Marlin code base is huge.   You can't do the search manually.   And you can't possibly know everything in it.  You have to learn how to use the electronic tools to help you.

studiodyne, 2017-10-23:
Thk you for help , now i understand 
I'm sorry i'm wrong 
All segments that have direction change and without stopping points are bufferised , to smooth the travel , but , when a ' stop 'appears , all buffer are emptied , because a stop if a final point or a start point , and no need to have a full buffer to calculate a stop
It's ok
So , now , if my machine is ' idle() ' and after 'steppersynchro' , what can happened when i go out of pause
One or two lines are made perfectly with correct extrusion movements and just after 50mm of extrusion at high speed .... 
Not a problem of resuming position , because the return is perfect 
I don't know
@Roxy-3D I make it , i use my brain at max to understand this JUNGLE , planner hashing and more , but i don't want to learn all the code , I'm trying , to just understant the little thing for me , and it's why i don't make a total immersion , i just want to make the minimum job 
i will follow what you say

studiodyne, 2017-10-23:
May be someone here have time , to test the code in dryrun , to analyse and help me
Look the PR[ Automatic Spool Finishing ](https://github.com/MarlinFirmware/Marlin/pull/8067)

Spawn32, 2017-10-23:
@Roxy-3D 
Thank's for the tip about Astro-Grep, that is a really nice utility :)

Roxy-3D, 2017-10-24:
Yeah...   If you have something to start with...  you can find exactly what is affecting the stuff you care about....

fiveangle, 2017-10-24:
Nice @Roxy-3D !  My work covers a copy of _Source Insight_ for me, but _Astro-Grep_ looks like it gets you 60-70% of that functionality.  I'm still using actual grep on Mac OS X 😝 

One thing really nice about _Sublime Text_ is that search across all files in the project is a single command modifier to "Find" and automatically shows context and I can just double-click in the resulting found lists to jump to that file/section, so I rarely have to dig into speciallized grep commands anymore.

[image]

Roxy-3D, 2017-10-24:
> Nice @Roxy-3D ! My work covers a copy of Source Insight for me, but Astro-Grep looks like it gets you 60-70% of the searching/conext functionality. I'm still using actual grep on Mac OS X

Yeah!   And if you like (and understand) Astro Grep...  There is a preference setting to tell it how to jump to your favorite editor with the specified line as the starting edit location.   That is super helpful when you need to edit 37 different example/configuration.h files.   (Because you added a new #define option)

fiveangle, 2017-10-24:
> There is a preference setting to tell it how to jump to your favorite editor with the specified line as the starting edit location

That does sounds useful !

Yeah, I've moved to _Sublime Text_ about 5yr ago only so I can have the same editor on both Windows, Mac, and Linux.  I was a long time jEdit user but the slow glitchy QT interface just finally got to me.

> I was already in that boat when Atom got forced on me because that is how PlatformIO is implemented.

This is why I've been trying to get the dev of DevIoT plugin for Sublime Text to get it working correctly for PlatformIO development so I never have to leave my editor :smile: .  It's also super easy compared to Atom (which would be good for new users).  There's a few minor kinks to work out but once it's solid I'll write up a quick getting-started thing for people.  Gawd Atom is atrocious :)

studiodyne, 2017-10-24:
# WORKS !!!

Thks to all that contribute to help me 
My beautifull feature works fine now ,  I LOVE IT ! my old ended spool are automaticly finished without my intervention !

The fix :  All worked well , but i have forgotten a little detail , FWRetract status , because i use FwRetract , and if someone use it , he knows that we are forced to initialise all extruders at starting with a G10 S1 for all . And this is this detail that made the issue , after resuming , one or two lines and when the first G11 appears , the swap statut high of the extruder make this long extrusion
I have transmit these status to the extruder and now IT S BEAUTIFULL !!!!!
@Roxy-3D thks for your help and your patience with me 😄 , and for your editor , i prefer notepad++ because all in one and plugin compare 👍

closed 2017-10-24 by studiodyne