MarlinFirmware/Marlin #8079: Informational: script to calibrate k-factor

token47, 2017-10-25:
I just made a (bash) script to ease the generation of GCODE to calibrate the k-factor on the LIN_ADVANCE. I was inspired after @Sebastianv650 's script and I hope I have improved it somehow (it not just made ir more complicated :-P)

Any contributions are welcome. I was going to write it in another language (probably python) but thought it should be more accessible this way. If anyone has the time and skill it would be amazing to put it online on a kfactor-o-matic style to generate and download the result on demand.

https://github.com/token47/k-factor

fiveangle, 2017-10-25:
You could submit a PR to the project adding the script here:

https://github.com/MarlinFirmware/Marlin/tree/1.1.x/buildroot/share/scripts

-=dave

MasterPIC, 2017-10-27:
What is the EXTRUSION_FACTOR?

Sineos, 2017-10-27:
This script is excellent. Many thanks for sharing it.

Wouldn't it be an option to bake such a calibration pattern right into Marlin, i.e. analog to the UBL validation pattern?

token47, 2017-10-27:
I'm not sure how to do all the math to come up with the extrusion values for each distance, and I doubt it would be feasible to do it in bash anyway. So I took a sample and isolated the value in mm of filament to a 1mm linear for that case (0.4mm nozzle, 0.2mm layer height and 1.75mm filament). I believe that is the most used config. But I didn't want to make it hardcoded and left as a variable so you can change it if you want to calculate for other situations. Ideally I could have a bunch of values for the user to choose.

Sebastianv650, 2017-10-28:
Excellent, great idea to turn the pattern into a script 👍

Sineos, 2017-10-30:
Here is the script a html variant:

https://github.com/Sineos/k-factorjs

All credits go to @Sebastianv650 and @token47

lrpirlet, 2017-10-31:
@token47 This script seems very nice... 

However, I have a 3mm filament and need to set the EXTRUSION_FACTOR? What is the relation between nozzle size, the layer height and the diameter of the filament...

The script says : 
```EXTRUSION_FACTOR="0.037418" # calculated for 0.4mm nozzle, 0.2mm layer height and 1.75mm filament```

Thanks in advance.

Sebastianv650, 2017-10-31:
It's the relation between mm movement of the nozzle and mm filament extruded to give the wanted line width. Seems like @token47 was assuming 0.45mm line width:
0.45mm * 0.2mm / ((1.75mm / 2)^2 * PI) = 0.037418

If you keep 0.2mm layer height and 0.45mm line width but with 3mm filament, the new factor would be 0.012732.

token47, 2017-10-31:
I took a sample GCODE and isolated the value, didn't really know how to do the math. Thank you for the explanation.

Sineos, 2017-10-31:
@Sebastianv650 
Would it be safe to assume:
`Nozzle Diameter == Line Width`
If so, we could easily add Filament and Nozzle Size to the script and go from there with the math above.

Sebastianv650, 2017-10-31:
Usually you want to go a bit higher. 1.05x nozzle diameter is the smallest I use for a clean extrusion. I would take 1.2x nozzle diameter or 1.1x.

Sineos, 2017-10-31:
Thanks @Sebastianv650 

Added filament and nozzle size to the html variant.

lrpirlet, 2017-11-01:
@token47 
Hello, not sure how I can submit a PR against your repository...

So, I'll submit the whole BASH procedure here bellow... Note that this is able to run in the "git bash" screen in Windows... so this is a quick and easy way to generate a code either from linux or from windows (I have no clear idea for mac world)

I also put a warning about heating the printer before issuing G28... I know at least one machine (myne) that would be damaged if G28 is issued while bed and/or hot-end is over plastic softening point...

I realy think this process should find it's way into the Marlin share scripts... 

```
#!/bin/bash

# Test generator for K-Factor calibration
# Version 0.9 by Andre Ruiz (aka Token47), andre (dot) ruiz (at) gmail (dot) com

# To Do:
# put a list of the values used on the begining of the gcode as comments on simplify3d style

set -u

# Parameter list, edit to your need

NOZZLE_TEMP="205" # C degrees
BED_TEMP="60" # C degrees
SLOW_SPEED="1200" # mm/min
FAST_SPEED="4200" # mm/min
MOVE_SPEED="7200" # mm/min
USE_UBL="0" # Set to 1 to enable bed levelling
RETRACTION="1.000" # mm
BEDSIZE_X="200" # mm
BEDSIZE_Y="200" # mm
LAYER_HEIGHT="0.200" # mm
NOZZLE_SIZE="0.4" # mm
NOZZLE_LINE_RATIO="1.2" # ratio between nozzle diameter and line width
FILAMENT_SIZE="1.75" # mm
EXTRUSION_MULT="1.1" # arbitraty extrusion multiplier, should be 1.0 normally

# End of parameter list

cat <<EOF
Value used to generate the test Gcode

NOZZLE_TEMP="205" # C degrees
BED_TEMP="60" # C degrees
SLOW_SPEED="1200" # mm/min
FAST_SPEED="4200" # mm/min
MOVE_SPEED="7200" # mm/min
USE_UBL="0" # Set to 1 to enable bed levelling
RETRACTION="1.000" # mm
BEDSIZE_X="200" # mm
BEDSIZE_Y="200" # mm
LAYER_HEIGHT="0.200" # mm
NOZZLE_SIZE="0.4" # mm
NOZZLE_LINE_RATIO="1.2" # ratio between nozzle diameter and line width
FILAMENT_SIZE="1.75" # mm
EXTRUSION_MULT="1.0" # arbitraty extrusion multiplier, should be 1.0 normally

EOF

echo "If those parameters do fit your need then type return"
echo "  else kill the procedure and edit to your need"
echo
echo "Caution, the generated Gcode program issues a G28 AFTER heating bed and hot-end"
read GO_ON

declare "EXTRUSION_FACTOR"="$(awk "BEGIN {
    printf \"%f\", ($LAYER_HEIGHT * $NOZZLE_SIZE * $NOZZLE_LINE_RATIO) /(3.14159 * ($FILAMENT_SIZE/2) * ($FILAMENT_SIZE/2))
    }")"

for a in 5 20 40; do
		declare "EXTRUSION_${a}mm"="$(awk "BEGIN {
				printf \"%f\", $EXTRUSION_FACTOR * ${a} * $EXTRUSION_MULT
		}")"
done

PRIME_Y1=$(awk -v var="$BEDSIZE_Y" 'BEGIN {print var - 162.5}')
PRIME_Y2=$(awk -v var="$BEDSIZE_Y" 'BEGIN {print var - 62.5}')
START_X=$(awk -v var="$BEDSIZE_X" 'BEGIN {print (var - 80)/2}')
START_Y=$(awk -v var="$BEDSIZE_Y" 'BEGIN {print (var - 125)/2}')

cat <<EOF
; K-FACTOR TEST
;
M190 S${BED_TEMP} ; set and wait for bed temp
M104 S${NOZZLE_TEMP} ; set nozzle temp and continue
G28 ; home all axis
EOF

if [ "$USE_UBL" = "1" ] ; then
	echo "G29 ; execute bed automatic leveling compensation"
fi

cat <<EOF
M109 S${NOZZLE_TEMP} ; block waiting for nozzle temp
G21 ; set units to millimeters
M204 S500 ; lower acceleration to 500mm/s2 during the test
M83 ; use relative distances for extrusion
G90 ; use absolute coordinates
;
; go to layer height and prime nozzle on a line to the left
G1 X20 Y${PRIME_Y1} F${MOVE_SPEED}
G1 Z${LAYER_HEIGHT} F${SLOW_SPEED}
G1 X20 Y${PRIME_Y2} E10 F${SLOW_SPEED} ; extrude some to start clean
G1 E-${RETRACTION}

;
; start the test (all values are relative coordinates)
;
G1 X${START_X} Y${START_Y} F${MOVE_SPEED} ; move to pattern start
G91 ; use relative coordinates
EOF

for K in 0 5 10 15 20 25 30 35 40 45 50 55 60 65 70 75 80 85 90 95 100; do
		cat << EOF
M900 K${K}
G1 E${RETRACTION}
G1 X20 Y0 E${EXTRUSION_20mm} F${SLOW_SPEED}
G1 X40 Y0 E${EXTRUSION_40mm} F${FAST_SPEED}
G1 X20 Y0 E${EXTRUSION_20mm} F${SLOW_SPEED}
G1 E-${RETRACTION}
G1 X-80 Y5 F${MOVE_SPEED}
EOF
		done

cat << EOF
;
; mark the test area for reference
;
G1 X20 Y0 F${MOVE_SPEED}
G1 E${RETRACTION}
G1 X0 Y20 E${EXTRUSION_20mm} F${SLOW_SPEED}
G1 E-${RETRACTION}
G1 X40 Y-20 F${MOVE_SPEED}
G1 E${RETRACTION}
G1 X0 Y20 E${EXTRUSION_20mm} F${SLOW_SPEED}
G1 E-${RETRACTION}
;
; finish
;
G4 ; wait
M104 S0 ; turn off hotend
M140 S0 ; turn off bed
G90 ; use absolute coordinates
G1 Z30 Y200 F${MOVE_SPEED}; move away from the print
M84 ; disable motors
M502 ; resets parameters from ROM (for those who do not have an EEPROM)
M501 ; resets parameters from EEPROM (preferably)
;
EOF
```

lrpirlet, 2017-11-01:
Please disregard previous previous procedure... I copied the wrong one... (deleted by now)

Here is the correct one.
```
#!/bin/bash

# Test generator for K-Factor calibration
# Version 0.9 by Andre Ruiz (aka Token47), andre (dot) ruiz (at) gmail (dot) com

# To Do:
# put a list of the values used on the begining of the gcode as comments on simplify3d style

set -u

# Parameter list, edit to your need

NOZZLE_TEMP="205" # C degrees
BED_TEMP="60" # C degrees
SLOW_SPEED="1200" # mm/min
FAST_SPEED="4200" # mm/min
MOVE_SPEED="7200" # mm/min
USE_UBL="0" # Set to 1 to enable bed levelling
RETRACTION="1.000" # mm
BEDSIZE_X="200" # mm
BEDSIZE_Y="200" # mm
LAYER_HEIGHT="0.200" # mm
NOZZLE_SIZE="0.4" # mm
NOZZLE_LINE_RATIO="1.2" # ratio between nozzle diameter and line width
FILAMENT_SIZE="1.75" # mm
EXTRUSION_MULT="1.1" # arbitraty extrusion multiplier, should be 1.0 normally

# End of parameter list

cat <<EOF
Value used to generate the test Gcode

NOZZLE_TEMP=$NOZZLE_TEMP C degrees
BED_TEMP=$BED_TEMP C degrees
SLOW_SPEED=$SLOW_SPEED mm/min
FAST_SPEED=$FAST_SPEED mm/min
MOVE_SPEED=$MOVE_SPEED mm/min
USE_UBL=$USE_UBL Set to 1 to enable bed levelling
RETRACTION="$RETRACTION mm
BEDSIZE_X=$BEDSIZE_X mm
BEDSIZE_Y=BEDSIZE_Y mm
LAYER_HEIGHT=$LAYER_HEIGHT mm
NOZZLE_SIZE=$NOZZLE_SIZE mm
NOZZLE_LINE_RATIO=$NOZZLE_LINE_RATIO ratio between nozzle diameter and line width
FILAMENT_SIZE=$FILAMENT_SIZE mm
EXTRUSION_MULT=$EXTRUSION_MULT arbitraty extrusion multiplier, should be 1.0 normally

EOF

echo "If those parameters do fit your need then type return"
echo "  else kill the procedure and edit to your need"
echo
echo "Caution, the generated Gcode program issues a G28 AFTER heating bed and hot-end"
read GO_ON

declare "EXTRUSION_FACTOR"="$(awk "BEGIN {
    printf \"%f\", ($LAYER_HEIGHT * $NOZZLE_SIZE * $NOZZLE_LINE_RATIO) /(3.14159 * ($FILAMENT_SIZE/2) * ($FILAMENT_SIZE/2))
    }")"

for a in 5 20 40; do
		declare "EXTRUSION_${a}mm"="$(awk "BEGIN {
				printf \"%f\", $EXTRUSION_FACTOR * ${a} * $EXTRUSION_MULT
		}")"
done

PRIME_Y1=$(awk -v var="$BEDSIZE_Y" 'BEGIN {print var - 162.5}')
PRIME_Y2=$(awk -v var="$BEDSIZE_Y" 'BEGIN {print var - 62.5}')
START_X=$(awk -v var="$BEDSIZE_X" 'BEGIN {print (var - 80)/2}')
START_Y=$(awk -v var="$BEDSIZE_Y" 'BEGIN {print (var - 125)/2}')

cat <<EOF
; K-FACTOR TEST
;
M190 S${BED_TEMP} ; set and wait for bed temp
M104 S${NOZZLE_TEMP} ; set nozzle temp and continue
G28 ; home all axis
EOF

if [ "$USE_UBL" = "1" ] ; then
	echo "G29 ; execute bed automatic leveling compensation"
fi

cat <<EOF
M109 S${NOZZLE_TEMP} ; block waiting for nozzle temp
G21 ; set units to millimeters
M204 S500 ; lower acceleration to 500mm/s2 during the test
M83 ; use relative distances for extrusion
G90 ; use absolute coordinates
;
; go to layer height and prime nozzle on a line to the left
G1 X20 Y${PRIME_Y1} F${MOVE_SPEED}
G1 Z${LAYER_HEIGHT} F${SLOW_SPEED}
G1 X20 Y${PRIME_Y2} E10 F${SLOW_SPEED} ; extrude some to start clean
G1 E-${RETRACTION}

;
; start the test (all values are relative coordinates)
;
G1 X${START_X} Y${START_Y} F${MOVE_SPEED} ; move to pattern start
G91 ; use relative coordinates
EOF

for K in 0 5 10 15 20 25 30 35 40 45 50 55 60 65 70 75 80 85 90 95 100; do
		cat << EOF
M900 K${K}
G1 E${RETRACTION}
G1 X20 Y0 E${EXTRUSION_20mm} F${SLOW_SPEED}
G1 X40 Y0 E${EXTRUSION_40mm} F${FAST_SPEED}
G1 X20 Y0 E${EXTRUSION_20mm} F${SLOW_SPEED}
G1 E-${RETRACTION}
G1 X-80 Y5 F${MOVE_SPEED}
EOF
		done

cat << EOF
;
; mark the test area for reference
;
G1 X20 Y0 F${MOVE_SPEED}
G1 E${RETRACTION}
G1 X0 Y20 E${EXTRUSION_20mm} F${SLOW_SPEED}
G1 E-${RETRACTION}
G1 X40 Y-20 F${MOVE_SPEED}
G1 E${RETRACTION}
G1 X0 Y20 E${EXTRUSION_20mm} F${SLOW_SPEED}
G1 E-${RETRACTION}
;
; finish
;
G4 ; wait
M104 S0 ; turn off hotend
M140 S0 ; turn off bed
G90 ; use absolute coordinates
G1 Z30 Y200 F${MOVE_SPEED}; move away from the print
M84 ; disable motors
M502 ; resets parameters from ROM (for those who do not have an EEPROM)
M501 ; resets parameters from EEPROM (preferably)
;
EOF
```

thinkyhead, 2018-02-09:
> I doubt it would be feasible to do it in bash anyway

Our preference is Javascript, so that we can provide a web page at marlinfw.org for easy access.

Sineos, 2018-02-09:
This one can be closed --> http://marlinfw.org/tools/lin_advance/k-factor.html

closed 2018-02-17 by thinkyhead