MarlinFirmware/Marlin #8080: ### Bug Report #define HOTEND_OFFSET not working

bruce356, 2017-10-25:
I am using Marlin v1.1.5
https://github.com/MarlinFirmware/Marlin/releases/tag/1.1.5
Dual X carriage Cartesian printer with RUMBA board
Trying to change / adjust the tool1 offset
#define HOTEND_OFFSET_X {0.0, -0.24}
#define HOTEND_OFFSET_Y {0.0, 0.18}  
Not sure why Z axis is not included in same location in Marlin.

Changing the numbers in firmware does not seem to make any difference to the tool1 offset.
I was able to once affect the Y axis offset but not X axis.
Maybe I am using the wrong #Defines for a dual X carriage printer?
I can not find these offset settings in the GLCD menus, would be handy to be able to adjust these in eeprom via GLCD if they work.

[Marlin_1.1.5.zip](https://github.com/MarlinFirmware/Marlin/files/1413238/Marlin_1.1.5.zip)

fiveangle, 2017-10-25:
1.1.6 is latest release: https://github.com/MarlinFirmware/Marlin/releases

If it does not resolve, please test latest bugfix and if still an issue, please provide details showing the resulting problem.

However, your X2_MIN_POS is set to 80, so you are stating that your offset for X2 is 80mm but you are still able to move X2 to X < 80 when X1 is parked ?

```

// Enable this for dual x-carriage printers.
// A dual x-carriage design has the advantage that the inactive extruder can be parked which
// prevents hot-end ooze contaminating the print. It also reduces the weight of each x-carriage
// allowing faster printing speeds. Connect your X2 stepper to the first unused E plug.
//#define DUAL_X_CARRIAGE
#if ENABLED(DUAL_X_CARRIAGE)
  // Configuration for second X-carriage
  // Note: the first x-carriage is defined as the x-carriage which homes to the minimum endstop;
  // the second x-carriage always homes to the maximum endstop.
  #define X2_MIN_POS 80     // set minimum to ensure second x-carriage doesn't hit the parked first X-carriage
  #define X2_MAX_POS 353    // set maximum to the distance between toolheads when both heads are homed
  #define X2_HOME_DIR 1     // the second X-carriage always homes to the maximum endstop position
  #define X2_HOME_POS X2_MAX_POS // default home position is the maximum carriage position
      // However: In this mode the HOTEND_OFFSET_X value for the second extruder provides a software
      // override for X2_HOME_POS. This also allow recalibration of the distance between the two endstops
      // without modifying the firmware (through the "M218 T1 X???" command).
      // Remember: you should set the second extruder x-offset to 0 in your slicer.

  // There are a few selectable movement modes for dual x-carriages using M605 S<mode>
  //    Mode 0 (DXC_FULL_CONTROL_MODE): Full control. The slicer has full control over both x-carriages and can achieve optimal travel results
  //                                    as long as it supports dual x-carriages. (M605 S0)
  //    Mode 1 (DXC_AUTO_PARK_MODE)   : Auto-park mode. The firmware will automatically park and unpark the x-carriages on tool changes so
  //                                    that additional slicer support is not required. (M605 S1)
  //    Mode 2 (DXC_DUPLICATION_MODE) : Duplication mode. The firmware will transparently make the second x-carriage and extruder copy all
  //                                    actions of the first x-carriage. This allows the printer to print 2 arbitrary items at
  //                                    once. (2nd extruder x offset and temp offset are set using: M605 S2 [Xnnn] [Rmmm])

  // This is the default power-up mode which can be later using M605.
  #define DEFAULT_DUAL_X_CARRIAGE_MODE DXC_FULL_CONTROL_MODE

  // Default settings in "Auto-park Mode"
  #define TOOLCHANGE_PARK_ZLIFT   0.2      // the distance to raise Z axis when parking an extruder
  #define TOOLCHANGE_UNPARK_ZLIFT 1        // the distance to raise Z axis when unparking an extruder

  // Default x offset in duplication mode (typically set to half print bed width)
  #define DEFAULT_DUPLICATION_X_OFFSET 100

#endif // DUAL_X_CARRIAGE
```

bruce356, 2017-10-25:
@fiveangle, thanks for the reply, the config you have posted above is not mine?
Mine is below:-

// Enable this for dual x-carriage printers.
// A dual x-carriage design has the advantage that the inactive extruder can be parked which
// prevents hot-end ooze contaminating the print. It also reduces the weight of each x-carriage
// allowing faster printing speeds. Connect your X2 stepper to the first unused E plug.
#define DUAL_X_CARRIAGE
#if ENABLED(DUAL_X_CARRIAGE)
  // Configuration for second X-carriage
  // Note: the first x-carriage is defined as the x-carriage which homes to the minimum endstop;
  // the second x-carriage always homes to the maximum endstop.
  #define X2_MIN_POS 0      // set minimum to ensure second x-carriage doesn't hit the parked first X-carriage
  #define X2_MAX_POS 291    // set maximum to the distance between toolheads when both heads are homed
  #define X2_HOME_DIR 1     // the second X-carriage always homes to the maximum endstop position
  #define X2_HOME_POS X2_MAX_POS // default home position is the maximum carriage position
      // However: In this mode the HOTEND_OFFSET_X value for the second extruder provides a software
      // override for X2_HOME_POS. This also allow recalibration of the distance between the two endstops
      // without modifying the firmware (through the "M218 T1 X???" command).
      // Remember: you should set the second extruder x-offset to 0 in your slicer.

  // There are a few selectable movement modes for dual x-carriages using M605 S<mode>
  //    Mode 0 (DXC_FULL_CONTROL_MODE): Full control. The slicer has full control over both x-carriages and can achieve optimal travel results
  //                                    as long as it supports dual x-carriages. (M605 S0)
  //    Mode 1 (DXC_AUTO_PARK_MODE)   : Auto-park mode. The firmware will automatically park and unpark the x-carriages on tool changes so
  //                                    that additional slicer support is not required. (M605 S1)
  //    Mode 2 (DXC_DUPLICATION_MODE) : Duplication mode. The firmware will transparently make the second x-carriage and extruder copy all
  //                                    actions of the first x-carriage. This allows the printer to print 2 arbitrary items at
  //                                    once. (2nd extruder x offset and temp offset are set using: M605 S2 [Xnnn] [Rmmm])

  // This is the default power-up mode which can be later changed using M605.
  #define DEFAULT_DUAL_X_CARRIAGE_MODE DXC_FULL_CONTROL_MODE

  // Default settings in "Auto-park Mode"
  #define TOOLCHANGE_PARK_ZLIFT   0.25      // the distance to raise Z axis when parking an extruder
  #define TOOLCHANGE_UNPARK_ZLIFT 1        // the distance to raise Z axis when unparking an extruder

  // Default x offset in duplication mode (typically set to half print bed width)
  #define DEFAULT_DUPLICATION_X_OFFSET 125

#endif // DUAL_X_CARRIAGE

Regards

fiveangle, 2017-10-25:
Ah, sorry...

So you are stating that you park your X1 carriage at -41, and you are able to make X2 travel to X < 0 ?

```
#define X_MIN_POS -41
#define X2_MIN_POS 0 // set minimum to ensure second x-carriage doesn't hit the parked first X-carriage
```

-=dave

bruce356, 2017-10-25:
X1 parks at -41 off the edge of the bed which is 250 wide and X2 parks at 291 also off the edge of the bed by the same amount (41mm), I assume those figures as entered are ok. I run the printer in full control mode and use a tool change script in Simplify3D. I also use S3D to set the T1 offsets and that works perfectly, but the Marlin firmware offsets do not, I do not know the reason as to why.
thanks

bruce356, 2017-10-25:
Sorry just tested to answer your question No X2 will not travel to <0 it stops at 0 and will not go any further.
regards - bruce

bruce356, 2017-10-25:
Just for your information the tool change script:-
;Start Tool Change
{IF NEWTOOL=0} T1; select Tool-1
{IF NEWTOOL=0} G1 X292 F10000 ; move T1 into dock
{IF NEWTOOL=0} T0; Left Extruder is active
{IF NEWTOOL=1} T0; select Tool-0
{IF NEWTOOL=1} G1 X-42 F10000 ; move T0 into dock 
{IF NEWTOOL=1} T1; Right Extruder is active
;End tool change script

fiveangle, 2017-10-25:
That doesn't tell us anything about what's not working.

How can you have X1 carriage extruder at 0.0mm and X2 offset extruder at -0.24mm ?  What are you expecting ?

btw- I agree it is confusing having these in Configuration.h and the other parameters in Configruation_adv.h, as the offsets in Configuration.h are not usable w/o configuration in Configuration_adv.h.  Once we figure out what your issue is, we may submit to change that.

bruce356, 2017-10-25:
Hi, it states in the Configuration.h file that X1 offset must remain set at 0 and offsets must apply to X2 only (not sure if this is meant for normal dual extruders or for dual X carriage or both).
Is there another way to adjust the X2 offset? Fine tuning and changing the X axis endstop positioning would be difficult and then there is Y Axis.

If I send X1 to a position on the bed for printing and then send X2 to the same position there is a slight mechanical misalignment between the two tools (X -0.24, Y 0.18), I would like to adjust this mechanical misalignment  using offsets Marlin makes provision for this in firmwar.  I am not sure if the offset is meant to be implemented for only single carriage dual extruder printers or dual X carriage or both (sorry repeating myself).

I am expecting to be able to adjust the mechanical misalignment between the tools T0 & T1 (no printer is mechanically perfect) in the X & Y and Z axis using the firmware offset settings which at the moment do not work on my printer (Z is not a problem for me as I have levelled both tools almost perfectly within 0.0005 inches). 
thanks

fiveangle, 2017-10-26:
http://marlinfw.org/docs/gcode/M218.html

bruce356, 2017-10-26:
@fiveangle, hi, I previously tried M218 T1 X1 Y1 Z0 and just again now (turning off the S3D Offsets) and found as before that the Y axis does respond to the M218 command offsetting T1 by 1mm, but there is no change or response by the X axis (no shift by 1mm of the X axis).

Are these offset settings saved to eeprom? and if so why can I not access these settings through GLCD menu, this would make changing the settings (once working) so very simple.

thanks again for your response.

thinkyhead, 2017-10-31:
> there is no change or response by the X axis (no shift by 1mm of T1 on X axis)

If you do this…

```gcode
M218 T1 X20
M503 S0
```
Does the `M503 S0` output show `M218 X20 ...`?

> why can I not access these settings through GLCD menu

Because no code exists in the LCD menu to edit them. Someone will have to add it. So far this hasn't been a high priority, as most probes stay in the same place once installed.

boelle, 2019-02-19:
@bruce356 have you tried latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

UticaTechClub, 2019-03-14:
@bruce356 @fiveangle @boelle @mikenz @boelle @thinkyhead 
Hi,
I found this thread while searching for answers on multi-extruder offsets issue (see my comments at https://github.com/MarlinFirmware/Marlin/pull/11623)
I am on cutting edge 2.0 code and it is either, I am doing something completely wrong or M218 simply doesn't work.

My baseline:
```
M851 Z0.6
M218 T0 X0.0 Y0.0 Z0.0
M218 T1 X-2.0 Y-2.0 Z-2.0
G28
T1
T0
T1
T0
...
```
Result:
Second toolhead position is not altered by `X-2.0 Y-2.0 Z-2.0` offsets.
I went through relevant source code and debug output and witnessed offsets properly applied to `current_position` global variable, but lost when tool head is moved to `destination` global variable. I cannot make any conclusion base on my limited understanding of Marlin implementation approaches.

- Did any of you ever witnessed M218 working on Marlin?
- Is it meant to be run in a specific sequence with other commands in order to take effect?
- Is it possibly a command scope issue?

@thinkyhead appears to be very busy, so maybe someone else could elaborate on this.

Thank you!

Phr3d13, 2019-06-10:
@thinkyhead Does this need a re-visit, or a close?

boelle, 2019-06-22:
missing label

InsanityAutomation, 2019-06-22:
@UticaTechClub They are applied on tool change or homing only. Regardless there have been several move fixes and one more open PR that should completely cover this at this point. Id say this can get closed.

UticaTechClub, 2019-06-24:
@thinkyhead @Phr3d13 @boelle @InsanityAutomation 
I think I experienced this issue only because of E3D like tool changer code. 

I got it to work about thee weeks ago on cutting edge Marlin, but I had to change some code in that tool changer section.

I will try to do the PR soon.

UticaTechClub, 2019-06-24:
#if ENABLED(SWITCHING_TOOLHEAD)

  inline void switching_toolhead_tool_change(const uint8_t tmp_extruder, const float fr_mm_s/*=0.0*/, bool no_move/*=false*/) {
    if (no_move) return;

    constexpr uint16_t angles[2] = SWITCHING_TOOLHEAD_SERVO_ANGLES;

    const float toolheadposx[] = SWITCHING_TOOLHEAD_X_POS,
                placexpos = toolheadposx[active_extruder] + hotend_offset[X_AXIS][active_extruder],
                grabxpos = toolheadposx[tmp_extruder] + hotend_offset[X_AXIS][active_extruder];
                
    const float toolhead_ypos = SWITCHING_TOOLHEAD_Y_POS + hotend_offset[Y_AXIS][active_extruder];

    /**
     * 1. Move to switch position of current toolhead
     * 2. Unlock tool and drop it in the dock
     * 3. Move to the new toolhead
     * 4. Grab and lock the new toolhead
     */

    // 1. Move to switch position of current toolhead

    if (DEBUGGING(LEVELING)) DEBUG_POS("Starting Toolhead change", current_position);

    current_position[X_AXIS] = placexpos;

    if (DEBUGGING(LEVELING)) {
      DEBUG_ECHOLNPAIR("(1) Place old tool ", int(active_extruder));
      DEBUG_POS("Move X SwitchPos", current_position);
    }

    fast_line_to_current(X_AXIS);
    planner.synchronize();

    current_position[Y_AXIS] = toolhead_ypos - SWITCHING_TOOLHEAD_Y_SECURITY;

    if (DEBUGGING(LEVELING)) DEBUG_POS("Move Y SwitchPos + Security", current_position);

    fast_line_to_current(Y_AXIS);
    planner.synchronize();

    // 2. Unlock tool and drop it in the dock

    if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("(2) Unlock and Place Toolhead");

    MOVE_SERVO(SWITCHING_TOOLHEAD_SERVO_NR, angles[1]);
    safe_delay(500);

    current_position[Y_AXIS] = toolhead_ypos;

    if (DEBUGGING(LEVELING)) DEBUG_POS("Move Y SwitchPos", current_position);

    planner.buffer_line(current_position,(planner.settings.max_feedrate_mm_s[Y_AXIS] * 0.2), active_extruder);
    planner.synchronize();
    safe_delay(200);
    current_position[Y_AXIS] -= SWITCHING_TOOLHEAD_Y_CLEAR;

    if (DEBUGGING(LEVELING)) DEBUG_POS("Move back Y clear", current_position);

    fast_line_to_current(Y_AXIS); // move away from docked toolhead
    planner.synchronize();

    // 3. Move to the new toolhead

    if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("(3) Move to new toolhead position");

    current_position[X_AXIS] = grabxpos;

    if (DEBUGGING(LEVELING)) DEBUG_POS("Move to new toolhead X", current_position);

    fast_line_to_current(X_AXIS);
    planner.synchronize();
    current_position[Y_AXIS] = toolhead_ypos - SWITCHING_TOOLHEAD_Y_SECURITY;

    if (DEBUGGING(LEVELING)) DEBUG_POS("Move Y SwitchPos + Security", current_position);

    fast_line_to_current(Y_AXIS);
    planner.synchronize();

    // 4. Grab and lock the new toolhead

    if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("(4) Grab and lock new toolhead ");

    current_position[Y_AXIS] = toolhead_ypos;

    if (DEBUGGING(LEVELING)) DEBUG_POS("Move Y SwitchPos", current_position);

    planner.buffer_line(current_position, planner.settings.max_feedrate_mm_s[Y_AXIS] * 0.2, active_extruder);
    planner.synchronize();

    safe_delay(200);
    MOVE_SERVO(SWITCHING_TOOLHEAD_SERVO_NR, angles[0]);
    safe_delay(500);

    current_position[Y_AXIS] -= SWITCHING_TOOLHEAD_Y_CLEAR;

    if (DEBUGGING(LEVELING)) DEBUG_POS("Move back Y clear", current_position);

    fast_line_to_current(Y_AXIS); // move away from docked toolhead
    planner.synchronize();

    if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("Toolhead change done.");
  }

#endif // SWITCHING_TOOLHEAD

UticaTechClub, 2019-06-24:
This is updated code.

boelle, 2019-08-28:
@UticaTechClub did you make an PR? just so we can pair the PR with this issue

boelle, 2019-11-03:
@bruce356 many updates have passed, is this still an issue on latest 2.0.x?

boelle, 2019-11-24:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-12-02:
closing this one assume the lack of response from @bruce356 means he have figured out his problem

can reopen if it turns out he has not

closed 2019-12-02 by boelle