MarlinFirmware/Marlin #8081: Bug Report - PID Hotend Not Holding Stable Temps

houseofbugs, 2017-10-25:
I have been running Marlin 1.1.5 on my machines for a while now and recently updated to 1.1.6. I have noticed that on ALL of them the temps are NOT being regulated in a very stable manner. On 1.1.5 they were within 1 degree C or less of the target temp. Now it is varying over 3C.

Any ideas?

[image]

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1413629/Marlin.zip)

jmdearras, 2017-10-25:
I have noticed reduced stability, but on a smaller scale.

fiveangle, 2017-10-25:
Can you provide same graph and terminal output during standard PID tune process ?

houseofbugs, 2017-10-26:
I will in a few days. I am running prints on all our machines right now and rolled back to 1.1.5 for now.

ruggb, 2017-10-27:
if you are running fan supply lines near thermo sense lines you will get something like this. Fans are very noisy electrically. Been there, done that in a very old issue.

thinkyhead, 2017-10-31:
Do you do your PID tuning with the extruder fan on? Do you do your PID tuning at typical print temperature? Do you do your PID tuning while the bed is warm or cold? Recommendation is to do your PID tuning with the bed and fan turned on if that's your typical usage.

houseofbugs, 2017-11-01:
@thinkyhead 

Extruder fan 100%, 250C (I print from 220c-260C). Bed warm (60C, before PID Marlin turns it off during PID).

houseofbugs, 2017-11-01:
@fiveangle 

[image]

fiveangle, 2017-11-01:
> Can you provide same graph and terminal output during standard PID tune process ?

houseofbugs, 2017-11-03:
It was late and I missed that part. I will post it when the prints are done. We use the machines for production.

fiveangle, 2017-11-03:
I took a closer look at the code and couldn’t see any obvious change that could result in this so I should tell you now that it probably won’t be useful without the terminal output from pid tune of the working version and the non-working version.

houseofbugs, 2017-11-03:
I am re-running the PID on 1.1.6 right now. Then I'll go back to 1.1.5 and do it again. There was mention in the 1.1.6 changelog that PID tuning was changed.

houseofbugs, 2017-11-03:
`Send: M303 C10 E0 S250
Recv: PID Autotune start
Recv:  T:24.12 /0.00 B:24.12 /0.00 @:127 B@:0
Recv:  T:24.97 /0.00 B:24.47 /0.00 @:127 B@:0
Recv:  T:28.03 /0.00 B:24.09 /0.00 @:127 B@:0
Recv:  T:31.62 /0.00 B:24.09 /0.00 @:127 B@:0
Recv:  T:34.87 /0.00 B:24.00 /0.00 @:127 B@:0
Recv:  T:39.03 /0.00 B:24.09 /0.00 @:127 B@:0
Recv:  T:42.92 /0.00 B:24.06 /0.00 @:127 B@:0
Recv:  T:47.31 /0.00 B:24.00 /0.00 @:127 B@:0
Recv:  T:51.37 /0.00 B:24.16 /0.00 @:127 B@:0
Recv:  T:55.10 /0.00 B:24.19 /0.00 @:127 B@:0
Recv:  T:59.41 /0.00 B:23.87 /0.00 @:127 B@:0
Recv:  T:63.40 /0.00 B:24.16 /0.00 @:127 B@:0
Recv:  T:67.15 /0.00 B:24.31 /0.00 @:127 B@:0
Recv:  T:71.00 /0.00 B:24.09 /0.00 @:127 B@:0
Recv:  T:74.56 /0.00 B:24.47 /0.00 @:127 B@:0
Recv:  T:78.20 /0.00 B:24.50 /0.00 @:127 B@:0
Recv:  T:82.12 /0.00 B:24.19 /0.00 @:127 B@:0
Recv:  T:85.62 /0.00 B:24.09 /0.00 @:127 B@:0
Recv:  T:89.03 /0.00 B:24.34 /0.00 @:127 B@:0
Recv:  T:92.34 /0.00 B:24.25 /0.00 @:127 B@:0
Recv:  T:95.59 /0.00 B:24.19 /0.00 @:127 B@:0
Recv:  T:98.98 /0.00 B:24.62 /0.00 @:127 B@:0
Recv:  T:102.27 /0.00 B:24.03 /0.00 @:127 B@:0
Recv:  T:105.37 /0.00 B:24.37 /0.00 @:127 B@:0
Recv:  T:108.43 /0.00 B:24.16 /0.00 @:127 B@:0
Recv:  T:111.42 /0.00 B:24.28 /0.00 @:127 B@:0
Recv:  T:114.66 /0.00 B:24.09 /0.00 @:127 B@:0
Recv:  T:117.50 /0.00 B:24.22 /0.00 @:127 B@:0
Recv:  T:120.23 /0.00 B:24.28 /0.00 @:127 B@:0
Recv:  T:123.18 /0.00 B:24.03 /0.00 @:127 B@:0
Recv:  T:125.91 /0.00 B:24.31 /0.00 @:127 B@:0
Recv:  T:128.84 /0.00 B:24.12 /0.00 @:127 B@:0
Recv:  T:131.40 /0.00 B:24.16 /0.00 @:127 B@:0
Recv:  T:134.12 /0.00 B:24.28 /0.00 @:127 B@:0
Recv:  T:136.62 /0.00 B:24.19 /0.00 @:127 B@:0
Recv:  T:138.99 /0.00 B:24.31 /0.00 @:127 B@:0
Recv:  T:141.73 /0.00 B:24.12 /0.00 @:127 B@:0
Recv:  T:144.35 /0.00 B:24.28 /0.00 @:127 B@:0
Recv:  T:146.80 /0.00 B:24.22 /0.00 @:127 B@:0
Recv:  T:149.00 /0.00 B:24.28 /0.00 @:127 B@:0
Recv:  T:151.66 /0.00 B:24.25 /0.00 @:127 B@:0
Recv:  T:153.95 /0.00 B:24.25 /0.00 @:127 B@:0
Recv:  T:156.31 /0.00 B:24.12 /0.00 @:127 B@:0
Recv:  T:158.35 /0.00 B:24.06 /0.00 @:127 B@:0
Recv:  T:160.40 /0.00 B:24.19 /0.00 @:127 B@:0
Recv:  T:162.96 /0.00 B:24.25 /0.00 @:127 B@:0
Recv:  T:165.02 /0.00 B:24.16 /0.00 @:127 B@:0
Recv:  T:167.24 /0.00 B:24.19 /0.00 @:127 B@:0
Recv:  T:168.88 /0.00 B:24.25 /0.00 @:127 B@:0
Recv:  T:171.09 /0.00 B:24.19 /0.00 @:127 B@:0
Recv:  T:173.13 /0.00 B:24.12 /0.00 @:127 B@:0
Recv:  T:175.58 /0.00 B:24.22 /0.00 @:127 B@:0
Recv:  T:177.36 /0.00 B:24.31 /0.00 @:127 B@:0
Recv:  T:178.98 /0.00 B:24.41 /0.00 @:127 B@:0
Recv:  T:181.08 /0.00 B:24.31 /0.00 @:127 B@:0
Recv:  T:183.32 /0.00 B:24.25 /0.00 @:127 B@:0
Recv:  T:185.37 /0.00 B:24.22 /0.00 @:127 B@:0
Recv:  T:186.99 /0.00 B:24.34 /0.00 @:127 B@:0
Recv:  T:188.81 /0.00 B:24.34 /0.00 @:127 B@:0
Recv:  T:190.76 /0.00 B:24.31 /0.00 @:127 B@:0
Recv:  T:192.04 /0.00 B:24.28 /0.00 @:127 B@:0
Recv:  T:193.88 /0.00 B:24.12 /0.00 @:127 B@:0
Recv:  T:195.66 /0.00 B:24.22 /0.00 @:127 B@:0
Recv:  T:197.27 /0.00 B:24.16 /0.00 @:127 B@:0
Recv:  T:198.65 /0.00 B:24.22 /0.00 @:127 B@:0
Recv:  T:199.97 /0.00 B:24.25 /0.00 @:127 B@:0
Recv:  T:202.10 /0.00 B:24.37 /0.00 @:127 B@:0
Recv:  T:203.17 /0.00 B:24.28 /0.00 @:127 B@:0
[... 131 lines cut ...]
Recv:  T:247.40 /0.00 B:24.56 /0.00 @:127 B@:0
Recv:  T:249.38 /0.00 B:24.06 /0.00 @:127 B@:0
Recv:  T:250.83 /0.00 B:24.19 /0.00 @:71 B@:0
Recv:  T:250.73 /0.00 B:24.56 /0.00 @:71 B@:0
Recv:  T:251.25 /0.00 B:24.44 /0.00 @:71 B@:0
Recv:  T:251.15 /0.00 B:24.00 /0.00 @:71 B@:0
Recv:  T:250.21 /0.00 B:24.37 /0.00 @:71 B@:0
Recv:  bias: 196 d: 58 min: 247.40 max: 251.88 Ku: 32.97 Tu: 17.20
Recv:  Classic PID
Recv:  Kp: 19.78 Ki: 2.30 Kd: 42.54
Recv:  T:248.33 /0.00 B:24.44 /0.00 @:127 B@:0
Recv:  T:247.71 /0.00 B:24.22 /0.00 @:127 B@:0
Recv:  T:248.65 /0.00 B:24.31 /0.00 @:127 B@:0
Recv:  T:249.27 /0.00 B:24.28 /0.00 @:127 B@:0
Recv:  T:251.04 /0.00 B:24.22 /0.00 @:69 B@:0
Recv:  T:251.77 /0.00 B:24.47 /0.00 @:69 B@:0
Recv:  T:251.15 /0.00 B:24.25 /0.00 @:69 B@:0
Recv:  T:251.04 /0.00 B:24.31 /0.00 @:69 B@:0
Recv:  bias: 197 d: 57 min: 247.29 max: 251.98 Ku: 30.97 Tu: 16.55
Recv:  Classic PID
Recv:  Kp: 18.58 Ki: 2.25 Kd: 38.43
Recv: PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h
Recv: #define  DEFAULT_Kp 18.58
Recv: #define  DEFAULT_Ki 2.25
Recv: #define  DEFAULT_Kd 38.43`

houseofbugs, 2017-11-03:
Auto tune fan 100% 1.1.5:

> Send: M303 C10 E0 S250
Recv: PID Autotune start
Recv:  T:20.84 /0.00 B:22.03 /0.00 @:127 B@:0
Recv:  T:21.22 /0.00 B:22.03 /0.00 @:127 B@:0
Recv:  T:22.81 /0.00 B:22.06 /0.00 @:127 B@:0
Recv:  T:25.53 /0.00 B:22.25 /0.00 @:127 B@:0
Recv:  T:29.00 /0.00 B:22.19 /0.00 @:127 B@:0
Recv:  T:32.00 /0.00 B:21.84 /0.00 @:127 B@:0
Recv:  T:35.58 /0.00 B:22.25 /0.00 @:127 B@:0
Recv:  T:39.46 /0.00 B:22.03 /0.00 @:127 B@:0
Recv:  T:43.17 /0.00 B:22.06 /0.00 @:127 B@:0
Recv:  T:46.88 /0.00 B:22.00 /0.00 @:127 B@:0
Recv:  T:50.77 /0.00 B:22.00 /0.00 @:127 B@:0
Recv:  T:54.25 /0.00 B:22.06 /0.00 @:127 B@:0
Recv:  T:58.39 /0.00 B:22.03 /0.00 @:127 B@:0
Recv:  T:62.03 /0.00 B:21.97 /0.00 @:127 B@:0
Recv:  T:65.57 /0.00 B:21.97 /0.00 @:127 B@:0
Recv:  T:69.37 /0.00 B:22.12 /0.00 @:127 B@:0
Recv:  T:72.86 /0.00 B:22.00 /0.00 @:127 B@:0
Recv:  T:76.73 /0.00 B:22.09 /0.00 @:127 B@:0
Recv:  T:80.33 /0.00 B:22.19 /0.00 @:127 B@:0
Recv:  T:83.69 /0.00 B:22.09 /0.00 @:127 B@:0
Recv:  T:87.12 /0.00 B:22.09 /0.00 @:127 B@:0
Recv:  T:90.49 /0.00 B:22.16 /0.00 @:127 B@:0
Recv:  T:94.01 /0.00 B:22.03 /0.00 @:127 B@:0
Recv:  T:97.28 /0.00 B:22.19 /0.00 @:127 B@:0
Recv:  T:100.46 /0.00 B:22.16 /0.00 @:127 B@:0
Recv:  T:103.73 /0.00 B:22.09 /0.00 @:127 B@:0
Recv:  T:107.19 /0.00 B:22.22 /0.00 @:127 B@:0
Recv:  T:110.26 /0.00 B:22.03 /0.00 @:127 B@:0
Recv:  T:113.41 /0.00 B:22.12 /0.00 @:127 B@:0
Recv:  T:116.43 /0.00 B:22.09 /0.00 @:127 B@:0
Recv:  T:119.33 /0.00 B:22.09 /0.00 @:127 B@:0
Recv:  T:122.67 /0.00 B:22.19 /0.00 @:127 B@:0
Recv:  T:125.61 /0.00 B:22.06 /0.00 @:127 B@:0
Recv:  T:128.41 /0.00 B:22.16 /0.00 @:127 B@:0
Recv:  T:131.24 /0.00 B:22.41 /0.00 @:127 B@:0
Recv:  T:134.21 /0.00 B:21.91 /0.00 @:127 B@:0
Recv:  T:137.11 /0.00 B:22.03 /0.00 @:127 B@:0
Recv:  T:139.67 /0.00 B:22.41 /0.00 @:127 B@:0
Recv:  T:142.43 /0.00 B:21.91 /0.00 @:127 B@:0
Recv:  T:145.25 /0.00 B:21.94 /0.00 @:127 B@:0
Recv:  T:147.80 /0.00 B:22.12 /0.00 @:127 B@:0
Recv:  T:150.59 /0.00 B:22.00 /0.00 @:127 B@:0
Recv:  T:153.31 /0.00 B:22.16 /0.00 @:127 B@:0
Recv:  T:155.81 /0.00 B:22.34 /0.00 @:127 B@:0
Recv:  T:158.22 /0.00 B:22.16 /0.00 @:127 B@:0
Recv:  T:160.85 /0.00 B:22.09 /0.00 @:127 B@:0
Recv:  T:163.40 /0.00 B:22.22 /0.00 @:127 B@:0
Recv:  T:165.81 /0.00 B:22.22 /0.00 @:127 B@:0
Recv:  T:168.14 /0.00 B:22.12 /0.00 @:127 B@:0
Recv:  T:170.42 /0.00 B:22.12 /0.00 @:127 B@:0
Recv:  T:173.15 /0.00 B:22.19 /0.00 @:127 B@:0
Recv:  T:175.65 /0.00 B:22.22 /0.00 @:127 B@:0
Recv:  T:177.87 /0.00 B:22.12 /0.00 @:127 B@:0
Recv:  T:180.00 /0.00 B:22.19 /0.00 @:127 B@:0
Recv:  T:182.39 /0.00 B:22.09 /0.00 @:127 B@:0
Recv:  T:185.00 /0.00 B:22.03 /0.00 @:127 B@:0
Recv:  T:187.33 /0.00 B:22.03 /0.00 @:127 B@:0
Recv:  T:189.35 /0.00 B:22.09 /0.00 @:127 B@:0
Recv:  T:191.45 /0.00 B:22.12 /0.00 @:127 B@:0
Recv:  T:193.62 /0.00 B:22.41 /0.00 @:127 B@:0
Recv:  T:195.82 /0.00 B:22.09 /0.00 @:127 B@:0
Recv:  T:197.63 /0.00 B:22.09 /0.00 @:127 B@:0
Recv:  T:199.41 /0.00 B:22.22 /0.00 @:127 B@:0
Recv:  T:201.56 /0.00 B:22.09 /0.00 @:127 B@:0
Recv:  T:203.97 /0.00 B:22.12 /0.00 @:127 B@:0
Recv:  T:206.21 /0.00 B:22.25 /0.00 @:127 B@:0
[... 142 lines cut ...]
Recv:  T:252.81 /0.00 B:22.19 /0.00 @:34 B@:0
Recv:  T:252.50 /0.00 B:22.50 /0.00 @:34 B@:0
Recv:  T:251.88 /0.00 B:22.41 /0.00 @:34 B@:0
Recv:  T:251.25 /0.00 B:22.37 /0.00 @:34 B@:0
Recv:  bias: 160 d: 94 min: 248.13 max: 252.81 Ku: 51.07 Tu: 22.28
Recv:  Classic PID
Recv:  Kp: 30.64 Ki: 2.75 Kd: 85.34
Recv:  T:249.90 /0.00 B:22.41 /0.00 @:127 B@:0
Recv:  T:249.17 /0.00 B:22.41 /0.00 @:127 B@:0
Recv:  T:248.54 /0.00 B:22.41 /0.00 @:127 B@:0
Recv:  T:248.54 /0.00 B:22.62 /0.00 @:127 B@:0
Recv:  T:248.96 /0.00 B:22.47 /0.00 @:127 B@:0
Recv:  T:249.17 /0.00 B:22.44 /0.00 @:127 B@:0
Recv:  T:251.46 /0.00 B:22.50 /0.00 @:33 B@:0
Recv:  T:251.88 /0.00 B:22.59 /0.00 @:33 B@:0
Recv:  T:252.29 /0.00 B:22.37 /0.00 @:33 B@:0
Recv:  T:251.88 /0.00 B:22.56 /0.00 @:33 B@:0
Recv:  T:251.25 /0.00 B:22.44 /0.00 @:33 B@:0
Recv:  bias: 161 d: 93 min: 247.92 max: 252.71 Ku: 49.42 Tu: 21.30
Recv:  Classic PID
Recv:  Kp: 29.65 Ki: 2.78 Kd: 78.95
Recv: PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h
Recv: #define  DEFAULT_Kp 29.65
Recv: #define  DEFAULT_Ki 2.78
Recv: #define  DEFAULT_Kd 78.95

fiveangle, 2017-11-04:
> There was mention in the 1.1.6 changelog that PID tuning was changed.

Where ?  There's nothing in temperature.cpp that I can find:

https://github.com/MarlinFirmware/Marlin/blame/8dc10fb4c206e4bbcbe9ea21a25d5a9802aa54a8/Marlin/temperature.cpp

The complete 115 - 116 diffs:  https://github.com/MarlinFirmware/Marlin/compare/1.1.5...1.1.6

I appoligize as I forgot to tell you to enable PID_DEBUG for these tests :(

But it appears there was definitely something different between the configs (or code controlling the process) on your 116 vs 115 autotunes above because the reported power levels look almost exactly double on 116 vs 115 for the cycling tests, which could mean a lot of stuff but tells me the first thing to do is confirm configs are identical between them.  But it's almost as if the printer is doubling the power of the heater from what it things it should be doing, resulting in the large overshoot.

115:
Recv: T:249.17 /0.00 B:22.44 /0.00 @:127 B@:0
Recv: T:251.46 /0.00 B:22.50 /0.00 **@:33** B@:0

vs.

116:
Recv: T:249.27 /0.00 B:24.28 /0.00 @:127 B@:0
Recv: T:251.04 /0.00 B:24.22 /0.00 **@:69** B@:0

and they're consistent down the line (see attached comparison report).

[image]

[115-LEFT_vs_116_RIGHT.html.zip](https://github.com/MarlinFirmware/Marlin/files/1442943/115-LEFT_vs_116_RIGHT.html.zip)

But I can't find a code change that correlates.

Are you absolutely sure you're starting with otherwise identical Configuration*.h ?  If no one else has an idea, I hate to ask but the PID_DEBUG output may be more enlightening.  But maybe someone else will have a better idea.

houseofbugs, 2017-11-04:
This is for my CR-10 Machine with a V6 Hotend:

1.1.5: https://www.dropbox.com/s/c558tturwo1740h/1.1.5.zip?dl=0
1.1.6: https://www.dropbox.com/s/1kg9y3p484njbcf/1.1.6.zip?dl=0

Differences: PID Tune values (I re-tuned them) and Autotemp is off on the 1.1.6, 

If I let the printer sit heated on 1.1.6 I get the over and undershoot for the temps. Could having Autotemp disabled do that? I will have to flash and check. Doesn't make sense if it does because Autotemp should only be adjusting the heater when it is printing, not idle.

jmdearras, 2017-11-04:
What does Autotemp do?

AnHardt, 2017-11-04:
Melting filament needs energy.
More filament needs more energy.
The idea of Autotemp is to deliver energy, to melt the filament before the temperature of the nozzle drops.

jmdearras, 2017-11-04:
it anticipates the load, then?
the tip regulates by reacting to the change.

houseofbugs, 2017-11-06:
Yeah it just compensates for the extra load.

houseofbugs, 2017-11-08:
I spent time comparing the 1.1.5 and 1.1.6 source and I can't see anything related to the PID/Hotend regulation that would be doing this. But I also setup the 1.1.6 bugfix and it has the same issue. You can see below that I flashed the 1.1.5 and 1.1.6 bugfix and the 1.1.6 builds had the same issue where the 1.1.5 was fine.

Also tried changing the thermistor tables to 1 and 5 and they both did the same thing.

Any ideas guys? 

[image]

houseofbugs, 2017-11-08:
Do you have your config handy? This was on a system with a Gen L as well.

jmdearras, 2017-11-08:
using the PID in the file,

jmdearras, 2017-11-08:
[image: Inline image 1]

houseofbugs, 2017-11-09:
Just an update. While auto tune was not returning working values and the old 1.1.5 ones were not working in 1.1.6 after turning the values down across the board I am able to get stable temps out of this. Not sure why but it's working now.

jmdearras, 2017-11-09:
turning which values down?

heineman-bob, 2017-11-10:
At first glance it looks like this is a relevant change in the code that may affect the tuning.  Looks like it was an optimization.

https://github.com/MarlinFirmware/Marlin/commit/bbb94b8f68d296f7ec9cb55fcc8d2c619640e946#diff-7e90b29c765c3047c7dc2aa8eff0856c

Roxy-3D, 2017-11-10:
> At first glance it looks like this is a relevant change in the code that may affect the tuning. Looks like it was an optimization.

The timing difference is very very small in this code.   And it will only come into play if there are outstanding Baby Steps in one or the other direction.   In the normal case (with no Baby Steps outstanding) the only difference is an extra assembly language compare of the sign bit and branch versus a single compare and branch.   Factoring in this happens inside a loop...  I'm guessing we are talking less than 20 us difference.

```cpp
   #if ENABLED(BABYSTEPPING)
      LOOP_XYZ(axis) {
        const int curTodo = babystepsTodo[axis]; // get rid of volatile for performance
 -      if (curTodo > 0) {
 -        stepper.babystep((AxisEnum)axis, /*fwd*/true);
 -        babystepsTodo[axis]--;
 -      }
 -      else if (curTodo < 0) {
 -        stepper.babystep((AxisEnum)axis, /*fwd*/false);
 -        babystepsTodo[axis]++;
 +      if (curTodo) {
 +        stepper.babystep((AxisEnum)axis, curTodo > 0);
 +        if (curTodo > 0) babystepsTodo[axis]--;
 +                    else babystepsTodo[axis]++;
        }
      }
```

fiveangle, 2017-11-10:
I don't have the wizbang assembler profiling capabilities like Roxy, but I did a quick test on actual HW before she provided the above. I was thinking of a possible long shot weirdness in case `const `declared var might not have been evaluating `FALSE ` if merely initialized to "0", but after testing on actual HW, the resulting machine logic is in fact identical.

heineman-bob, 2017-11-11:
Thanks for the follow up.  I was in a different group and only had a bit of time to inspect the source and didn't see a reference to this change so I figured I'd just point out the difference as at least a talking point to see if it was possible it had adverse effects.  Appreciate the further explanation!

RowanMeara, 2017-11-12:
I tested autotuning on the current 1.1.x and 1.1.5 with and without fans and babystepping.  I was unable to see any difference in the temperature stability and the resulting values were almost exactly the same.

closed 2017-12-30 by houseofbugs