MarlinFirmware/Marlin #8083: [FR]E1 heating when using E1 for another Z motor[/FR]

number12, 2017-10-25:
Hi there new to marlin and working with a MKS Gen 1.4 and been doing good on it, Got it all working and I noticed after testing the hotend and heatbed for stability for an hour, I looked at the LCD and it says "E1 Heating...] and when I try to move the X axis i doesn't move unless I turnoff heating for the extruder. Im using my E1 port for my second Z motor which I changed in the configuration_adv.h file and works. Now I just got this issue. I still haven't done a test print as im calibrating the extruder motor atm, but if anyone can give me a solution to this "E1 heating..." while im doing this that would be great. Thank you!

Phr3d13, 2017-10-25:
Attach configuration.h and configuration_adv.h by dragging & dropping, selecting them, or pasting from the clipboard, please.

number12, 2017-10-26:
Also today when calibrating my extruder I noticed the E motor doesnt push filament when it reaches the hotend. It will push and retract before that, but once it gets there it stops. I can push filament manually so there isnt no blockage, so not sure if this is related. Thank you for your help!

configuration.h
> /**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010100

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// example_configurations/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
[... 3033 lines cut ...]
 * display. Requires 3 signal wires. Some useful debug options are included to demonstrate its usage.
 *
 * Fully assembled MAX7219 boards can be found on the internet for under $2(US).
 * For example, see https://www.ebay.com/sch/i.html?_nkw=332349290049
 */
//#define MAX7219_DEBUG
#if ENABLED(MAX7219_DEBUG)
  #define MAX7219_CLK_PIN   64  // 77 on Re-ARM       // Configuration of the 3 pins to control the display
  #define MAX7219_DIN_PIN   57  // 78 on Re-ARM
  #define MAX7219_LOAD_PIN  44  // 79 on Re-ARM

  /**
   * Sample debug features
   * If you add more debug displays, be careful to avoid conflicts!
   */
  #define MAX7219_DEBUG_PRINTER_ALIVE    // Blink corner LED of 8x8 matrix to show that the firmware is functioning
  #define MAX7219_DEBUG_STEPPER_HEAD  3  // Show the stepper queue head position on this and the next LED matrix row
  #define MAX7219_DEBUG_STEPPER_TAIL  5  // Show the stepper queue tail position on this and the next LED matrix row

  #define MAX7219_DEBUG_STEPPER_QUEUE 0  // Show the current stepper queue depth on this and the next LED matrix row
                                         // If you experience stuttering, reboots, etc. this option can reveal how
                                         // tweaks made to the configuration are affecting the printer in real-time.
#endif

#endif // CONFIGURATION_ADV_H

Phr3d13, 2017-10-26:
Are you using a Marlin release, or bugfix branch?

number12, 2017-10-26:
Its Marlin 1.1.6

thinkyhead, 2017-10-31:
On the LCD "E1" refers to what geeks call "E0."

closed 2017-10-31 by thinkyhead