MarlinFirmware/Marlin #8085: Cant use OpenDACT with Marlin

vido89, 2017-10-25:
To set delta printer correctly we need to measure couple of things, I dont know how they are called in Marlin but in repetier they are called (Alpha Rotation, Delta Radius, XYZ offset, Horizontal Radius, and Diagonal Rod length). So we need to set them to value which they ARE on our printer and NOT to ideal value.To achieve that we just need to measure z hight with micrometer and we can calculate real values of our delta. I came across OpenDACT its designed to work with Repetier but Im sure that we can use it Marlin to, but we need to change those parameters. Sadly those parameters are not available in Marlin EEPROM (should we make feature request?)

My question is what is name for Alpha Rotation, Delta Radius, XYZ offset, Horizontal Radius in Marlin and where we can change those ? I know that "Diagonal Rod length" is "DELTA_DIAGONAL_ROD"
[Here](https://www.youtube.com/watch?v=im5MDpTkzCM) you can watch how to perfectly calibrate your Delta to be within 0.01 from end stop.

kevintong116, 2017-10-26:
It is not designed for Marlin...
It's for Repetier only.

vido89, 2017-10-26:
> It is not designed for Marlin...
> It's for Repetier only.
What Im saying is we should be able to use same technique with Marlin

fiveangle, 2017-10-26:
Looks more like you need to submit a feature request to OpenDACT to request support for Marlin:

https://github.com/RollieRowland/OpenDACT/issues

There's no changes needed by Marlin as all the GCODE to set these values (and surely perform whatever GCODE movement commands OpenDACT performs as necessary to do its "calibration") already exist.

-=dave

vido89, 2017-10-26:
Right that is good thing @fiveangle but what about adjusting Alpha Rotation, Delta Radius, XYZ offset, Horizontal Radius ?

fiveangle, 2017-10-27:
Those are Repetier terms so someone who 1) knows about Repetier, and 2) cares to make OpenDACT work with Marlin, needs to translate what those are in Repetier to what they are in Marlin and provide to OpenDACT so they can translate to their needs.  The OpenDACT devs will know how they do this more than anyone.

vido89, 2017-10-30:
```
Marlin                           Repetier

DELTA_DIAGONAL_ROD               Diagonal Rod length
DELTA RADIUS (A,B,C)             Horizontal radius
                                 XYZ offset
                                 ALPHA (A,B,C)

```

vido89, 2017-10-30:
There is some progress on Open DACT gut hub page https://github.com/RollieRowland/OpenDACT/issues/23

thinkyhead, 2017-10-31:
Have you played with `G33` delta calibration at all?

vido89, 2017-10-31:
Yes I tried it but Im not satisfied completely with it because my prints are of in dimensions and almost always I have elephant feet artefact

LVD-AC, 2021-08-28:
I guess:

Marlin                           Repetier

DELTA_DIAGONAL_ROD                   Diagonal Rod length
DELTA RADIUS (A,B,C)                      Horizontal radius
Endstop Offsets                                 XYZ offset
Tower Angle Corrections                   ALPHA (A,B,C)
Delta Height                                       ???

Those need to be set with M665 and M666 to have a "flat bed"; they can be calculated with G33 (or OpenDACT)

@vido89 "almost always I have elephant feet artefact"

You need to tweak the Z-offset to cure that.

@vido89 "my prints are of in dimensions"

That smells like a stepper calibration error.

closed 2022-04-09 by thisiskeithb