MarlinFirmware/Marlin #8090: Travis vs Neopixel

SCiunczyk, 2017-10-25:
In the .travis.yml in line 64 there is still the old NEOPIXEL_RGBW_LED definition. I think that should be changed to NEOPIXEL_LED ?

Tannoo, 2017-10-26:
Change it in your current PR.

fiveangle, 2017-10-26:
I would make the change.

🤣

SCiunczyk, 2017-10-26:
LOL. Done .... but I can't pass travis checks because of compile error fixed in PR #8089. Original version of travis just skipped neopixel check.

fiveangle, 2017-10-26:
Do yourself a favor and put meaningful descriptions on your PRs and they migh have a higher chance of getting reviewed faster.

> "_[bf2.0] Fix for #1234_" 

^**shrug - could be anything including someone trying to merge a new framework that offloads GCODE processing to an outboard processor ;)**

vs:

> "_[bf2.0] Fix for compile error due to missing `set_led_white();` function_"

^**ooooh, that sounds important and a 3 minute job to merge !**

:wink: 

Scott sounds like he's been pretty busy actually messing with his printers lately so any additional things that make it easier for him to gauge importance can only help I think.

-=dave

SCiunczyk, 2017-10-26:
@fiveangle better now ?

closed 2017-10-27 by SCiunczyk