MarlinFirmware/Marlin #8091: [FR] Extruder duplication

SydneyMakers, 2017-10-25:
First off, please forgive the terrible formatting.
I have an interesting feature request that is very similar to the current Nozzle Duplication Mode M605, and Ditto Printing in Makerbot.

In short, I have a long bowden tube (around 1m) going from my hotend carriage to the back corner frame, down the frame edge, and to a second extruder motor in the base of the printer.
The purpose was to have a spool in the bottom with one extruder to push from one end, and the carriage extruder to pull from the other end, since one extruder does not have enough torque to accomplish this.
I have made a firmware modification that does this, but it required significant pin changes, setting 

> #define EXTRUDERS 2

and

> extruder_duplication_enabled = true

in marlin_main.cpp.
This also created the problem of having a second thermistor to wire in (i set the pins to the E0 thermistor), setting the heater pin to an unused pin (and subsequently disabling all instances where that heater could be activated and cause a sanitycheck failure)
As well as creating multiple LCD options for the fake extruder.

---------
My hope with this Issue is to hopefully provide a method where a second extruder (presumably E1) will completely mimic the first (E0) (with reversing options for left-hand vs right-hand motor direction) without requiring thermistor pins and heater pins to be set, or defining more than one extruder in the Configuration.h file

I know this is potentially a massive feature request, and as far as I know, it hasnt been implemented in a straight forward fashion.

--------
The other upside to this feature is it would allow a motor breakout to an external motor or external device to be synced to the extruder motor for devices such as enclosed filament spools or other applications.

Phr3d13, 2017-10-25:
As a workaround, I just came up with this:
#define EXTRUDERS 2
#define SINGLENOZZLE
#define MIXING_EXTRUDER
Always set the mix to 50/50

fiveangle, 2017-10-25:
If extruders are same, why not just connect both to the same driver ?

fiveangle, 2017-10-25:
That said, you have given me an idea on how to enable geared extruder with two regular steppers connected to same driver so they drive filament from each side.  A bit heavy, but they can be half the size :)

SydneyMakers, 2017-10-25:
because the x,y,e0,e1 steppers only have one set of pin headers, I did try a version where E0 was swapped with Z to use the dual headers, but there was not enough torque across the steppers, this would also only work for motors that are the exact same.

psavva, 2017-10-27:
Would it not be easier to wire the 2 steppers in series?  In that case, no Firmware changes needed at all...

thinkyhead, 2018-02-09:
The demand for this kind of feature is extremely low and devs have a lot on their plates dealing with bugs and 32-bit migration so it's unlikely this will be implemented anytime soon. But, you may be able to hire a developer for $$$ if you need this feature very much.

InsanityAutomation, 2019-10-15:
@boelle @shitcreek this is done as well.

closed 2019-10-15 by boelle