MarlinFirmware/Marlin #8096: hey guys! i have tevo tarantula classic. i have a problem with x pin on motherboard. so i want to use E1 pin for control x step motor. i try to change on configration.h file x pin to E1 pin but i couldn't. if you can do this for me i will be happy. :)

goker1, 2017-10-25:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

Phr3d13, 2017-10-25:
An excerpt from the template you deleted to post this 'issue':
```
Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
```

goker1, 2017-10-26:
I added configration files. TEVO 3D Printer bord is  MKS Base V1.4 3D Printer Control Board With USB Mega 2560 R3 Motherboard RepRap Ramps1.4 Compatible. so i don't know which configration file is correct, i added 3 of them. thank you so much.

[New folder.zip](https://github.com/MarlinFirmware/Marlin/files/1416913/New.folder.zip)

hey guys! i have tevo tarantula classic. i have a problem with x pin on motherboard. so i want to use E1 pin for control x step motor. i try to change on configration.h file x pin to E1 pin but i couldn't. if you can do this for me i will be happy. :)

Phr3d13, 2017-10-27:
I'll take a look next time I'm in front of my computer

thinkyhead, 2017-10-31:
If you want to make this kind of customization, change the pins in your board's pins file so that the X stepper pins use the E1 pin numbers. This is not something you can do from your configuration.

marcio-ao, 2017-12-05:
I second @thinkyhead's suggestion. We remap stuff all the time in "pins_<boardname>.h"

closed 2018-02-10 by thinkyhead