MarlinFirmware/Marlin #8097: [FR] T1 as a ride-a-long with offset to nozzle

empakoso, 2017-10-25:
I have installed a pen (like a 2D plotter) on my Cartesian 3D printer.

How Can I enable extruder = 2 so that I can gain the benefit of entering an offset to my permanently mounted pen/marker tip.

I would like to do this without all the extra heater and T/C values on my display or taking up the space in the processor.

Is this possible with the current combination of features in Marlin?

fiveangle, 2017-10-28:
ensure NO_WORKSPACE_OFFSETS is not defined in Configuration_adv.h, then use M206 (or M428) to adjust offsets due to your Pen holder not being where your nozzle tip is:

http://marlinfw.org/docs/gcode/M206.html

-=dave

empakoso, 2017-10-29:
Great advise! Thanks.  I did not know that definition related to M206.
Thank you.  This exactly what I was looking for.

closed 2017-10-29 by empakoso