MarlinFirmware/Marlin #8102: Re-Arm compile is failing after changes too alpha sort

apballard, 2017-10-26:
commit 3d2b2eb1ca16b576b8138668b74f022654a0664a

https://github.com/MarlinFirmware/Marlin/commit/3d2b2eb1ca16b576b8138668b74f022654a0664a

### Compile Error
When I compile with `SDSUPPORT` disabled and `SDCARD_SORT_ALPHA` I get an error:
```
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:40:0,
from Marlin/src/HAL/HAL_LPC1768/HAL.cpp:23:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:332:8: error: #error "SDSORT_CACHE_VFATS must be 2 or greater!"
#error "SDSORT_CACHE_VFATS must be 2 or greater!"
^~~~~
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:40:0,
from Marlin/src/HAL/HAL_LPC1768/HAL_timers.cpp:31:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:332:8: error: #error "SDSORT_CACHE_VFATS must be 2 or greater!"
#error "SDSORT_CACHE_VFATS must be 2 or greater!"
^~~~~
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:40:0,
from Marlin/src/HAL/HAL_LPC1768/HardwareSerial.cpp:25:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:332:8: error: #error "SDSORT_CACHE_VFATS must be 2 or greater!"
#error "SDSORT_CACHE_VFATS must be 2 or greater!"
^~~~~
Compiling .pioenvs/Re-ARM/src/src/HAL/HAL_LPC1768/HAL_timers.o
```

-----------

isaacfank, 2017-10-26:
I cant seem to find how to compile for the re-arm. What am i missing or what program/options do i choose/site with info.

apballard, 2017-10-26:
@isaacfank 

I'm using platformio using atom.

In platformio.ini change 'env_default = Re-ARM'
In the configuration.h '#define MOTHERBOARD BOARD_RAMPS_14_RE_ARM_EFB'

apballard, 2017-10-26:
oh man, if I enable sdsupport, things go even worse......

What I'm not sure about and is confusing me is that the ENABLED macro is ignored in the sanity check.
if someone could explain why I'd be very grateful because I've seen it before but never to to the reason.

small excerpt of a lot of errors and warnings with sdsupport enabled:

```
In file included from Marlin/src/module/../inc/../pins/pins.h:245:0,
from Marlin/src/module/../inc/MarlinConfig.h:35,
from Marlin/src/module/stepper_indirection.h:47,
from Marlin/src/module/stepper_indirection.cpp:34:
Marlin/src/module/../inc/../pins/pins_RAMPS_RE_ARM.h:277:0: warning: "SDSORT_USES_RAM" redefined
#define SDSORT_USES_RAM    true   // Pre-allocate a static array for faster pre-sorting.
 

In file included from Marlin/src/module/../inc/MarlinConfig.h:32:0,
from Marlin/src/module/stepper_indirection.h:47,
from Marlin/src/module/stepper_indirection.cpp:34:
Marlin/src/module/../inc/../../Configuration_adv.h:532:0: note: this is the location of the previous definition
#define SDSORT_USES_RAM    false  // Pre-allocate a static array for faster pre-sorting.
 
In file included from Marlin/src/module/../inc/../pins/pins.h:245:0,
from Marlin/src/module/../inc/MarlinConfig.h:35,
from Marlin/src/module/stepper_indirection.h:47,
from Marlin/src/module/stepper_indirection.cpp:34:
Marlin/src/module/../inc/../pins/pins_RAMPS_RE_ARM.h:279:0: warning: "SDSORT_CACHE_NAMES" redefined
#define SDSORT_CACHE_NAMES true   // Keep sorted items in RAM longer for speedy performance. Most expensive option.
 
In file included from Marlin/src/module/../inc/MarlinConfig.h:32:0,
from Marlin/src/module/stepper_indirection.h:47,
from Marlin/src/module/stepper_indirection.cpp:34:
1 result found for 'SDSUPPORT'Finding with Options: Case Insensitive
```

apballard, 2017-10-26:
Seems to be resolved after #8063

closed 2017-10-26 by apballard