MarlinFirmware/Marlin #8103: Marlin 115 tried to burn my house down during PID tuning

jarrodasmith, 2017-10-26:
### Bug Report
- Description: I started a PID tune from Octoprint's terminal.  My thermistor went out during the PID tune (or maybe before I started it, not sure on this detail).  Heater turned on and just never shut off until I intervened.

- Expected behaviour: Firmware should detect mintemp before/during PID tuning and shut the heat off (it was reading -14C at the time).
- Actual behaviour: Smoking PLA and scorched hotend parts.
- Steps to reproduce:
  - Install TH3D115 hex binary firmware (th3dstudio.com) on CR-10
  - Install Octopi 0.14 and Octoprint 1.3.5
  - Disable the thermistor
  - Issue M303 from octoprint

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/1418930/config.zip)

guiseco, 2017-10-26:
AUTOTUNE does not respect MAXTEMP. If you set 500C ... KABUUUM !!!!

jarrodasmith, 2017-10-26:
I did:

M303 E0 S205 C10

The issue isn't that I set a high number, it's that the thermistor went offline (it read -14C) so the firmware kept pushing amps to the heater even though it was probably close to 400C by the time I could run downstairs and manually switch off the power.  The bottom ~20mm of PTFE in my throat was in a smoking puddle on the build plate.... 

This is a serious safety concern.

studiodyne, 2017-10-26:
Say ' THANK YOU ' to marlin 
Now you know , what is the maker life that have thermistors and these eternal wiring short circuit between nude wires or on the hotend

Buy thermocouple type k or pt100 ; i suggest PT100 because all hotend have the 3 mm holes for it , volcano , or normal , and no limits of temp
and ....... NO PROBLEMS OF ETERNAL MIN TEMP BRUTAL CRASHES

I love thermistors but not on a machine that can shake while 30 hours a  pair of nude wires !!!!!!!!!!

Grogyan, 2017-10-26:
I agree, just get a thermocouple, or PT100, these are sensors designed for high temperature, and are resilient to vibration.
Those cheap thermistors, break.

Or get an E3D V6 hotend, which have temperature cartridges

jarrodasmith, 2017-10-26:
Thanks for responding.  This is a SERIOUS safety concern.  I prefer infinity min temp crashes to one fire.  

I do agree about the thermocouple - I already have parts on the way to build M3 cartridge type thermistors for this application.  But even when I get those, it does not mean I want to run the risk of a fire under any circumstance just to avoid losing a few prints here and there....

This is a severe issue that needs to be addressed by both hardware and software fixes.

jarrodasmith, 2017-10-26:
> On Oct 26, 2017, at 6:13 PM, Jarrod Smith <JARROD.SMITH@GMAIL.COM> wrote:
>  But even when I get those, it does not mean I want to run the risk of a fire under any circumstance just to avoid losing a few prints here and there....

In hindsight what I wrote there doesn't even make sense.  But I think you get my point.  The good news is that we all appear to be in agreement that the thermistor issue is very real and annoying.  I think that adds more weight to the argument for a software fix.  Otherwise you are going to have some newbies like me buy a $300 printer off aliexpress with those crap thermistors and burn half his house down with it.

Roxy-3D, 2017-10-26:
So...   It is impossible to please everybody.    The thermal protections numbers for THERMAL_PROTECTION_HOTENDS and THERMAL_PROTECTION_BED are much too tight for me.   I always end up relaxing them significantly for my printers.   But the people that did the work want them that tight and I respect their opinions.

If you are worried about burning your house down, you probably should only use the printer while you are watching it.   And you should probably only use it with the printer placed outside in the middle of a concrete driveway.    Maybe that isn't enough.   Maybe you should stack some concrete blocks around it to make sure anything that catches fire is isolated and the fire can't spread.    Also, it might be wise to put a 5 amp fuse on the circuit where you are plugging in the printer to help limit what can go wrong.

While you are doing those things we can continue to discuss other ways the firmware can help you have a confidence in the safety of your using your new AliExpress printer!

Spawn32, 2017-10-26:
or use dual thermistors on the hotend... not sure Marlin has support for it ?

jarrodasmith, 2017-10-26:
Why the hostility?  I think a request to obey Max Temp and Min Temp during PID tuning is reasonable.

Roxy-3D, 2017-10-26:
> Why the hostility?  I think a request to obey Max Temp and Min Temp during PID tuning is reasonable.

Did M115 REALLY try to burn your house down.   You are basically accusing the author (and everybody that has contributed to the M115 code) of intentionally trying to cause you harm.   Perhaps it would make sense to change the title to something like "Suggested improvement for M115's thermal behavior"

studiodyne, 2017-10-27:
STOP !
The only thing we want to say , is our experience of marlin and the brutality of the temperature alarm algorithms
If temps make shit , print stop
And all here knows enough marlin to say , that  it's possible there is an issue about this , but , we are all sure that , before to bug , we have a hardware that don't give enough stable value 
Here , we are all sure that if you haven't theses thermistors 'sorry ' these nude wires that touch the hotend and receive your 3G/4G cell phone waves and more lolllllllll , you never had this issue

Yes , we must work on several wiring problems and  your issue will be seriously explored

Give us , all , the circonstances , to make this crash 
In a perfect order , because , we all want to try to make this at home , without your hardware , but to have the same exact situation , and after we will say if it's an issue

No war here ,  the only war , is to have the first perfect layer with UBL loll

jarrodasmith, 2017-10-27:
Thank you.  If I was impolite in my wording of the subject line, it was because this incident freaked me out.  I hope all you reasonable and generous people can understand that.  That does not in any way diminish the otherwise extremely clever bit of software you've managed to put together so don't read any more into it than that.

Roxy-3D, 2017-10-27:
From another perspective....   Everybody here has spent countless hours trying to make things better for the generic user.   My biggest contribution is the UBL Bed Leveling.   I did **_EVERYTHING_** possible to make it **_SAFE_** so it doesn't needlessly hurt anybody's machine.   There are 1000 different ways the bed leveling and mesh editing software can drive a nozzle into the bed and damage the user's machine.   That still does happen occasionally.   But I've done everything in my power to make it not happen.   The people that work on the thermal code are at least that contentious.  

If people are volunteering their time (and expertise)...   It is best to proceed gently and make them want to help you.   

With that understanding...   I suspect somebody is going to find the problem interesting and be willing to add some safety checks to M303.

RowanMeara, 2017-10-27:
I know that everyone who has contributed has put in many hours to make Marlin better and safe and I appreciate their effort.  However, I found the pid auto-tuning code and it seems to be a little rough (one confusing 200+ line function).  I'm interested in contributing and will look around and see if I can find the issue.

jarrodasmith, 2017-10-27:
> If people are volunteering their time (and expertise)... It is best to proceed gently and make them want to help you.

Roxy-3D, thank you for your contributions (seriously).  We all want to be recognized for our hard work, that is human nature.

Since something I said seems to have really rubbed you the wrong way, I'm going to get very specific so you and others who I may have put off will understand where I'm coming from.  I am a person who watched their printer nearly go up in flames.  Politeness was not at the top of mind when I came here to alert you all to an issue that could potentially cause loss of property or even life.  Getting your attention was.  My frank tone is not coming from a place of anger, or ego, or spite.  It's out of concern that the next person who uncovers this issue may not be in a position to intervene as quickly as I was.

If someone reading this has the expertise to diagnose and fix this, don't do it for me - I know how to avoid it.  Do it for the next guy or gal, who doesn't.

And thank you all for an amazing piece of software.  I authored some open source software back in the '90s, so I fully understand the commitment.  I once had someone at a LUG meeting tell me to my face that my software was a piece of shit.  And later I got threatened with a lawsuit by a publicly traded company who decided I was infringing on their trademark by supporting their products in Linux.  So I stopped doing it.  Then they offered me a job.  I said no.

This is not that.

jarrodasmith, 2017-10-27:
Thank you Rowan.  If I can provide any info from my end, let me know.

studiodyne, 2017-10-27:
@Roxy-3D 
Fuckkkkkkkkkkkkkkkkkkkkkkk !!!! UBL is your child  ? 
Damned !!!!
RESPECT FOR YOU !!!!!
All menus , EEPROM storing , G26 unit , 
Fuck it's great !!!!!!!!!!!!!!!!!!!!!!

studiodyne, 2017-10-27:
Contributing to marlin ,or other sofware is a big donation of time , every line of code generate 1500 mistakes and errors
But , in robotics like 3d printer , it's the worst !!!!  Upload are long , Heating too ,  you write line ' one minute ' and to test this line ' 30 min ' , each bad writting cost 30 min
You can waste a days , or a night just for a forgotten ' { ' , or bad writting of some conditionnals and more
Sometimes , just to understand something in this jungle can cost you 4 night and days , and no written line yet  ...

RowanMeara, 2017-10-27:
This might be wrong because I've never really looked at this code base before.

I think the issue is that pid_autotune disables all heaters thus turning off the temperature sanity checking that would happen when Marlin sees the heaters as on.  I believe your problem occurred because pid_autotune directly sets the PWM values without doing any error checking at all.

Tannoo, 2017-10-27:
I run my printer near an open can of gasoline... is that an issue?

studiodyne, 2017-10-27:
Oh !!! No temp alarm on M303 😲 ....
It's not important because the user is here and close.... but ... if a m303 appears by mistake. ..or anything else ... and no human presence ..... it's dramatic. 
Tonight i make a test to unconnect my sensors to be sure there's no thermal protection

studiodyne, 2017-10-27:
If you're right .. this must be quickly fixed

RowanMeara, 2017-10-27:
I think I found a pretty noninvasive fix so I will probably submit a pull request tomorrow.

AnHardt, 2017-10-27:
```
Steps to reproduce:

Install TH3D115 hex binary firmware (th3dstudio.com) on CR-10
Install Octopi 0.14 and Octoprint 1.3.5
Disable the thermistor
Issue M303 from octoprint
```

If the problem is reproducible in this order, it's not M303s fault. Min temp error should kill the machine in the moment the sensor is disconnected.

Are you sure this downloaded binary is produced with the appended config?

There is one way to disable min temp errors - setting mintemp to a value lower than the lowest value in the thermistor table.

studiodyne, 2017-10-27:
hex ? Or marlin 1.1.5 ?

studiodyne, 2017-10-27:
@RowanMeara ,  do you really have identified M303 is not under thermal prot ?

AnHardt, 2017-10-27:
Think i got it.
Since the introduction of the high temperature thermistors we check for min/max temp errors only when the heater is on. We detect that by checking if the target temp. is > 0. target temp. is not used by M303.

jarrodasmith, 2017-10-27:
> On Oct 27, 2017, at 6:50 AM, AnHardt <notifications@github.com> wrote:
> 
> Are you sure this downloaded binary is produced with the appended config?

I have no way of knowing without asking Timothy.  It came from the source folder of the binary download, but he may only provide that to comply with licensing.

I have never built Marlin from source.

guiseco, 2017-10-27:
I did this:

**#define HEATER_0_MAXTEMP 275**

After that, I ran the autotune:

**M303 E0 S2750 C10**

Autotune does not respect HEATER_0_MAXTEMP and will attempt to reach 2750 C.

ruggb, 2017-10-27:
IMHO, this sounds like a real problem. So why are all the developers defending themselves instead of looking for a solution. It only makes a long thread that gets nothing done.
Max temp is defined in the config file. If one enters 2750 instead of 275, or any number > Max, marlin should produce an error.
If the temp stays at -14 or for that matter is there in the first place, marlin should error.
sounds easy to me, but what do I know?

Tannoo, 2017-10-28:
My low temp is set for -15. I don't need an error when the ambient temp is that low.

But, I'm not usually starting a print when it's that cold.

jarrodasmith, 2017-10-28:
I really hope your thermistor never fails open because you have effectively defeated an important safety feature.

studiodyne, 2017-10-28:
My safety thermal settings are  120 seconds and 10 degrees for bed , and 120 seconds and 4 degree for hotends 
Under this setting the risk to crash MY PRINTER  is too high
The problem is the default setting too low , and all users are afraid to set stronger values , 
For me a serious Safety settings for thermal must be ' SERIOUS ' and not too short
To make a fire 120 seconds is really too short , and 120 is risk free for normal printing temp curves in all situations

bobc, 2017-10-28:
You should never rely on software for safety.  You have heated surfaces, and plastic that burns really well, a recipe for disaster. Commercial plastic products must use flame retardant additives by law*, filament used for 3DP does not contain retardants.  (* depending on national laws I guess).

The safety of home 3d printers is very poor, and although a known issue is often overlooked. Personally, I always regard printing as a "attendance required" activity. Now I know a lot of folks do long unattended prints and say "its fine", but those are all accidents waiting to happen. Unfortunately people take a binary approach to risk:
- it never happened to me , therefore risk is <0.001%, i.e. not worth thinking about
- it happened to me, therefore risk is 99%.

Fire services here are concerned about a rise in house fires due to domestic appliances, such as fridges and washing machines. Those are appliances that are supposed to have passed safety regs. Home builds or cheap machines imported from China won't even have been tested against safety standards.

We can add all the features we like to Marlin, but it will never be "safe" to an official standard - it requires hardware support. I would rather people recognize that 3d printing is inherently risky, rather than be complacent and falsely rely on "safety features" in the software.

If you are concerned about your house burning down, do not print unattended. If you are not concerned, you should be!

fiveangle, 2017-10-28:
If anything this would make a great clickbait YouTube video…

**Guy starts PID autotune a defective thermistor.
YOU'LL NEVER GUESS WHAT HAPPENS NEXT !!!**

thinkyhead, 2017-10-31:
> obey Max Temp and Min Temp during PID tuning

Very much appreciate the addition! Safety is vital, especially around the heaters. I think up to now this hasn't been looked at because `M303` is usually done once and forgotten. You've patched a pretty big hole, and Marlin is better for it!

marcio-ao, 2017-12-11:
There is only so much software can do. Has anyone considered using a [thermal cutoff fuse](https://www.digikey.com/products/en/circuit-protection/thermal-cutoffs-thermal-fuses/146?utm_adgroup=Circuit%20Protection&gclid=EAIaIQobChMI_oHQtrCC2AIVk19-Ch0qmgN7EAAYASAAEgJvzvD_BwE) in the hotend and/or thermal bed?

PR #8126 (2017-10-29): [1.1.x] Fix M303 thermal protection #8103
The temperature sanity checking logic was not applied during M303
(pid autotuning) because instead of setting a target temperature, PID_autotune
directly manipulated the pwm values.  When PIDTEMP/PIDTEMPBED is
enabled, PWM values rather than the target temperature determine whether
the heater is on.  I changed the temperature sanity checking logic to look directly at the PWM values
when pid is enabled.

I tested M303 with the changes and they fix #8103 .

Edit: I also added calls to disable the heaters if PID_autotune errors out.