MarlinFirmware/Marlin #8105: Re-Arm Compile is failing after #8077

apballard, 2017-10-26:
commit 90a14b4f5be5dea76728181d28b5647c9fb62c46

### Compile Error

unsure what what config is enabled to do this....  thinking sdcard?

```
Compiling .pioenvs/Re-ARM/src/src/HAL/HAL_LPC1768/persistent_store_impl.o
Compiling .pioenvs/Re-ARM/src/src/HAL/HAL_LPC1768/watchdog.o
Marlin/src/HAL/HAL_LPC1768/persistent_store_impl.cpp: In function 'bool HAL::PersistentStore::access_start()':
Marlin/src/HAL/HAL_LPC1768/persistent_store_impl.cpp:56:73: error: cannot convert 'uint16_t* {aka short unsigned int*}' to 'UINT* {aka unsigned int*}' for argument '4' to 'FRESULT f_write(FIL*, const void*, UINT, UINT*)'
res = f_write(&eeprom_file, &eeprom_erase_value, 1, &bytes_written);
^
Marlin/src/HAL/HAL_LPC1768/persistent_store_impl.cpp: In function 'bool HAL::PersistentStore::write_data(int&, const uint8_t*, uint16_t, uint16_t*)':
Marlin/src/HAL/HAL_LPC1768/persistent_store_impl.cpp:114:64: error: cannot convert 'uint16_t* {aka short unsigned int*}' to 'UINT* {aka unsigned int*}' for argument '4' to 'FRESULT f_write(FIL*, const void*, UINT, UINT*)'
s = f_write(&eeprom_file, (void *)value, size, &bytes_written);
^
Marlin/src/HAL/HAL_LPC1768/persistent_store_impl.cpp: In function 'bool HAL::PersistentStore::read_data(int&, uint8_t*, uint16_t, uint16_t*)':
Marlin/src/HAL/HAL_LPC1768/persistent_store_impl.cpp:142:60: error: cannot convert 'uint16_t* {aka short unsigned int*}' to 'UINT* {aka unsigned int*}' for argument '4' to 'FRESULT f_read(FIL*, void*, UINT, UINT*)'
s = f_read(&eeprom_file, (void *)value, size, &bytes_read);
```

Spawn32, 2017-10-26:
This seems to be the same problem i mentioned in:
https://github.com/MarlinFirmware/Marlin/issues/8099

There is a fix for it, but it's not merged....

Roxy-3D, 2017-10-26:
I'll do it...  But I'm having trouble figuring out how to merge this:

Could someone be so kind and merge 1b6497a

I can't find the higher level Pull Request associated with it.

apballard, 2017-10-26:
Ok, I'm gonna close this in favour of #8099

closed 2017-10-26 by apballard

Roxy-3D, 2017-10-26:
I merged #8063   Please let us know if that resolves your issue.