MarlinFirmware/Marlin #8116: [FR] Printer wait, I applying the glue on the bed.

cedcaz, 2017-10-27:
Hello,
I haven't LCD screen (an old computer with Linux and Repetier-Host pilots the printer).
But I would want to use the M0 command with a switch button.
-M28 etc... (Preparation of the printer, Bed temperature ok...)
-M0 printer waiting for me to spread the glue on the bed.
-When I'm finished applying the glue I click on the button and the printer starts its work
My question: how to activate M0 without LCD screen? On what pins of Ramps 1.4 to connect the switch?

mcmatrix, 2017-10-27:
Simple workaround is to use wait or delay.
http://marlinfw.org/docs/gcode/G004.html
Insert into your startup script in right position.
For example. (Check needed gcodes from docs).
....
Beep
Wait a minute or two
Beep
....

cedcaz, 2017-10-27:
Ok, that's not true what I wanted, but it's actually the simplest solution. Thank you mcmatrix

mcmatrix, 2017-10-27:
There’s also a pin state wait gcode
http://marlinfw.org/docs/gcode/M226.html
If you have unused endstop socket or actually any digital pin will work.

For example, cheap endstop switch
https://m.aliexpress.com/s/item/32549454222.html?trace=storeDetail2msiteDetail

Connect it to unused endstop socket and find what is the needed pin number and use gcode to wait until you click endstop

mcmatrix, 2017-10-27:
Any arduino button module will work too.
Search module from google.
Just find a right pins (signal, +5v, ground) from board.

cedcaz, 2017-10-27:
YES! that's exactly what I was looking for. BIG thank you

mcmatrix, 2017-10-27:
I'm glad it helped. You can close this issue.